mongoose-os-apps/shelly-homekit #2: help flashing

andyblac, 2020-02-07:
hi,

my Shelly 1 does not have original firmware on it, I purchase from Voltio.cz it has custom HomeKit firmware, I would like to test your firmware, could you help me flash via https://github.com/espressif/esptool ? or do know a better way to refresh your firmware on a Mac via FTDI (Serial header) ?

Thanks
Andy.

kzyapkov, 2020-02-07:
You can flash the Shelly1 over a wired connection using mos. Download and install mos following these instructions: 

https://mongoose-os.com/docs/mongoose-os/quickstart/setup.md

You can then use mos to flash the shelly-homekit-Shelly1.zip file. It may be a good idea to first dump your current flash contents, so you can go back to the Voltio.cz-provided firmware if needed.

andyblac, 2020-02-07:
ok got it installed, just waiting for my FTDI tool to be delivered, could you explain how to dump firmware ?, I'm new to all this

rojer, 2020-02-07:
once you've connected the serial port and put the device into boot-loader mode (by driving gpio0 low), to dump entire flash:

```
$ mos flash-read --platform esp8266 --esp-baud-rate 921600 flash.bin
```

this should produce a 2M flash.bin file.

to flash this (or any other Mongoose OS-based) firmware:

```
$ mos flash --esp-baud-rate 921600 shelly-homekit-Shelly1.zip
```

to later flash it back:

```
$ mos flash-write --platform esp8266 --esp-baud-rate 921600 0 flash.bin
```

mos will use the first available serial port, which is usually the correct one. but if you have multiple in your system, add `--port PORT` to each command, where PORT is the name of the serial port of the FTDI adapter - one of the COMx devices on Windows, /dev/ttyUSBx on Linux, etc.

andyblac, 2020-02-07:
really appreciate all the help, when you say boot-loader mode (by driving gpio0 low), how do I do that ?

rojer, 2020-02-07:
ok, i will write up detailed instructions with pictures tonight

andyblac, 2020-02-07:
thanks a lot, you got a PayPal for donations?

rojer, 2020-02-07:
i am not accepting donations at this time, but thanks for offering!

rojer, 2020-02-07:
@andyblac see [here](https://github.com/mongoose-os-apps/shelly-homekit/blob/master/docs/flashing.md).

closed 2020-02-07 by rojer

bugfixer1992, 2021-02-03:
Hello,
I am experiencing a problem when trying to flash the shelly-homekit-Shelly1.zip
mos flash --esp-baud-rate 921600 shelly-homekit-Shelly1.zip
leads to an error message. "Image 0x1fc000 overlaps with system params area (16384 @ 0x1fc000)"
Does anyone have an idea how to fix this?
Thanks o lot in advance!

timoschilling, 2021-02-03:
Maybe this wiki pages will help:
- https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development
- https://github.com/mongoose-os-apps/shelly-homekit/wiki/Serial-connection

bugfixer1992, 2021-02-03:
Thank you for the fast answer, @timoschilling.
I followed the wiki and get the same mistake again.
"mos flash http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip"
leads to the error
"Image 0x1fc000 overlaps with system params area (16384 @ 0x1fc000)"
I attached a screenshot for further investigation.
[image]
Do you have any further idea? 

Thanks :-)

andyblac, 2021-02-03:
maybe @rojer can offer some assistance here ?

rojer, 2021-02-03:
this looks like old mos utility version what does `mos --version` report?

bugfixer1992, 2021-02-03:
Thanks a lot @rojer!!!!!
My version was
[image]
when I downloaded the new mox.exe everything works like a charm. 
[image]
Thank you sooooo much! You all made my day!

rojer, 2021-02-03:
any time :)

Bodie2005, 2021-02-20:
> really appreciate all the help, when you say boot-loader mode (by driving gpio0 low), how do I do that ?

You were successful to change voltio.cz firmware? I can't start my Shelly1 with voltio.cz to boot-loader mode.

andyblac, 2021-02-20:
> > really appreciate all the help, when you say boot-loader mode (by driving gpio0 low), how do I do that ?
> 
> You were successful to change voltio.cz firmware? I can't start my Shelly1 with voltio.cz to boot-loader mode.

i have reflashed a few voltio.cz devices to this firmware via serial with no issues.

Bodie2005, 2021-02-22:
I have changed USB / serial from CH340 chipset to FDTI FT232 and I can start in boot-loader mode now. But unfortunately if I try dump flash I have error: 

$ mos flash-read --platform esp8266 --esp-baud-rate 921600 flash.bin
Using port /dev/cu.usbserial-A50285BI
Opening /dev/cu.usbserial-A50285BI @ 115200...
Connecting to ESP8266 ROM, attempt 1 of 10...
  Connected, chip: ESP8266EX
Running flasher @ 921600...
  Flasher is running
Reading 2097152 @ 0x0...
Error: /private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/common/slip.go:54: invalid SLIP starting byte: 0xff
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/esp/flasher/flasher_client.go:400: flash read failed @ 0xd00
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/esp/flasher/read_flash.go:52: failed to read 2097152 @ 0x0
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash_read.go:97: 
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/main.go:198: flash-read failed
exit status 1
Command completed.

Could anybody help me? I am using Mac.

andyblac, 2021-02-22:
what version of mos ?

rojer, 2021-02-22:
this looks like baud rate is too high. try removing `--esp-baud-rate 921600` or setting `--esp-baud-rate 0`

Bodie2005, 2021-02-23:
> what version of mos ?

Serial console
2.19.1~brew

Bodie2005, 2021-02-23:
> this looks like baud rate is too high. try removing `--esp-baud-rate 921600` or setting `--esp-baud-rate 0`

Hello rojer, unfortunately the same issue:

mos flash-read --platform esp8266 --esp-baud-rate 0 flash.bin
Using port /dev/cu.usbserial-A50285BI
Opening /dev/cu.usbserial-A50285BI @ 115200...
Connecting to ESP8266 ROM, attempt 1 of 10...
  Connected, chip: ESP8266EX
Running flasher @ 0...
  Flasher is running
Reading 2097152 @ 0x0...
Error: /private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/common/slip.go:54: invalid SLIP starting byte: 0x30
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/esp/flasher/flasher_client.go:400: flash read failed @ 0x800
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/esp/flasher/read_flash.go:52: failed to read 2097152 @ 0x0
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash_read.go:97: 
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/main.go:198: flash-read failed
exit status 1
Command completed.

--------------

$ mos flash-read --platform esp8266 flash.bin
Using port /dev/cu.usbserial-A50285BI
Opening /dev/cu.usbserial-A50285BI @ 115200...
Connecting to ESP8266 ROM, attempt 1 of 10...
  Connected, chip: ESP8266EX
Running flasher @ 921600...
  Flasher is running
Reading 2097152 @ 0x0...
Error: /private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/common/slip.go:54: invalid SLIP starting byte: 0x00
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/esp/flasher/flasher_client.go:400: flash read failed @ 0x200
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/esp/flasher/read_flash.go:52: failed to read 2097152 @ 0x0
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash_read.go:97: 
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/main.go:198: flash-read failed
exit status 1
Command completed.

Bodie2005, 2021-02-23:
> this looks like baud rate is too high. try removing `--esp-baud-rate 921600` or setting `--esp-baud-rate 0`

Hello rojer, unfortunately the same issue:

mos flash-read --platform esp8266 --esp-baud-rate 0 flash.bin
Using port /dev/cu.usbserial-A50285BI
Opening /dev/cu.usbserial-A50285BI @ 115200...
Connecting to ESP8266 ROM, attempt 1 of 10...
  Connected, chip: ESP8266EX
Running flasher @ 0...
  Flasher is running
Reading 2097152 @ 0x0...
Error: /private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/common/slip.go:54: invalid SLIP starting byte: 0x30
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/esp/flasher/flasher_client.go:400: flash read failed @ 0x800
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash/esp/flasher/read_flash.go:52: failed to read 2097152 @ 0x0
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/flash_read.go:97: 
/private/tmp/mos-20210208-36530-1bwv7ue/mos-9bf406d90c65fca0b377220940735015dcfbb76c/cli/main.go:198: flash-read failed
exit status 1
Command completed.

--------------

Bodie2005, 2021-02-23:
> this looks like baud rate is too high. try removing `--esp-baud-rate 921600` or setting `--esp-baud-rate 0`

Hi rojer. It looks, that problem was with USB-C (I am using original Apple USB C to USB A), it doesn't work. At the end I have tried my old Air with native USB A 
[image]
and I can dump original firmware + add new. 
[image]

Thank you for you patience and help.

Bodie2005, 2021-02-23:
> what version of mos ?

Thank you andyblac for you patience and help. It is working now.

simonfield76, 2021-02-23:
With my interface I need to disconnect and reconnect usb after any command.

Read fw disconnect and reconnect usb
Erase disconnect and reconnect usb
Flash like above

I’ve seen many people using this trick and it always went through.

Inviato da iPhone

simonfield76, 2021-02-23:
Ah sorry I didn’t see that you were using a Mac this hint was for windows users

Inviato da iPhone

Il giorno 23 feb 2021, alle ore 09:37, simone campo <concierge-bentley@hotmail.it> ha scritto:

 With my interface I need to disconnect and reconnect usb after any command.

Read fw disconnect and reconnect usb
Erase disconnect and reconnect usb
Flash like above

I’ve seen many people using this trick and it always went through.

Inviato da iPhone