mongoose-os-apps/shelly-homekit #3: Shelly1 PM

RRostron, 2020-02-07:
Good evening

Would this work on the Shelly1 PM obviously without the PM features?

Thanks

rojer, 2020-02-07:
i think it should but i have no device to test.
the safest way to test would be to make a backup using the serial connection as described [here](https://github.com/mongoose-os-apps/shelly-homekit/blob/master/docs/flashing.md) and give it a go.

RRostron, 2020-02-07:
No probs, thanks I will give it a try and let you know.

hfagelnour, 2020-02-08:
Any luck testing?

RRostron, 2020-02-08:
It will be some time later today I will be trying it.

RRostron, 2020-02-08:
i built the latest shelly1 firmware and flashed it to the shelly1 pm model, after the flash i temporarily powered it up and the setup went ok and its now added to homekit ok, i have setup as a momentary switch.  but i cannot fit it to the lighting circuit till tomorrow to test it actually works or not  :D

rojer, 2020-02-08:
if it clicks when you set it to on and off, then it's fine

RRostron, 2020-02-08:
hi 
no it doenst make any clicking noise, i have only L and N connected just to power the device.

rojer, 2020-02-08:
the relay should still be clicking when you turn it on and off via the home app. maybe it uses different pin for relay control, i don't know

RRostron, 2020-02-08:
i will try with the original firmware,  this was new out of the box,  backed up and then flashed 1.2.0

rojer, 2020-02-08:
yes, it uses different pins. but these are configurable without even rebuilding the firmware, try this:

mos --port ws://192.168.X.Y/rpc config-set sw1.in_gpio=4 sw1.out_gpio=15

use the device's IP address for 192.168.X.Y

RRostron, 2020-02-09:
hi, 
that gives an error

Unknown Command

im running the command in the mos tool in windows 10, my laptop is on the same wifi network as the shelly

RRostron, 2020-02-09:
ok,  ignore the above, seems the command doenst want to work inside the mos tool,
but on the normal windows command line it works fine

c:\mos>mos --port=ws://192.168.6.88/rpc config-set sw1.in_gpio=4 sw1.out_gpio=15
Getting configuration...
Setting new configuration...

c:\mos>

now it clicks :D

rojer, 2020-02-09:
can you check the input too?

RRostron, 2020-02-09:
hi yes that works too

andyblac, 2020-02-10:
created a PR, not sure if I did it correct, please review code.
https://github.com/mongoose-os-apps/shelly-homekit/pull/10

rojer, 2020-02-10:
Added to README and makefile, should be done now.

closed 2020-02-10 by rojer