mongoose-os-apps/shelly-homekit #4: Devices will not obtain IP address

bolagnaise, 2020-02-08:
I successfully installed the firmware on some shelly 1s and 2.5s in my home. After 1 day many of them will no longer connect to my wifi and get an IP, some are working fine. I can reset them and force them into AP mode and then re-enable wifi and they still do not connect. It’s very strange.

rojer, 2020-02-08:
it is indeed... to have any hope of debugging this, we will need logs from one of the devices - we need to see wifi activity.
logs can be obtained via serial connection. please set up wiring as described [here](https://github.com/mongoose-os-apps/shelly-homekit/blob/master/docs/flashing.md#wiring) but do not connect GPIO0 to GND and 3v3 pins, only RX< TX and GND. in this mode firmware will run and print diagnostic information to the serial port, which should include wifi connection attempts and errors. capture that output with any terminal emulator (at 115200 8-N-1) or use `mos console` which should do the trick.

bolagnaise, 2020-02-10:
I have fixed the issue. I deleted the pairing code in AP mode and then re-enabled wifi and then reset up with new code and it’s all working fine now. Weird issue, might be related to me turning main power off an on multiple times to setup Shellys

rojer, 2020-02-10:
yeah, not sure what the deal is.
anyway, feel free to reopen if the issue comes back.

closed 2020-02-10 by rojer