mongoose-os-apps/shelly-homekit #5: Shelly 2.5 as roller shutter?

realguarroman, 2020-02-08:
Hello

Congratulations! I was waiting for a firmware for using only home kit, now I'm using homebridge and a shelly plugin.
If I install this firmware... Will I be able to configure my shelly 2.5 as a roller shutter in homekit?

Thanks

macguru341, 2020-02-08:
Like for this , I’m also need it

rojer, 2020-02-08:
it's a pretty big feature, no ETA.
if anyone is willing to contribute, PRs welcome.

andyblac, 2020-02-08:
I do not yet know Mongoose, I am a python coder, but I am willing to learn. once I get setup so I can build / flash the fun begins 😄.

@rojer do you have a discord server so we can chat direct ?

rojer, 2020-02-08:
most of your questions will probably be general mongoose os dev questions, you can use https://gitter.im/cesanta/mongoose-os
i'm hanging out there, and there are other folks that may be able to help

andyblac, 2020-02-08:
thanks I'll take a look.

macguru341, 2020-02-27:
Hi, again great work, is there any ETA on Shelly 2.5 for shutter?

rojer, 2020-02-27:
it's the first thing on my list, but haven't had the time yet

PatrickHrdl, 2020-03-08:
@rojer 
Hi! GREAT WORK! Thank you!
It would be great if it would soon be possible to use SHELLY 2,5 as a shutter switch in HomeKit. May I give you some money, maybe it will be faster? :-) :-) :-)

Thank you! BR, Patrick

rojer, 2020-03-08:
tell me in more detail, how is your setup wired?

PatrickHrdl, 2020-03-08:
Currently I have not yet installed the SHELLY 2.5.

I have a Rollershutter double switch. (With mechanical interlock)

One switch for UP, the other for DOWN. 

What else do you want to know? 
It should work like the SHELLY firmware, but for HomeKit. 

The SHELLY takes the switch impulse and then drives itself (after calibration) completely down or up. Or even a percentage setting is possible.

rojer, 2020-03-09:
i am not familiar with how shutters work, so i have all sorts of questions, for example - i assume one output is used to close and the other is used to open, but should output stay on or just produce a single pulse? if it should stay on, for how long?

PatrickHrdl, 2020-03-09:
Yes One to Close and one to Open.

I think (!) the SHELLY 2,5 Makes the calibration and than knows how long (in seconds) to close and open.

Liebe Grüße,
Patrick Hrdliczka

+43 676 / 76 76 706
office@p-sign.at

realguarroman, 2020-03-09:
Maybe this video could help https://www.youtube.com/watch?v=fhp8DooeguI&list=PLl9nrhAAwWskc4jQxhlaedn0M57QOu_yj
It seems that there are two variables storing the amount of time used for open and close the shutters.
In the video they use the app Eve for adjusting this values and more things

eurich, 2020-03-10:
Shelly 2 & 2.5 stock firmware  has got different types of open / close working time:

1) calibration - use power consumption in order to determine the open / close working time.
2) manual setup open / close working time.. (in seconds)

Most important feature is the "interlock", so you can use relay1 (open) or relay2 (close) but not both at the same time..

PatrickHrdl, 2020-03-10:
But with one and the Same Button in HomeKit and also with the percentage of the opening:) 

Liebe Grüße,
Patrick Hrdliczka

+43 676 / 76 76 706
office@p-sign.at

Thotty1986, 2020-03-19:
upup :=) Need the funktion :)

rojer, 2020-03-19:
> Maybe this video could help https://www.youtube.com/watch?v=fhp8DooeguI&list=PLl9nrhAAwWskc4jQxhlaedn0M57QOu_yj
> It seems that there are two variables storing the amount of time used for open and close the shutters.

that's right. and they probably use current sensor for auto learning - when end is reached and shutters stop moving, current consumption will go up significantly.

> In the video they use the app Eve for adjusting this values and more things

unfirtunately, i am very short on spare time now, so no promises...

PatrickHrdl, 2020-03-23:
Can i help you with Money die programming Roller Shutter? :)

rojer, 2020-03-23:
at this point i think the biggest barrier for me personally is just not having the hardware and not being familiar with it.
can someone suggest what i can buy that would be representative of what's out there in the field?
if someone is willing to contribute hardware for me to work with, that would be even better.

PatrickHrdl, 2020-03-23:
I will support you with € 200,- if you can offer me a HomeKit Firmware for Roller shutter 
(Like SHELLY 2.5 over homebridge!) :)))

Liebe Grüße,
Patrick Hrdliczka

+43 676 / 76 76 706
office@p-sign.at

GrafKanex, 2020-04-01:
On Amazon (https://www.amazon.de/gp/offer-listing/B085HKH8LR/ref=sr_1_6?s=diy&search-type=ss&keywords=HomekitBG&qid=1585689734&sr=1-6) you can buy a shelly 2.5 with the HomeKit firmware as a roll shutter. But nowhere I can find the firmware to download or find the right settings to set it as a shutter, maybe one of you can help me.

BlancoDeMallloca, 2020-04-07:
Hi all,

Thank you very much for the flash software. This is great work. 

I am really interested in this development too. Unfortunately I am not a software expert. 

> that's right. and they probably use current sensor for auto learning - when end is reached and shutters stop moving, current consumption will go up significantly.

There is another video in German to show the usecase. https://m.youtube.com/watch?v=h0uxqDSpTz8

I think the endpositions (fully opened and fully closed) of the Roller are also determined by the motor of the roller shutter. So this feature could be implemented in a second step. 

The main thing is to implement the feature that HomeKit detects Shelly 2.5 as a roller shutter and not as two separate switches. So the interlock would be Implemented to save the life of the roller shutter motor. 

Thanks for your efforts.

kaeferfreund, 2020-04-12:
I'm currently working on this

PatrickHrdl, 2020-04-12:
Wow! This would be very nice!

Liebe Grüße,
Patrick Hrdliczka

+43 676 / 76 76 706
office@p-sign.at

BlancoDeMallloca, 2020-04-12:
Cool! Thank you very much. This is awesome. 

Will there be a pre release of the interlock? I am a little bit worried to „schredder“ my roller shutter if I am switching on both Shelly 2.5 channels at the same time

kaeferfreund, 2020-04-12:
Currently I'm testing the software interlock with my own shutters, no problems so far. As development takes some time and @rojer provided such a great foundation, he should def. receive the compensation for his mos development claimed in his licensing program. This is why I'm still figuring out how and when the whole thing will be released. But I' glad to see that there is interest in this solution :)

rojer, 2020-04-12:
thanks for the kind words. yeah, there is definitely interest, as you can see.
go ahead, get it working and send in a PR.

BlancoDeMallloca, 2020-04-19:
Good morning rojer,
Good morning kaeferfreund,
Once again thanks for the great efforts you provide within this project. 
I saw that there had been an update tonight. Is there any release plan for the shutter functionality? 
Thanks

kaeferfreund, 2020-04-19:
no set timeline - testing works fine so far, but I couldn't find any time to work on this project since last week

Thotty1986, 2020-04-20:
If you are interested, I can help you testing if you want

pintopaulofrancisco, 2020-05-02:
You guys did an amazing job with this!! Awesome 👍 
Can this feature be prioritized! for my end will make a lot of diference !!

At this moment I am using the API provided by Shelly Cloud on shortcuts app.
you can take a look at this documentation, I don't know if it helps: https://shelly.cloud/support/shelly-cloud-api-access/

shanghaix, 2020-05-04:
For those who are impatient you can always this firmware from another great project: https://github.com/RavenSystem/esp-homekit-devices Been working everyday for the last 2 months without any issues, works natively on Shelly 2.5 without the need for a hub.
[image]

Thotty1986, 2020-05-05:
Looks great but I would prefer a easy Firmeware to put on the device

shanghaix, 2020-05-05:
By all means, feel free to make a commit and add the feature. As for myself, I don’t know that much about coding, I’m going back to my previous post, while the contributors of this firmware are hard at work adding features voluntarily, if you are impatient I recommend trying the firmware I linked my previous post.

Sursulapitschi, 2020-05-17:
@shanghaix : How can the firmware from https://github.com/RavenSystem/esp-homekit-devices be installed on a shelly 2.5?

Borja90, 2020-05-17:
> thanks for the kind words. yeah, there is definitely interest, as you can see.
> go ahead, get it working and send in a PR.

Hi rojer,

I see a hight interest in the roller functionality and one of the most wanted ones. I know it might be complex but why do not do it easy in the first wave? One idea would be to be able to select the time it works instead of using the “auto” detection as the official Shelly firmware. This will allow you to say for example it takes 8 seconds to close and 12 to open. I completely lack the knowledge of the programming but just an idea of what could be an “easy” way ;)

Thanks for all the work done, this is a great project.

shanghaix, 2020-05-17:
> @shanghaix : How can the firmware from https://github.com/RavenSystem/esp-homekit-devices be installed on a shelly 2.5?

You need to flash using a serial adapter, I recommend using wires from an Ethernet cable as the GPIO holes are very small in the shelly 25 and standard DuPont wires will not fit. The flashing process is very straightforward and doesn’t take more than 2min.

Jaguaza, 2020-05-17:
> > @shanghaix : ¿Cómo se puede instalar el firmware de https://github.com/RavenSystem/esp-homekit-devices en un shelly 2.5?
> 
> Necesita flashear con un adaptador en serie, le recomiendo usar cables de un cable Ethernet ya que los orificios GPIO son muy pequeños en el shelly 25 y los cables DuPont estándar no encajan. El proceso de flasheo es muy sencillo y no lleva más de 2 minutos.

He looks knowledgeable could help @rojer  to program the roller.
My programming knowledge is nil. If not it would help to implement it and I do not have a roller.

Thotty1986, 2020-05-19:
i can try to help programming if you want to

rojer, 2020-05-19:
> i can try to help programming if you want to

please, by all means! you'll be a hero :)

Patisone, 2020-05-19:
First of all thanks for your works, this project is great. Shelly at this time seems the only small devise to bring roller shutter support via wifi without any bridges for Z-Wave and Zigbee. So i'm in the line for this feature too ))

As it was mentioned — main thing is the interlock to be sure there won't be voltage on both outputs at the same time. But also many motors have it's own limiters for up and down positions, that prevents then from rotating even if the on or off signal (voltage) continue. So as a suggestion for some kind of mvp you can make only "open" and "closed" modes than just turns one of the relays and keeps it on, cause the motor will stop by itself anyway. Just an idea...

Pixel-Chris, 2020-05-20:
When it is in HomeKit.... with the „dummy“ switches (See issue #9 and issue #74)
The time can be set in an HomeKit Automation. (When Button x is pressed, then switch on relay switch for 10 seconds and then switch off)

Only the interlock must be set in the Shelly web gui, when the roller shutter mode is active.

Lasso77, 2020-06-18:
Thank you very much for bringing Shelly to Homekit. All your work is highly appreciated!
As many others I am definitely in the queue for the roller shutter support. I urgently need it.
Unfortunately I‘m a total noob when it comes to coding, but I would contribute or pay 200 bucks for a fully configurable Shelly HK firmware, too. 
@all Let‘s collect some money for the project.

alexanderbjelos, 2020-06-18:
@rojer You can create a PayPal MoneyPool where we can collect money for the rolling shutter integration.

rojer, 2020-07-14:
i have test hardware now and starting to work on this.
question to everyone: is there a convention which channel to use for opening and closing?
i.e., channel 1 for output, channel 2 for close or vice-versa?

alexanderbjelos, 2020-07-14:
> i have test hardware now and starting to work on this.
> 
> question to everyone: is there a convention which channel to use for opening and closing?
> 
> i.e., channel 1 for output, channel 2 for close or vice-versa?

Hello @rojer, I use the first channel to open and the second one to close. An option to change the channel (when someone reversed the cables) would be nice.

Alex

rojer, 2020-07-14:
naturally. just want to understand what to choose as the default.

BlancoDeMallloca, 2020-07-15:
Hi rojer, 
the default wiring scheme is displayed in the following graphic:
https://forum.creationx.de/index.php?attachment/3332-shelly2-5-rollladenschaltung-png/
SW1 open
SW2 close
However as Alex already mentioned, it would be great to change the default. 
Thanks for your effort. 
Robert

Thotty1986, 2020-07-16:
If you need somebody to test, please let me know I will do my best to help here .

mientki, 2020-07-30:
Is it possible to join the beta test? I have a lot of experience with HomeKit and roller shutter controls.

rojer, 2020-07-30:
i'm still fighting with #30, i will let you know once i have something to test.

for now, i'm wondering how many are using motors with built-in stop switches and how many would like to use shelly inputs as such?
i have a bunch of [these](https://www.amazon.co.uk/Roller-Shutter-Solroyal-Powerful-Warranty/dp/B00SFXPF5E) to play with, they have stop switches built-in.

andyblac, 2020-07-31:
> i'm still fighting with #30, i will let you know once i have something to test.
>
> for now, i'm wondering how many are using motors with built-in stop switches and how many would like to use shelly [...]

would I be able to control the blind without a physical switch ?, how would these work within homekit, ie a % open, how would HomeKit know when they are open or closed ?

Thotty1986, 2020-07-31:
You can make it by time. But that is not recommend.
Normal is, that the end switch is integrated bei the Rollo Motor.
If you have Raffstors there is a physical switch.

rojer, 2020-07-31:
> would I be able to control the blind without a physical switch ?, how would these work within homekit, ie a % open, how would HomeKit know when they are open or closed ?

it would be represented in homekit as a "Window Covering" device. partial opening will be judged by movement time, we'll know full movement time after calibration.

so my current thinking is that we should expect end switch to be integrated into the circuit, so we expect current to drop to 0 when a limit is reached in either direction.
we will also have a configurable timeout so movement will stop after preconfigured number of seconds no matter what.
inputs will be used as optional external open/close controls.

pivale, 2020-08-03:
I now have a motorized roller shutter with mechanical toggle switch with two buttons. Line always stays "on" on the switch and motor, since it's the motor that stops when it reaches the end.

I'm buying a shelly 2.5 and a replacement momentary (push) switch with two buttons.

Quick questions:
- would it still work with the mechanical toggle? How would it interpret the current always on in the switch? what if I moved the shutters through homekit, how would then the work with the mechanical switch?
- the push buttons only has line "on" for as long as I push. can there by some "intelligent" action for short push (always keeps the line on to the motor) and long push (line "off" after the use let's go of the button)?

Thank you and keep up the great work!

yehuda2525, 2020-08-06:
hi mate good job on firmware,
any news about the roller shutter?
its hard to control the roller in a switch mode...

rojer, 2020-08-06:
still `ENOTIME`, attending to more urgent matters. this is first on my list to do.

Ludelag, 2020-08-07:
good job Rojer!, look at this video, probably you have seen it already, there is a firmware outhere call multy accessory homekit, and it looks like is programmed thru EVE app and you get the response as well in homekit:
https://www.youtube.com/watch?v=_hFVtwb0wug&t=546s if you follow up you can also find the second part and there is also a firmware installation explanation. But I am not so expert to install this firmware but could give a clue to the topic of of roller shutter and garage door-

greensouth, 2020-09-02:
> working on it. when planning the implementation, i realized that i need to do significant refactoring to avoid creating even more of a mess. i'm re-writing existing implementation in proper c++ that will make rearranging parts (e.g. combining two switches into a roller-shutter service) simpler.

When doing it, don't forget to add a pause when quickly switching from one direction to another. I was using the Ravencore firmware just before it was implemented in code, and fried one roller shutter because it hadn't the pause to let the motor stops completely before changing the direction.

;)

michaelnozbe, 2020-09-06:
@rojer thanks again for all your hard work. Also hoping to use Shelly 2.5 as roller-shutter.

On a side note, when I saw this "HomekitBG" person selling Shelly 2.5 with HomeKit support I thought it was a standard Shelly 2.5 with THIS firmware, but turns out they've got a different firmware. You think they developed it themselves?

[160 more replies]

closed 2020-10-02 by rojer