mongoose-os-apps/shelly-homekit #6: Toggle switch

bulyshko, 2020-02-08:
Hello, 

Original Shelly firmware had an option to choose "Button type". Can you please add "Toggle switch" support? e.g. Shelly should change state on every button hit.

Thanks

andyblac, 2020-02-08:
already in I asked for that https://github.com/mongoose-os-apps/shelly-homekit/commit/3fa6854b4822f825bb5bc71720f3f6072de3c4aa

rojer, 2020-02-08:
it is in fact the default. but point taken - we should match all the modes.

bulyshko, 2020-02-08:
Yes, I have seen that change, but for some reason it doesn’t work for me.

andyblac, 2020-02-08:
haven't had chance to test it yet, waiting for flashing leads to arrive should be Monday.

rojer, 2020-02-08:
ok, done, pushed 1.2.0

closed 2020-02-08 by rojer

bulyshko, 2020-02-08:
Thanks for adding more modes. I can confirm that now everything works as expected.