mongoose-os-apps/shelly-homekit #7: fails to build

andyblac, 2020-02-08:


rojer, 2020-02-08:
works for me.

(please provide more details)

andyblac, 2020-02-08:
here is log 
[build.log](https://github.com/mongoose-os-apps/shelly-homekit/files/4174180/build.log)

rojer, 2020-02-08:
to build locally (--local) you'll need to install Docker. Docker COmmunity Edition will do.
alternatively, you can use remote build server - just drop `--local` and it will use a default one.

andyblac, 2020-02-08:
ok I'll try thanks

andyblac, 2020-02-08:
now get

```Error: /private/tmp/mos-20200204-9510-12i29jp/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/cli/build_remote.go:333: error response: 400: max size 5242880 exceeded (upload size is 5849621)
/private/tmp/mos-20200204-9510-12i29jp/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/cli/build.go:221: 
/private/tmp/mos-20200204-9510-12i29jp/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/cli/build.go:164: 
/private/tmp/mos-20200204-9510-12i29jp/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/cli/main.go:197: build failed
```

rojer, 2020-02-08:
and before that?

andyblac, 2020-02-08:
```Connecting to https://mongoose.cloud, user test
Uploading sources (5850346 bytes)
Error: /private/tmp/mos-20200204-9510-12i29jp/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/cli/build_remote.go:333: error response: 400: max size 5242880 exceeded (upload size is 5849621)
```

rojer, 2020-02-08:
ok, delete deps folder

andyblac, 2020-02-08:
👍 built ok

closed 2020-02-08 by andyblac