mongoose-os-apps/shelly-homekit #8: question about HomeKit Serial number

andyblac, 2020-02-08:
is the taken from Mac address so we can add multiple devices into home kit, or is it fixed ?

rojer, 2020-02-08:
serial number is determined by the device.sn configuration value.
default value for this setting is set [here](https://github.com/mongoose-os-apps/shelly-homekit/blob/16b22c6f1216e9e882a7e1bd2078318a95a0c880/mos.yml#L18) but it can be later set via any of the (many) methods that can be used to change config values. 
for example:

```
$ mos --port=ws://shelly1-68b743.local/rpc config-get device.sn
000000
$ mos --port=ws://shelly1-68b743.local/rpc config-set device.sn=012345
Getting configuration...
Setting new configuration...
$ mos --port=ws://shelly1-68b743.local/rpc config-get device.sn
012345
```

andyblac, 2020-02-08:
so atm it static ?, so only 1 device can be added to HomeKit, without changing code and rebuild ?

andyblac, 2020-02-08:
oh I understand it can be change in config on shelly1 after flashing, it might be an idea to auto generate it from the last 4 digits from Mac address, something like `Shelly1-A2DE`

rojer, 2020-02-08:
i haven't had any problems with devices with the same serial number.
what homekit really cares about is the accessory id. if you duplicate that, then things get weird (device not appearing in the list, etc).
however, in this firmware ID is randomly generated on every homekit reprovisioning (first boot or whenever you press "reset" in the UI). it is not static and not derived from serial number, so having the same serial should not be an issue. in this firmware it is used purely for cosmetics - presented in the UI and maybe for your own tracking.
so, i'm going to close this issue but if you find any problems, please reopen.

closed 2020-02-08 by rojer