mongoose-os-apps/shelly-homekit #9: would it be possible to expose Single / Double press / Long Press in HomeKit

andyblac, 2020-02-08:
as the title, would it be possible to expose Single / Double Press & Long Press actions in HomeKit, this would be handy to control the switch differently for those actions, obviously this would require push button type to be enabled.

example that could be setup in HomeKit GUI.
Single Press =  normal switch operation.
Double Press = turn off after xx
Long Press = invoke a HomeKit scene.

andyblac, 2020-02-08:
when doing a double or long press, this would not have to switch the shelly, it would purely give HomeKit actions that you could use.

if you wanted the shelly to switch you could just link to the HomeKit Shelly switch.

andyblac, 2020-02-08:
this is what I mean

[image]

rojer, 2020-02-08:
in terms of HAP services, that would mean implementing "8.37 Stateless Programmable Switch" instead of "8.38 Switch".
i think we can make it configurable, whether shelly switch exports a stateful or stateless HAP switch.
sure, it's possible, not that difficult either.

andyblac, 2020-02-08:
but not both, understood. so shelly would become a virtual switch only, got it.

rojer, 2020-02-08:
ok, i'm talking nonsense here. "Switch" is an output gadget, taking commands from the controller while "Stateless Programmable Switch" is an input gdaget, sending events to the controller.
so in fact, i think the right way to do is to export the input pin as a stateless programmable switch if the mode for it is set to "detached".

mgoeller, 2020-02-10:
It should be possible to deploy two HomeKit Services (e.g. 1: Stateless Programmable Switch and 2: Lightbulb) both on the same Accessory. So the input GPIO would correspond to the service 1 and the output to service 2. Now of course the input is not directly switching the Lightbulb anymore, but one can link that action as a single / double / long press in HomeKit. Otherwise the Lightbulb can be controlled only via HomeKit.

Pros:
Use different button actions

Cons:
Increased delay for switching Lightbulb
Only works over HomeKit protocol (e.g. not without WiFi and only with an 0iOS device available that executes the action)

andyblac, 2020-02-10:
would that also mean the manual switch would not work anymore ?

mgoeller, 2020-02-10:
I would do both possibilities, the drawbacks are too huge. But some may need it: e.g. replace a switch next to the door:
-don't use the traditional switch anymore to switch the corridor light, instead do it motion based with a separate sensor
-use the traditional switch and link it with Homekit and single/double/long press to power off the whole house

That is basically what I have done using esp-homekit and a custom firmware. But all other switches that I replaced (not next to the apartment door) I used a single service to have the traditional switch working without delay and even if WiFi is down or similar.

rojer, 2020-02-10:
> It should be possible to deploy two HomeKit Services (e.g. 1: Stateless Programmable Switch and 2: Lightbulb) both on the same Accessory. 

that's right, that's the idea. but only do it if the user chooses the function for the input in the web interface. we have the "detached" mode for the input, so we can use that to mean "i want the input detached from the switch and exposed as a HAP button instead".

rojer, 2020-02-10:
in terms of implementation, it's not too difficult - there's already an example for switch service definiiton in the code, can be used as a template for the stateless switch as well.
a bit of tap-dancing will need to be done around adding and removing services from the accessory, including managing the HAP configuration version (so controller re-enumerates services), but it shouldn't be too difficult.

andyblac, 2020-02-10:
this all sounds good, I'm trying to learn HAAP / mongoose

jonny9412, 2020-04-01:
Hey there, 
I recently installed my shelly1 with the over the air update and it works very fine.
Just like @andyblac said I would love to activate HomeKit-scenes via the shelly switch.

I'm sorry but I didn't understand all of your chats - is it already possible to do so? Or will you implement this feature in one of the next versions?

Thanks!

andyblac, 2020-04-08:
@rojer maybe an option would be to have a separate fw for each shelly that only allows HomeKit mode as laid out above, but on pressing the physical switch is just actions the HomeKit ie Single, Long, Double presses.

you could add new machines into build "shelly1-hk", "shelly25-hk", and so on, this would save ram on the devices ?

Pixel-Chris, 2020-04-14:
Hello, so this ist my first post...

If that could be done... this will be the best solution for all HomeKit users.
The „Killer“ feature in my opinion.

The best would be to expose a virtual Switch (with single, double and long press function) and a second virtual switch, which controls the real relay  to the Apple HomeKit System.
Both can enabled/disabled in the web gui. So the user can choose if he needs the relay variant and/or HomeKit function variant in HomeKit.

Why both:
I think not only me... so better „many“ are equipped with HomeKit smart bulbs

With „normal“ switches you can cut off the power to these bulbs - so the bulb is offline and can not be controlled over HomeKit.
With HomeKit Switches you can not cut off the power to these bulbs, you can „only“ control it Smart via HomeKit.

So that is the reason why many of us use actually only „normal“ switches and not HomeKit switches

So a combination would be the best solution...
Normally you control the bulbs over the HomeKit functions, but when there is a fault/error you need the cut off the power for a restart, or if you are on holidays you want to cut off the power.

Best regards
PixelChris

Jaguaza, 2020-05-17:
Seria interesante esta función

Pixel-Chris, 2020-05-19:
The user should be able to select what he wants to see in HomeKit...

a) HomeKit Switch to switch the real relay
b) A HomeKit Switch (Single, Double, Longpress)
c) An extra dummy HomeKit Switch (if needed)

prrtzt, 2020-05-19:
+1, would be super interested as well.

geokvant, 2020-05-25:
hi everyone. 

I'd like to understand only one thing. is it possible to keep my Shelly 1 as a Simple switch (installed in momentary state) but add longpress for some scenarios activation? b/c I don't want to do single click for ON and double for "OFF", but want to make long press for "Leaving home" scenario activation.
Is it possible to make in Homebridge installation?

andyblac, 2020-05-25:
you could try your shelly with original firmware and this https://www.npmjs.com/package/homebridge-shelly homebridge plugin.

geokvant, 2020-05-25:
> you could try your shelly with original firmware and this https://www.npmjs.com/package/homebridge-shelly homebridge plugin.

I'm on original firmware now and see longpress option in shelly's admin page but this stuff has no options in homebridge page or in homekit app so setup.
should I do some additional settings in homebridge config to see options in homebridge/homekit?

andyblac, 2020-05-25:
looks like the answer is no then, you'll have to wait for @rojer to hopefully add this support to his firmware.

rojer, 2020-05-25:
so the plan for this is to export a separate "stateless programmable switch" service when mode for the input is set to "detached". it will generate single, double and long press events that you can then use in your automations.

Pixel-Chris, 2020-05-25:
> hi everyone.
>
> I'd like to understand only one thing. is it possible to keep my Shelly 1 as a Simple switch (installed in momentary state) but add longpress for some scenarios activation? b/c I don't [...]

In HomeKit you can use automations for that,
For example for the single press:

If lamp is off turn it on
If Kamp is on turn it off

So you can use it as a „normal“ switch

Two more examples:

Double press:
For controlling the relay in the Shelly to cut the power
(So the Shelly real relay Switch must also exposed to HomeKit)
If power is off turn it on
If power is on turn it off

Long press:
If Alarm scene is on turn it off
If Alarm scene is off turn it on

Or

If leaving home is off turn it on
If it is on turn it off

Pixel-Chris, 2020-05-25:
> so the plan for this is to export a separate "stateless programmable switch" service when mode for the input is set to "detached". it will generate single, double and long press events that you can then use in your automations.

Will it be possible to control the relay in HomeKit?

All available HomeKit switches can only do one thing:

They can cut the power to the device
or they are only smart switches which can control smart devices over HomeKit.

So again (sorry for that):

If both is possible this would be the KILLER Feature !!!!

Can we speed it up?
Are here some more people, who need this feature? can we spend some money to rojer???

geokvant, 2020-05-25:
> 
> In HomeKit you can use automations for that,
>...
> If leaving home is off turn it on
> If it is on turn it off

There is no option to setup long press in homekit in automation scenarios b/c there is no such event available for this Shelly 1 with original firmware.

I don't need double press, I just want to have in addition to standard function on/off (momentary setting) to get longress and bind some action for it in homekit (directly or over homebridge) but without any third party firmware

Pixel-Chris, 2020-05-25:
Original Firmware is Not HomeKit compatible!
And actually this firmware „Shelly-HomeKit“ actually exposes a Shelly 1 only as a „normal“ switch - this switch controls the real relay.

So the idea behind this issue, is to expose a second switch to HomeKit (a smart switch) with single / double and long press function (see screenshot above) 
So you can control the real relay switch and/or other things in HomeKit.

geokvant, 2020-05-25:
> Original Firmware is Not HomeKit compatible!
> And actually this firmware „Shelly-HomeKit“ actually exposes a Shelly 1 only as a „normal“ switch - this switch controls the real relay.

Sorry for my misunderstanding.
I was talking about solution with external gateway based on [Homebridge](https://homebridge.io/) by [nfafina](https://github.com/nfarina) and in this case you don't need to modify original firmware of Shelly devices,  but you need external gateway.

So I was thinking that there is a solution to keep an option for Sinlge Switch to work as it should do and just add another virtual button to homekit that will be programmable over homekit interface for long press as standard scenario.  (without double press b/c for "8.38 Switch". it's not applicable in terms of Shelly) 

So the idea in general sounds like this:
for one physical device (Shelly 1) to homekit interface should be added two items:
1) one logical switch - manages relay to on|of
2) one virtual switch - takes long press and sends info to homekit to start user scenario
NB) _NO double press in this case b/c there is no difference with single press_

This "two switches" option gives ability to control lights directly even without wifi, but adds one more option to initiate user scenario from any of switches.

Pixel-Chris, 2020-05-25:
So you Need a) and c)

`a) HomeKit Switch to switch the real relay
b) A HomeKit Switch (Single, Double, Longpress)
c) An extra dummy HomeKit Switch `

geokvant, 2020-05-25:
I would say that I need A (relay switch) and C (dummy switch), but C should be linked to Longpress of Shelly's brain

Pixel-Chris, 2020-05-25:
I think that will not work

The relay needs to be detached from the switch for the „normal“ HomeKit smart switch to work (single/double/long press)

geokvant, 2020-06-08:
> I think that will not work
> 
> The relay needs to be detached from the switch for the „normal“ HomeKit smart switch to work (single/double/long press)

b/t/w I've found solution for my "problem" with help of [this plug-in](https://www.npmjs.com/package/homebridge-button-platform)
you can read it in this post
https://github.com/alexryd/homebridge-shelly/issues/118#issuecomment-640807308

Pixel-Chris, 2020-06-16:
@rojer 
Will that be integrated in the next time?

rojer, 2020-06-16:
yes, this is the next thing on my list

vladbisceanu, 2020-09-03:
Hi! IS this live in the latest build? Sorry for commenting on an old issue, but this is of high interest to me. Actually, I intend on purchasing around 20 Shellys for all my dumb light switches and this is the feature that will make it or break it.

I want to use the Shellys on dumb switches together with smart lights. The goal is to have dimmable/colored lights but also be able to control them via the old dumb switches. For that, I need to be able to re-write the single press on the switches to not trigger the relay and instead turn on my smart light via homekit (and ideally also be able to trigger the relay via a long switch, for redundancy).

If I understand correctly, this thread debates exactly that and the feature should be available. However, I want to make sure it is doable before purchasing the devices.

Thanks a bunch! This sounds like a lifesaver feature for me.

rojer, 2020-09-03:
no, it hasn't been implemented yet

vladbisceanu, 2020-09-03:
Thanks for the fast reply! Any idea when it will be implemented?

rojer, 2020-09-03:
absolutely no idea. i've been very busy at my day job lately, very little time and energy for side projects...

vladbisceanu, 2020-09-04:
Ok, mate. Can 100% understand how that feels. Thanks for letting me know and good luck with your work! Just to be aware of all the options here: is there any known way of disabling the relay switching on button press with this firmware and just have the button do something else entirely?

rojer, 2020-09-04:
not at present but that is exactly the plan - decouple input from output, export it as an independent hap service.

vladbisceanu, 2020-09-04:
Understood. Really looking forward to that feature, whenever you got the time for it. Would love to offer my support, but I truly have no clue what sorcery you did with this project so I'm at a loss. Again, thanks for being so responsive!

geokvant, 2020-09-04:
> Ok, mate. Can 100% understand how that feels. Thanks for letting me know and good luck with your work! Just to be aware of all the options here: is there any known way of disabling the relay switching on button press with this firmware and just have the button do something else entirely?

you can detach switch from relay over web-interface in original FW 1.8.3 by this menu
[image]
with this setup relay will be controlled only by button in web-interface or over API. I use for my needs this plugin in HB https://www.npmjs.com/package/homebridge-button-platform with Actions menu for every button.

This solution gives me Single press and Long press.

vladbisceanu, 2020-09-04:
> with this setup relay will be controlled only by button in web-interface or over API. I use for my needs this plugin in HB https://www.npmjs.com/package/homebridge-button-platform with Actions menu for every button.
> 
> This solution gives me Single press and Long press.

That's actually perfect as an alternative solution for me. I have a HB setup already running for PC control and other minor things. Can you share more details about how the plugin is configured in your case. I've ordered a Shelly 1 and a 2.5, which should arrive in 1-2 days and I'll be able to test this.

geokvant, 2020-09-04:
In my case I've made usage only Long press for Actions, but Switches not detached. I wanted to add to my switches Long press Actions in HB, so I added 11 Virtual Buttons with those scenarios I wanted for LONG PRESS and added links for them as it described for homebridge-button-platform 
[image]
you can add (with Detached mode switch) short press for Switch to call process in homebridge-button-platform you want to

vladbisceanu, 2020-09-04:
Ok, it all makes sense. Once I have my shellys I'll try this setup and post here on how it works, in case anyone else is interested in this. I might follow-up with noob questions here, if I get stuck. Many thanks for your help!

geokvant, 2020-09-04:
here you can find solution how to setup for single short press of Switch (in detached mode) to send command for one of your virtual buttons so the button single (or double or long) press (they are just 3 virtual actions in this case) toggle any action you want to be done through the shortcuts automation
https://linkdhome.com/articles/how-to-toggle-in-homekit

my case for this is next:
I have a Philips hue light in bathroom that has no wired switch - so it's always powered on. But I can manage it over the Home App and Hue App. But I wanted to turn it on and off (without changing colors or brightness) with same Switch I use for real lights there. So I've added for long press of my Shelly switch this option to toggle my LED Light.

vladbisceanu, 2020-09-12:
Hi everyone. Just following-up on the thread above, here are my updates:

I got my shelly 1 the other day and tried to install it with my dumb light switch. I am using a Livolo touch light switch [like this one](https://www.livoloeurope.eu/livolo-wifi-touch-light-switch-with-glass-panel-vl-c701-11).

1. The first thing I did was try and install it in my wall light box. However, there was not enough space in there for the shelly AND I did not have a neutral. Fortunately, I did have an additional phase wire I was not using, so I wired that to the power and decided to install the Shelly in my ceiling, where neutral is available.
2. The installation went pretty smooth and in a few minutes I was able to boot up the shelly app and switch the lights on and off. However, the actual switch was not working. After some investigation, I found out that Shelly does not send enough power on the SW line to power the tactile switch I'm using (that also has a status LED on it). After some searching, I found [an adapter](https://www.livoloeurope.eu/livolo-lighting-adapter-the-saviour-of-the-low-wattage-led-lamp-VL-PJ01?search=adapter) that the [guys from Livolo mentioned](https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/2116678198431533/) should solve the issue. I ordered that and it should arrive in a few days. Yes, the adventure continues.
3. To still be able to test the "detached" integration mentioned above, I re-installed one of my old on/off light switched in place of the Livolo capacitive one. I then proceeded to install the button-platform plugin for Homebridge and setup a button, as per the instructions.
4. In the Shelly interface, I switched to the Detached mode and configured the actions. Since I have an on/off switch, I configured actions for button on and button off, instead of the short/long press. I'm sending a double-press event on "turn off: because I need two separate triggers in Home. There is no actual double press happening.
[image]
5. In Home app, I simply added turn on and off actions of the button to the single and double press triggers.

All in all, the solution proposed here works great. I am able to control my smart lights with the physical buttons without messing them up. The only downside I can see is that I have no redundancy. In the event that my smart lights and shelly for some reason break, I need to re-wire stuff. However, the likelyhood of that happening is so low I am not concerned.

My end-goals was to use the Livolo touch switches with Shellys in my whole house. I still need to find a solution to make Shelly and Livolo cooperate, which I hope will be that adapter they recommended. I am no electrician, but from my understanding, that is supposed to add Null to the wire running between the light and the switch (sorry if I'm talking nonsense.

If anyone is interested, I can share pictures of my wiring and/or share more details on how that adapter works, once I have it.

Cheers!

vladbisceanu, 2020-09-12:
Later realised the single/double press approach is redundant, so I configured both turn on and turn off to call the single-press event and handled the toggle logic from Shortcuts, like a previous post was mentioning
[image]

rojer, 2020-09-14:
please test this firmware - https://rojer.me/files/shelly/2.1.0/
set input mode to "detached" and you should see a stateless switch show up in the app.
if everything is ok, this will become the next release.

andyblac, 2020-09-14:
> please test this firmware - https://rojer.me/files/shelly/2.1.0/
> set input mode to "detached" and you should see a stateless switch show up in the app.
> if everything is ok, this will become the next release.

got it loaded, 

first issue changing input name does not save.

2nd issue you now have double SW's in HomeKit, when in detached mode, can we hide the original SW please, so only the detached one shows.

3rd everything is treated as a Long Press, via physical switch, ie once you have setup the 3 mode options in homekit, pressing the physical switch everything acts what has been setup as Long Press

vladbisceanu, 2020-09-14:
> please test this firmware - https://rojer.me/files/shelly/2.1.0/
> set input mode to "detached" and you should see a stateless switch show up in the app.
> if everything is ok, this will become the next release.

Tried to update as well, but I'm not getting the new version for some reason.
Used this URL http://192.168.1.2/ota?url=https://rojer.me/files/shelly/2.1.0/shelly-homekit-Shelly1.zip where 192.168.1.2 is my Shelly's IP. I get this message `{"status":"updating","has_update":false,"new_version":"20200827-065344/v1.8.3@4a8bc427","old_version":"20200827-065344/v1.8.3@4a8bc427"}` but nothing happens.

 Do I need to first flash the stable build or something?

andyblac, 2020-09-14:
> Tried to update as well, but I'm not getting the new version for some reason.
> Used this URL http://192.168.1.2/ota?url=https://rojer.me/files/shelly/2.1.0/shelly-homekit-Shelly1.zip where [...]

you have to manually upload via webUI, download the file 1st then open webUI, scroll to bottom click choose file and then upload.

vladbisceanu, 2020-09-14:
Ok, found the issues. File needs to be HTTP, not HTTPS. This was mentioned in the instructions, but I missed it the first time. My bad.

andyblac, 2020-09-14:
> Ok, found the issues. File needs to be HTTP, not HTTPS. This was mentioned in the instructions, but I missed it the first time. My bad.

would adding a `test` option to my flash script be worth doing? so the script could flash test firmware.

andyblac, 2020-09-14:
@rojer looks like the HomeKit connection has gone strange, I'll flash the dbg version so you can take a look

[image]

rojer, 2020-09-14:
> > please test this firmware - https://rojer.me/files/shelly/2.1.0/
> > set input mode to "detached" and you should see a stateless switch show up in the app.
> > if everything is ok, this will become the next release.
> 
> got it loaded on Shelly 2.5 and have SW1 in momentary mode & SW2 in detached mode,
> 
> 1st issue changing input 2 name in webUI overwrites SW1 name, and input reverts back to default.

fixed this one, https://rojer.me/files/shelly/2.1.1/

> 
> 2nd issue you now have double SW's in HomeKit, when in detached mode, can we hide the original SW please, so only the detached one shows. (this is confusing if you have 'show as separate tiles enabled)

well, it still works as an output via ui or homekit. if you want it hidden, set its HAP type to "disabled".

> 
> 3rd everything is treated as a Long Press, via physical switch, ie once you have setup the 3 mode options in homekit, pressing the physical switch everything acts what has been setup as Long Press
> 
> update:
> checking the webUI after pressing single, double or long, everything shows as `long press`

how it's supposed to work:
 * single press: input activated for less than 1s, then released
 * double press: input activated for less than 0.5 s, released, activated again before 0.5.s, released before 1s
 * long press: input held for 1 second or more

rojer, 2020-09-14:
> @rojer looks like the HomeKit connection has gone strange, I'll flash the dbg version so you can take a look

> 4/3/9 pending/active/max

it's normal to have pending connections for a short time after reboot or reconfiguration. device will perform 1 new HAP handshake every 3 seconds to avoid resource spikes, newly acceptet connections will be held pending in the interim. it's only a problem if this persists.

rojer, 2020-09-14:
> > Ok, found the issues. File needs to be HTTP, not HTTPS. This was mentioned in the instructions, but I missed it the first time. My bad.
> 
> would adding a `test` option to my flash script be worth doing? so the script could flash test firmware.

we need to figure something out with beta/test firmware, yeah. what do you suggest? please file as a separate issue though.

vladbisceanu, 2020-09-14:
> please test this firmware - https://rojer.me/files/shelly/2.1.0/
> set input mode to "detached" and you should see a stateless switch show up in the app.
> if everything is ok, this will become the next release.

Tested it myself just now. I can confirm the 1st and 3rd issues @andyblac found (name and long press). Will flash the new 2.1.1 to see if name issue is fixed.

I am also getting the pending connection indefinitely (up for 30 min now and still pending).

Additionally, I am using a traditional on/off switch ([like this one](https://images.app.goo.gl/83zKPB1awQV25MKA8)). With the stock firmware, I would be able to trigger an action when the switch is turned either way (on and off). With this firmware, it seems that action is detected only when switch is turned from on to off. This might not be related to the test firmware just released, but guidance would be appreciated.

andyblac, 2020-09-14:
loaded dbg version 2.1.1 

saving names now fixed. did not noticed the new option to disable HAP thx. but 3 still exists, everything does Long Press actions.

andyblac, 2020-09-14:
@rojer can you take a look at switch `shellyswitch25-8CAAB505B323` it still has 0/7/9 connections. whilst your in see if you see why everything is being actioned as Long Press

[174 more replies]

closed 2020-09-18 by rojer