mongoose-os-apps/shelly-homekit #14: Shelly Dimmer support possible ?

andyblac, 2020-02-12:
as title do you think Shelly Dimmer would possible ?

rojer, 2020-02-12:
in principle, yes. but it has secondary microcontroller, which is an additional complication.

andyblac, 2020-02-13:
I could do with it, would you be willing to take a look if I send you one ?

andyblac, 2020-02-13:
could this help ? https://github.com/alexryd/homebridge-shelly

rojer, 2020-02-13:
no, that's not relevant.

i would not recvommend starting with dimmer. implementing input as push button would be easier and get you acquainted with what implementing a HAP service entails.

andyblac, 2020-02-13:
an a dimmer act as on/off too ?

rojer, 2020-02-13:
what's the use for the dimmer without, you know, dimming functionality?
i think we should hold off declaring support for shelly dimmer until dimming is implemented.
in terms of HAP, that would mean implementing a lightbulb service, which supports variable brightness.

andyblac, 2020-02-13:
so if I purchased a shelly dimmer I could try adding as a switch to start ?

kzyapkov, 2020-02-13:
The dimmer is indeed not the device to start with. If lighting is your thing, try the LED drivers, they should be much simpler, using PWM for output control.

As for the dimmer -- we plan on releasing info about the UART protocol used to control the small MCU which does the actual dimming, but no ETA.

andyblac, 2020-02-13:
ahhhh. atm I'm using Philips Hue, but tbh it is unreliable, found your shelly code match more so and it is instant, I was hoping to replace my Hue with Shelly's.

kzyapkov, 2020-02-13:
please drop a note to developers@shelly.cloud, we may be able to help

andyblac, 2020-02-13:
atm I have zero knowledge of HAP or MOS, I am a python coder. mOS looks vastly different, I am trying to read through this git code to try to get an understanding of it.

rojer, 2020-02-13:
right, so dimmer is not a good start. better pick a simpler issue to tackle.
for HAP, you bible is `HAP-Specification-Non-Commercial-Version.pdf`. you can download it from [here](https://developer.apple.com/homekit/). you'll need to login with apple id but it's free of charge.
get familiar with the concepts - controller, accessory, service, characteristic, event.
then read existing code to understand how services are defined and characteristics and events managed. fortunately, it's not that much - all of the "switch" service implemented by shellies is contained [here](https://github.com/mongoose-os-apps/shelly-homekit/blob/master/src/shelly_sw_service.c).

in terms of coding - yeah, it's the good old C, not Python. not much do be done about it, i'm afraid. either you pick it up, or you won't be able to do much.

andyblac, 2020-02-13:
thanks, I have some homework to do 😄

sebastianheise, 2020-02-24:
Hi guys! How are you doing? I've caught up on this thread and had a look at the HAP. 
From my understanding a _dimmer_ would be implemented using the _brightness_ characteristic. Correct? So instead of a switch, the device would need to authenticate as a _light bulb_ service. Is that assumption correct? Unfortunately I also don't have any experience in HAP and MOS, but it sounds like a fun project. If I can help with anything I'm all keen. Just give me a yell. Would be awesome to have support for the Shelly dimmer!

rojer, 2020-02-24:
yes, that is correct.

Olly68, 2020-08-08:
@rojer 
Do you have plans for Shelly Dimmer 2 at the moment?
Or is time still your problem?

It would be great if you can find a solution👌🏻😀

ree109, 2020-08-20:
Would this help with the dimmer - https://github.com/curla92/LED-Dimmer-Homekit

vanrri, 2020-08-20:
Anyone tried it yet?

andyblac, 2020-10-02:
Just so you don't get Bored!!! HAHA, Can This Be Next Please!!!

rojer, 2020-10-02:
yes, i think i'll tackle either dimmer or rgbw2 next. we'll see.

andyblac, 2020-10-02:
cool cheers!!!

andyblac, 2020-10-02:
> yes, i think i'll tackle either dimmer or rgbw2 next. we'll see.

do you need any devices ?, I don't mind donating one of each for you to work with.

rojer, 2020-10-02:
no, i'm all set. i've had a couple for a while now.

andyblac, 2020-10-02:
ok np.

freddair, 2020-10-10:
+1 for Dimmer next :)

rojer, 2020-10-11:
yes, dimmer is next (after garage door is out), in particular dimmer2, which is what i have for testing.

andyblac, 2020-10-11:
> yes, dimmer is next (after garage door is out), in particular dimmer2, which is what i have for testing.

if I can help with debug info for Dimmer1 let me know.

freddair, 2020-11-04:
+1 for Dimmer1

Leo-W536, 2020-11-04:
+1 for Dimmer 1 👍🏻

andyblac, 2020-11-04:
i now have dimmer2, so can help testing that too 😄

ToblerCGN, 2020-11-06:
I got two dimmers 2. I'd like to help, too!

Przemyslaw-Wosko, 2020-11-13:
@rojer can we implemnet shelly dimmer as light bulb instead? it has characteristics to on / off and also dimm 
i will try to update some code, and rise PR. 

What do you think?

rojer, 2020-11-13:
that's exactly the plan, dimmer will be represented as a lightbulb HAP service with "brightness" characteristic.

Przemyslaw-Wosko, 2020-11-13:
Have you started work somewhere? i've noticed it will be lot of work, since everything relays on Switch 
I'd love to collaborate instead of writing whole stuff from scratch

rojer, 2020-11-13:
the biggest chunk of work will be implementing the UART protocol to communicate with the MCU that does the dimming.
adding a HAP service, will be the easy part.
no, i haven't written any code yet, but i have test setup and have looked into the protocol details, i have a reasonable grasp of it now.
will probably get something written this weekend.

rojer, 2020-11-16:
well, this did not go to plan at all.
i was having issues getting ,y firmware to run after OTA, and while messing with serial connections, briefly had both non-isolated serial connected while i turned on the mains power... BOOM! sparks, smoke, lights go out.
when the smoke cleared, the casualties are: 2 of 2 dimmers, usb hub and, crucially, my main laptop.
nothing much to show for it. not a good day...
but i'll be back :)

pivale, 2020-11-16:
> well, this did not go to plan at all.
> i was having issues getting ,y firmware to run after OTA, and while messing with serial connections, briefly had both non-isolated serial connected while i [...]

damn :( sorry to hear that!

freddair, 2020-11-16:
@rojer glad to understand that yourself have not been harmed though! I have a few stock Dimmer1 laying around, so let me know if and how I can provide you with one!

andyblac, 2020-11-16:
@rojer Dam, thats not good, the pesky none isolated serial is dangerous. Just glad to here your OK though, kit can be replaced. I think we need to get you some donations in to help you replace that laptop. One from me on the way!!

Przemyslaw-Wosko, 2020-11-16:
@rojer thanks for effort, i didn't knew this could go soo bad and i feel sorry for your laptop, i hope it wasn't something extremally expensive

freddair, 2020-11-16:
@rojer do you already have the needed Dimmer UART specification? Otherwise maybe the sample config of HAA firmware could help:

`{"t":30,"d":0.1,"b":[{"g":12,"i":1,"p":0},{"g":12,"t":0}],
"0":{"u":[{"v":"010202060000f401000000ff04"}]},
"y0":[{"v":1,"0":{"u":[{"v":"010202060a00f4010000010904"}]}},
{"v":5,"0":{"u":[{"v":"010202063200f4010000013104"}]}},
{"v":10,"0":{"u":[{"v":"010202066400f4010000016304"}]}},
{"v":15,"0":{"u":[{"v":"010202069600f4010000019504"}]}},
{"v":20,"0":{"u":[{"v":"01020206c800f401000001c704"}]}},
{"v":25,"0":{"u":[{"v":"01020206fa00f401000001f904"}]}},
{"v":30,"0":{"u":[{"v":"010202062c01f4010000012c04"}]}},
{"v":35,"0":{"u":[{"v":"010202065e01f4010000015e04"}]}},
{"v":40,"0":{"u":[{"v":"010202069001f4010000019004"}]}},
{"v":45,"0":{"u":[{"v":"01020206c201f401000001c204"}]}},
{"v":50,"0":{"u":[{"v":"01020206f401f401000001f404"}]}},
{"v":55,"0":{"u":[{"v":"010202062602f4010000012704"}]}},
{"v":60,"0":{"u":[{"v":"010202065802f4010000018b04"}]}},
{"v":65,"0":{"u":[{"v":"010202068a02f4010000010904"}]}},
{"v":70,"0":{"u":[{"v":"01020206bc02f401000001bd04"}]}},
{"v":75,"0":{"u":[{"v":"01020206ee02f401000001ef04"}]}},
{"v":80,"0":{"u":[{"v":"010202060032f4010000013104"}]}},
{"v":85,"0":{"u":[{"v":"010202065203f4010000015404"}]}},
{"v":90,"0":{"u":[{"v":"010202068403f4010000018604"}]}},
{"v":95,"0":{"u":[{"v":"01020206b603f401000001b804"}]}},
{"v":100,"0":{"u":[{"v":"01020206e803f401000001ea04"}]}}]}`

rojer, 2020-11-16:
@freddair that's a neat hack. i'd like to have something more sophisticated than that, though.

freddair, 2020-11-16:
of course! 👍

Przemyslaw-Wosko, 2020-11-16:
@rojer done is better than perfect :D wouldn't that be sufficient as proof of concept for dimmers? it could be upgraged in future

andyblac, 2020-11-16:
> @rojer done is better than perfect :D wouldn't that be sufficient as proof of concept for dimmers? it could be upgraged in future

imo, HAA is not the way to go, it far to hacky for my liking, i would rather it done correctly in the place.

freddair, 2020-11-16:
> > @rojer done is better than perfect :D wouldn't that be sufficient as proof of concept for dimmers? it could be upgraged in future
> 
> imo, HAA is not the way to go, it far to hacky for my liking, i would rather it done correctly in the place.

Yeah I tried HAA on my Shelly Dimmer 1 before and it was less than perfect - from time to time I had issues with on/off blinking lights and input confusions / crashes... but I thought that UART commands to the dimmer chip would likely not have been the cause.

andyblac, 2020-11-16:
i did find a github thread that gave @rojer the info he needed, we just need to be patient.

https://github.com/arendst/Tasmota/issues/6914#issuecomment-562109267

rojer, 2020-11-16:
i don't intend to polish every last bit for the first iteration, but having just a couple fixed commands to bang out is a bit too low a bar :)
since no actual progress is expected until i get my laptop back and a couple test devices,  let me dump my thoughts so far instead :)

so, the way dimmer works is quite complicated. it has a slave mcu that does the PWM and communication is over UART. the same UART that is normally used for flashing and debug, so that's complication number 1.
second, the protocol is not documented. there are multiple commands.
third, the slave MCU formware is tightly coupled to the stock ESP firmware. STM firmware is shipped with the ESP firmware and is stored on the filesystem. ESP will flash it first thing after boot if it detects version mismatch.
finally, there is the calibration procedure - calibration data is stored on the filesystem, and is sent to the STM on boot.

so, there's quite a lot of ground to cover here to get to feature parity with the stock firmware, but we can do it in stages.

stage 0, early testing:
 * implement uart protocol
 * only support setting brightness, nothing else
 * ignore calibration data
 * ignore STM firmware

this is essentially what HAA bunch of strings does, but a little more sophisticated - we'll actually use proper sequence numbers and check responses.

stage 1:
 * support uploading existing calibration data, when calibration was preformed on stock firmware.
i think this could be first release. we'll ask user to calibrate usinng stock fw, then flash. if they need to recalibrate, they'll have to revert to stock, calibrate, then flash homekit fw again.

stage 2:
 * add calibration procedure to homekit fw - perform calibration sequence, record data

stage 3:
 * add STM firmware update, so we always know what version STM is running

that's the path i see. of course, i stumbled straight out of the gate - i didn't even get as far as running my code on the ESP before everything went up in smoke :)

freddair, 2020-11-16:
@rojer great path! I would love to help in beta testing stage 0.

And I would also love to start with a simple switch implementation (light on/100% and off/0%), as the dimmer is already installed behind the wall switch and I fiddle with it since ages (HAA, home bridge-shelly didn't work either)... 

I am currently on stock fw again and really can't wait to have a working basic HomeKit implementation.

Let me know how I can send you one of my spare Dimmer1 :)

rojer, 2020-11-16:
@freddair i'm focusing on dimmer2 at first, as that is the current product for sale. that said, extending support to dimmer1 is also planned, so if you can send me a device to test with, it'd be great. sned me an email to rojer@rojer.me and i'll give you the address.

freddair, 2020-11-16:
@rojer done. I hoped that Dimmer 1 and 2 would be the same fw wise but yeah... i can('t) wait a few more weeks ;-)

andyblac, 2020-11-16:
> @rojer done. I hoped that Dimmer 1 and 2 would be the same fw wise but yeah... i can('t) wait a few more weeks ;-)

could always setup homebridge, that what i doing atm, Dimmer 1/2 work ok, just dont even bother trying RGBW2 it just crashes the device.

freddair, 2020-11-16:
> could always setup homebridge, that what i doing atm, Dimmer 1/2 work ok, just dont even bother trying RGBW2 it just crashes the device.

unfortunately home bridge-shelly did not work for me as for some reason it did not discover the dimmer1 (it's the only stock shelly I use)... https://github.com/alexryd/homebridge-shelly/issues/214

bdshet, 2020-11-17:
> could always setup homebridge, that what i doing atm, Dimmer 1/2 work ok
@andyblac read from top to bottom — love to see the work you all are doing for native Shelly Dimmer HK support. In the meantime, what plugin are you using on HomeBridge?

andyblac, 2020-11-18:
> > could always setup homebridge, that what i doing atm, Dimmer 1/2 work ok
> 
> @andyblac read from top to bottom — love to see the work you all are doing for native Shelly Dimmer HK support. In the meantime, what plugin are you using on HomeBridge?

https://github.com/alexryd/homebridge-shelly#readme

andyblac, 2020-11-18:
> > could always setup homebridge, that what i doing atm, Dimmer 1/2 work ok, just dont even bother trying RGBW2 it just crashes the device.
> 
> @andyblac unfortunately home bridge-shelly did not work for me as for some reason it did not discover the dimmer1 (it's the only stock shelly I use)... [alexryd/homebridge-shelly#214](https://github.com/alexryd/homebridge-shelly/issues/214)

the new firmware 1.8.x is very picky on networks due CoAP, you could try reverting to version 1.7.0.

bdshet, 2020-11-18:
@andyblac that's great advice. Thank you!

Przemyslaw-Wosko, 2020-11-18:
@andyblac @rojer i'm waiting for this feature, since i have wall to finish on my flat :D 

Guys, what are changes, that this firmware will not require physical access to dimmer after this feature will be finished?

andyblac, 2020-11-19:
> @andyblac @rojer i'm waiting for this feature, since i have wall to finish on my flat :D
> 
> Guys, what are changes, that this firmware will not require physical access to dimmer after this feature will be finished?

it shouldn't, but i would not try any beta versions, if any are released, wait till it's complete.

rojer, 2020-11-19:
right, flashing will be done OTA, as with other devices, and won't require physical access. but as Andy said, staying away from early versions would be prudent.

[60 more replies]