mongoose-os-apps/shelly-homekit #16: possible web interface bug on shelly 2.5

RRostron, 2020-02-17:
Good Afternoon

i setup a shelly 2.5 yesterday, everything seems to work as it should in homekit etc, but
in the web interface if i turn on SW1 it actually switches on SW2 and vice versa

is this a bug in the web interface or with my wiring?

thanks

RRostron, 2020-02-17:
After some more testing the issue appears in HomeKit too it was because I had changed the names of the inputs I thought it was working.

This is what is happening, 
I have wired light 1 to SW1 and light 2 to SW2, 
The switch for SW1 turns on SW2
The switch for SW2 turns on SW1

Strange as my wiring looks ok to me, I have wired in shelly1 and shelly1pm with no probs.

andyblac, 2020-02-17:
sounds like GPIO config issue for SW1 IN and SW2 IN, they could need swapping.

rojer, 2020-02-17:
um... that would be kinda embarrassing :)
i'm pretty sure it matches _my_ shelly.
i'm traveling now, so can't verify. can someone else confirm this?

andyblac, 2020-02-17:
would it IN need swiping or OUT ?

in HOMEKIT SW2 does indeed. turn on O1

rojer, 2020-02-17:
ok, i guess i was naive assuming that since GPIO 5 on 2.5 is an input too, it would be input for the same switch as on the 1 :)

closed 2020-02-17 by rojer

RRostron, 2020-02-17:
Thanks rojer the fix fixed it :)