mongoose-os-apps/shelly-homekit #18: Shelly 2.5 room assignment

RRostron, 2020-02-18:
Good evening :)

Now the Shelly is working fine, when you add it to HomeKit it shows both switches under 1 tile, there is an option to separate them into 2 tiles, when I have done this I am unable to assign the switches to different rooms

For eg, 
SW1 is the loft light
SW2 is the landing light

In HomeKit you can only assign 1 room to both switches I can’t assign SW1 to the loft and SW2 to the Landing

If I change the room on 1 of them it changes both of them to that room.

Is there anyway to be able to assign the switches to different rooms?

Thanks

andyblac, 2020-02-18:
I would say, only if the code was changed so each switch was a separate device, the issue you have spoken about is the same for a power extension cable with 3 sockets in it, As HomeKit see it as a single device with 3 power outlets you can not have each outlet in a operate room, which kind of. makes sense on a device level (you can not physically split the device), but I do see what your requirement is.

rojer, 2020-02-18:
yeah, it is doable. like Andy said, if we export two separate accessories.

RRostron, 2020-02-25:
hi rojer,  any joy with the above request?  :D

bulyshko, 2020-04-04:
so are there any plans to add a checkbox to export multiple independent accessories?

rojer, 2020-04-05:
it's a possibility but is by no means a priority.
anyone who needs this - feel free to take a stab.
for models that have more than one switch, add a checkbox to the hap settings section "[ ] Separate accessories", unchecked by default.

rojer, 2020-04-05:
there is additional consideration here: from the cursory look, it seems that creating additonal accessory will require another instance of HAPAccessoryServer, which comes with its own statically allocated buffers.
this may stretch our memory budget beyond capacity, so this change may be predicated on making buffers dynamic.

andyblac, 2020-04-14:
@rojer how about just splitting it in to 2. and if people want it as one, they can just group them. I have a shelly 2.5 I got from Voltio and it is 2 perpetrate switches.

rojer, 2020-04-14:
no objection there. main problem right now is that we cannot support second accessory due to resource usage by the library.
it's all doable, just needs time.

andyblac, 2020-04-14:
we all have that atm, we're all in fracking lock down 😂 , wish I knew C, I would love to help.

rojer, 2020-04-14:
@andyblac if you know JS and have some time, maybe you can take a look at #50.

andyblac, 2020-04-14:
no, but, I'll try to learn some JS, I am python guy, it can't be that different can it ?

rojer, 2020-04-14:
oh, yes it can... :)
anyway, let's stop spamming this issue.

joe210599, 2020-08-31:
Where should I put this: "[ ] Separate accessories". To separate the devices into two?

rojer, 2020-09-18:
please test https://rojer.me/files/shelly/2.1.5/
see https://github.com/mongoose-os-apps/shelly-homekit/issues/9#issuecomment-694418580 for additional context

rojer, 2020-09-18:
released in [2.1.7](https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.1.7)
due to compatibility mode being enabled by default, you may need to remove the device first and add it back - this will disable compatibility mode and reconfigure device as a bridge with independent accessories that can then be assigned to different rooms.

closed 2020-09-18 by rojer