mongoose-os-apps/shelly-homekit #20: Air Conditioner Support needed

mandarkyiz, 2020-03-02:
Is it possible to configure an air conditioner support for a Shelly 2.5 so HomeKit will see it as an air conditioner?

It doesn’t have to control the temperature. An On/Off feature would be enough.

Thank you so much in advance.

mandarkyiz, 2020-03-02:
According to Shelly’s website “Shelly 2.5 can control the heating and air conditioning, and help you decrease the energy consumption up to 30%. electrical consumption. Appliances Control all household appliances in your home”
But is it possible through HomeKit?

kzyapkov, 2020-03-02:
Shelly 2.5 can control two AC appliances with its relays, be it a heater, an AC unit or something else.

Using this project as a starting point, you can implement any Homekit-supported service in firmware, so the device shows up as the appropriate type in the Home app. Ultimately, however, the Shelly2.5 hardware can only turn on or off two electrical appliances.

mandarkyiz, 2020-03-02:
Hey man thank you for your respond.
Since its only a 1 air conditioner thermostat I'm going to connect the shelly is it a better idea to use a shelly 1? and will it be suitable to control the temprature or juts On/Off?
Furthermore, how can I implement/find such firmware? 
Thanks !

rojer, 2020-03-06:
@mandarkyiz 
Thermostat HAP service must implement temperature control, so we cannot export a thermostat service unless we can report current temperature and specify target temperature.
so i'm afraid no, it's not possible.

closed 2020-03-06 by rojer

andyblac, 2020-03-06:
might be possible with Shelly1 and Temperature add-on, but support would have to be coded of course.

https://shop.shelly.cloud/temperature-sensor-addon-for-shelly-1-1pm-wifi-smart-home-automation#312