mongoose-os-apps/shelly-homekit #22: Default ON, OFF or last state at device start

one4many, 2020-03-05:
Hi there, thank for for the homekit support. I would like to have a configurable default state when the Shelly powers up, preferably configurable through the web interface (not sure if something is possible at all in homekit). It would be great if I can have the Shelly either ON, OFF or at the last state before it rebooted (lost power) on a new start up. Thank you!

andyblac, 2020-03-05:
kind of already is "persistent sate" that will return the device to previous state after power cycle.

andyblac, 2020-03-07:
this can be closed now ?

one4many, 2020-03-07:
The persistant state is nice, but I have a pump for example, which after a powerloss should be ON for a few minutes, regardless if it was ON or OFF before.

rojer, 2020-03-08:
@one4many that may be beyond what a simple switch like shelly can be expected to do. does the stock firmware support this use case?

mjkout, 2020-04-05:
can it be set to On on power back? ( similar to the stock firmware )

andyblac, 2020-04-05:
what ever the state was before power cycle it will return to that state.

mjkout, 2020-04-05:
I know but is there a way it can be set to On  only?

rojer, 2020-04-05:
it's a good idea to implement it. convert current "Persist state" checkbox into a select "State on power-up", with "on", "off", "previous" and "input".

mjkout, 2020-04-06:
This would help me a lot in my setup I'm not using the sw wire ( have a 4 switch setup will endup laying alot of wires to the lights ),  if the state was On, i will not have to lay another wire for the switch port in case someone switches it off and then on again

one4many, 2020-04-06:
+1 
This is what I am looking for 'on' / 'off' / 'previous' / 'input'

andyblac, 2020-04-08:
@rojer could 'persist_state' be changed to a menu option like 'in_mode' 

0 = always off after power cycle
1 = always on after power cycle
2 = remember previous state

this coupled with the pull request for 'auto power off' could do what people are seeking ?

rojer, 2020-04-18:
@andyblac this is exactly what i proposed, plus the "input" choice, which would take input value.

rojer, 2020-04-18:
released in 1.8.0

closed 2020-04-18 by rojer