mongoose-os-apps/shelly-homekit #24: MQTT function

italoc, 2020-03-05:
with this firmware is possible to mantain MQTT function?

andyblac, 2020-03-05:
I believe not, it is homekit only, it more secure that way.

sja, 2020-03-08:
I would also like to still have MQTT support. Think about using switch functionality with HomeKit UI and reading measurements with MQTT in some sort of persistence or Home Automation solution.

rojer, 2020-03-09:
makes sense. it would be possible, though we are currently very tight on space, we'd need to solve that first.

pnuding2, 2022-02-09:
I'd love this, too. Especially useful with power measurement, where Homekit isn't very helpful

timoschilling, 2022-02-09:
@pnuding2 we already expose the power measurement to HomeKit via the Eve specification, try the Eve app or Home+ App to view them or creating related automatisations.

opctim, 2022-04-26:
@rojer How is it going with this? I would really like to see an mqtt feature on shelly-homekit! Would also be nice to have a click counter on the devices so we can see in mqtt how many times the switch was pressed in a given timespan (milliseconds) to switch other devices on "doubleclick"

rojer, 2022-05-01:
no update on MQTT

kaese2002, 2023-07-16:
still request that as well...
these are the only devices i cannot get into iobroker...