mongoose-os-apps/shelly-homekit #26: Shelly Plug-s

Domo-Dana, 2020-03-09:
Hello, is a development planned to have the consumption data read by the Plug-S as on the Shelly Cloud app?

Thank you.

rojer, 2020-03-10:
the question is how to expose this info in the Home app?

Domo-Dana, 2020-03-12:
> the question is how to expose this info in the Home app?

Exactly, I hadn't thought of this point. :-/

Domo-Dana, 2020-03-12:
The Shelly Plug S socket is well recognized and controllable with Homekit, however the automation mode does not seem to act on it. Did you have the same problem ?

andyblac, 2020-03-12:
only 3rd party apps like EVE can show this data.

cschildhorn, 2020-05-21:
Perhaps this helps. 

https://gist.github.com/gomfunkel/b1a046d729757120907c

Rappelsack, 2021-01-30:
I wrote this feature request: https://github.com/mongoose-os-apps/shelly-homekit/issues/452

I do not understand why this should be placed in this post, but again:
This feature would be great.

andyblac, 2021-01-30:
> I wrote this feature request: #452
> 
> I do not understand why this should be placed in this post, but again:
> This feature would be great.

it about getting power consumtion data in to HomeKit, so you can create an automation to do what you describe.

Rappelsack, 2021-01-30:
No. I do not request power consumption in homekit.

Just a setting simmilar to the: switch off after xx minutes:
[image]

Like this.

andyblac, 2021-01-30:
I'll reopen, post this info in the issue, i'll let rojer decide if it something he wants to code.

timoschilling, 2021-01-30:
That is only supported as a characteristic of a valve. I will work on that after #433 is merged.

Fireball-AT, 2021-02-22:
My use case would be to trigger an action if e.g. the washing machine is finished. I would refer to the power consumption in stand-by over a certain timeframe.

closed 2021-06-11 by rojer

Rappelsack, 2021-06-11:
Why this is closed? :-(

Still waiting for this to use the shelly plug s to reduce the standby consumption of my TVs, etc.

andyblac, 2021-06-11:
> Why this is closed? :-(
> 
> Still waiting for this to use the shelly plug s to reduce the standby consumption of my TVs, etc.

will be in next build via https://github.com/mongoose-os-apps/shelly-homekit/pull/690

PR #690 (2021-06-11): support for eve power chars
closes #26

This characteristics aren't supported by the Home App, they can be view with the Eve or Home+ App.

This PR explicitly don't contains support for the eve power history char. That is out of scope for this PR, because it requires general change like storing the power consumption first.