mongoose-os-apps/shelly-homekit #27: Shelly 1PM power consumption

ellavas, 2020-03-11:
Hi,

I have installed the shelly homekit firmware on a Shelly 1PM. 
It works well but I can't find power consumption.

Would it be possible to add this feature ?

Thanks a lot.

ellavas, 2020-04-15:
related to https://github.com/cesanta/mongoose-os/issues/531#issuecomment-605852246

ellavas, 2020-04-28:
I’ve ported the Arduino HLW8012 power meter sensor library.

It's working fine with the Shelly 1PM.
You can have a look here:

https://github.com/ellavas/mongoose-lib-hlw8012

edgarclerigo, 2020-09-28:
Hi guys,

Whats the current status of this ? Anyone with this firmware is not able to see any data about the power consumption, even on the Eve App ?

rojer, 2020-09-28:
current state: not supported, no ETA

Dag0d, 2020-11-10:
Is this something you've been working on? Would love to see that feature.
For the other power measurement Shelly devices which are already supported, do you expose the power consumption to the eve app?

rojer, 2020-11-10:
power measurement is only supported on shelly 2.5 at the moment and is only surfaced on the web UI because there is no service or characteristic defined in the HAP protocol for such measurements.

ellavas, 2020-11-11:
@Dag0d you can have a look to the library I have ported. It works fine on shelly 1PM.

https://github.com/ellavas/mongoose-lib-hlw8012

rojer, 2020-11-11:
thanks @ellavas this is handy! i'll take a look

freddair, 2020-11-29:
@rojer why not exposing power consumption to the eve app which would allow homekit automations like detecting when a washing machine / dish washer / water boiler is ready... would be really handy

rojer, 2020-12-05:
Added support for power management for 1PM, Plug and PlugS.
Please help test by installing http://rojer.me/files/shelly/2.7.0-alpha4/ - you should see power figures in the web UI.
I am especially interested in accuracy, as I only have one of each of the device and cannot tell if coefficients hold across different samples. In theory, 1PM and PlugS should be the same, but mine differ by about 10%

andyblac, 2020-12-05:
> Added support for power management for 1PM, Plug and PlugS.
> Please help test by installing http://rojer.me/files/shelly/2.7.0-alpha4/ - you should see power figures in the web UI.
> I am especially [...]

whats the minimum power draw required, i have mine just turning the indicator light on a fused spur, but it does not detect any power consumption.

[image]

[image]

rojer, 2020-12-05:
that's definitely under 1 watt, you won't see anything.
i'd say any kind of accurate reading is from 10W.

Dag0d, 2020-12-06:
did you make the power measurement values already persistent to create charts and so on?

rojer, 2020-12-06:
no

Dag0d, 2020-12-06:
okay thanks, I wondered why this wasn't working, I will continue testing this on my 1PM, till now it seems to run great. this evening I will have time to double check the power consumption measurements.
if it is off, should we think about a calibrating process?

rojer, 2020-12-06:
correct

Dag0d, 2020-12-07:
I´ve tested it a bit further, this firmware shows approx 1 watt more than the original firmware, if I double check with a multimeter, the multimeter value is exactly between this value and the value of the original firmware. I would be fine with that

rojer, 2020-12-07:
thank you! indeed, that is acceptable. probably just a matter of rounding.

rojer, 2021-01-05:
this is done

closed 2021-01-05 by rojer

masterc09, 2021-04-08:
I've got updated Shelly PlugS with latest shelly-homekit firmware to trigger a scene depending on power consumption like mentioned above from @freddair 

> @rojer why not exposing power consumption to the eve app which would allow homekit automations like detecting when a washing machine / dish washer / water boiler is ready... would be really handy

Can somebody confirm this functionality? I tried to access the value for power consumption via controller app with no luck, even not in eve app. If this should work could somebody give me a short instruction please? Thanks.

timoschilling, 2021-04-08:
@masterc09 this issue is only about the measurement, that providing to HomeKit isn’t part of this issue. There is a other issue about that.

masterc09, 2021-04-08:
@timoschilling Thanks, now I got it.

ingenierodevops, 2023-01-18:
Do you hace an y idea to make a workaround to this? 
maybe launch a web call  or similar to HomeKit from the device configured in the web ui

I saw in homebridge someone used light sensor element to import power measurement. Can this be a good idea?