mongoose-os-apps/shelly-homekit #28: Shelly2 does not accept new fw via OTA

sja, 2020-03-16:
I have a Shelly2 and some Shelly2.5.
For 2.5 everything works fine doing a curl to the OTA link.

For Shelly 2 it does not work:

    ❯ curl 'http://10.0.20.40/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly2.zip'
    {"status":"updating","has_update":false,"new_version":"20200309-104038/v1.6.0@43056d58","old_version":"20200309-104038/v1.6.0@43056d58"}%

Are there any known issues for 2 and this method?

consoleman, 2020-03-17:
I have 2 Shelly 2 and no problems with them. Firmware flashed and in 1 Minute they worked with the new firmware

andyblac, 2020-04-15:
I having same issue with new shelly1, any ideas?

andyblac, 2020-04-15:
had to flash via serial. strange.

andyblac, 2020-05-12:
ok fixed issue, check to make sure you shelly's have a working internet connection, my Shelly was not getting a working DNS.

andyblac, 2020-09-15:
@sja is this still an issue ?

andyblac, 2020-09-19:
@rojer original OP has not responded, think we can close this.

rojer, 2020-09-19:
@consoleman please reopen if it's still an issue

closed 2020-09-19 by rojer

TonyJacobi, 2023-07-02:
Hello, I have the same issue with a Shelly 1:

{"status":"updating","has_update":false,"new_version":"","old_version":"20191217-140757/1047-long-id-for-shelly-devices@a0661583"}

[image]

timoschilling, 2023-07-03:
@TonyJacobi your Shelly is on a over 3 years old version. Please update the stock firmware first

TonyJacobi, 2023-07-03:
Hello Timo, thanks for your response.The Shelly 1 mentioned that there was no firmware update in the website of the switch (192.168.33.1). I read on the internet that sometimes Shelly can’t find a firmware update OTA. So, I added the Shelly in the Shelly app. The Shelly app said indeed there was a firmware update 2023…. . I could install the firmware update and then I did the update to HomeKit. So, the Shelly 1 is now compatible with HomeKit.The reason I bought a new Shelly 1 switch is that I believe that another Shelly 1 switch which I use as a garage door opener, has failed. That one switches WiFi on and off, all the time. I already went back to stock software and added the HomeKit software, but still it loses contact with WiFi. I did not connect the new Shelly 1, so I can’t say if that solves the problem, or that there is another issue.Greetings and thank you very much for all the great work you do for the HomeKit community, Tony