mongoose-os-apps/shelly-homekit #30: Shelly 1 disconnecting

CHCHcc, 2020-03-23:
After successfully pairing shelly 1 with HomeKit, the device disconnects from WiFi and won't reappear. Only cutting power helps to make a new connection. After some minutes up to a couple of hours it disconnects again.

andyblac, 2020-03-23:
I would say check your wifi strength in that room.

CHCHcc, 2020-03-23:
I have two of them running (rather sleeping). One in the room with the WiFi router and one in the next room. So strength should not be a problem. Since it drops the WiFi connection I can’t go to their IP addresses either.

tsdrummer, 2020-03-25:
Do you have running 2,4 and 5 GHZ with the same name / SSID?

CHCHcc, 2020-03-27:
Yes, 2,5 and 4 GHz on the same name. But all other devices in my network work flawlessly.

thp1105, 2020-03-30:
I have exactly the same issue here. Two shelly 1 connected to WIFI. Working flawlessly with the stock firmware, but when flashing the homekit firmware both start disconnecting at random, only eventually reconnecting. Access to the devices ist possible through browser, even if homekit sees them offline.

tsdrummer, 2020-03-30:
Does switching off 5GHz signal solve issue? (Even OEM Firmware works with both)

kzyapkov, 2020-03-30:
Activity on the 5GHz band would be irrelevant here. ESP8266 doesn't have a 5.8G transciever.

thp1105, 2020-03-30:
I just monitored both Shellys, they in fact stay on 2.4 GHz all the time. Right now one is reactive on homekit, the other is not. Both can be contacted in a browser through their address and acting normally.

CHCHcc, 2020-03-30:
I have no connection to the shellys at all. They simply drop out of the network and cannot be reached by browser.

rojer, 2020-03-31:
somehow we will need to retrieve console logs from one of the devices experiencing this.
can anyone with this problem connect serial console to one of the switches?

CHCHcc, 2020-03-31:
If you give me a detailed guide how to do it, I can try. I just received a USB to TTL adapter today.

rojer, 2020-03-31:
http://www.thesmarthomehookup.com/wp-content/uploads/2018/08/diagram.jpg

Do not connect GPIO0 to ground though.
After that, use any terminal program to connect at 115200 baud, 8-N-1.
Log output from the device to file and capture the moment when it becomes unreachable.

CHCHcc, 2020-04-01:
I solved the problem by flashing HAA onto my shellys. wasn't as convenient to install as this firmware but it works flawlessly in my network environment.

rojer, 2020-04-03:
ok, it may no longer be affecting you but the issue is still there.
can somebody else who is affected please help by providing console logs?

tsdrummer, 2020-04-03:
I have no issues, but router is only 8 feet away.

andyblac, 2020-04-03:
I have the issue, but it imo it related to WiFi strength, I have my shelly 1 in a Shelly Button, if I put in on the floor next to lamp, it drops off AP, but if I have in on table next to lamp it seems fine.

Supereg, 2020-04-03:
Got said issues. Definitely not a connectivity router caused by my network.

This is a serial capture AFTER it was already disconnected and continuously tries to reconnect.
I try to capture know when it actually encounters the disconnect.

```
mgos_wifi.c:345         WiFi STA: Connect timeout
mgos_wifi.c:88          WiFi STA: Using config 0 (Tasty Door)
mgos_wifi.c:270         WiFi STA: Connecting to Tasty Door
mgos_net.c:86           WiFi STA: connecting
shelly_main.c:254       Uptime: 2011.61, RAM: 32936, 23692 free
shelly_main.c:254       Uptime: 2012.61, RAM: 32936, 23692 free
shelly_main.c:254       Uptime: 2013.61, RAM: 32936, 23692 free
scandone
mgos_wifi.c:119         WiFi STA: Disconnected, reason: 201
mgos_net.c:82           WiFi STA: disconnected
```

Supereg, 2020-04-03:
Additionally I get some errors like the following: 
```
HAPPlatformLog.c:65     IPAccessoryServer Failed to allocate session. (Number of supported accessory server sessions should be consistent with the maximum number of concurrent streams supported by TCP stream manager.)
```
Seems not related and don't know if that error is somehow a problem

Supereg, 2020-04-06:
Did the serial output help?

Okay some theory. Is there the possibility, that this is some issue which is only present when booting up the **first time** after a firmware upgrade (coming from stock firmware)? 😅
Couldn’t reproduce it yet. Shelly1 is running rock solid (with some reduced response times sometimes) for three days straight 😅

rojer, 2020-04-07:
code 201 means it cannot find the wifi network, as in - doesn't see any APs to connect to. this is weird.
unfortunately, there isn't much i can do about it - the wifi machinery is in the libraries provided by Espressif, if it says "not found", i can't do much here...
one thing i have in mind is update to latest version of the SDK and libraries they provide, we haven't done an update for a while.

rojer, 2020-04-07:
the sessions problem suggests that you have more than 3 clients trying to conenct to the device. we currently have a limit of 3 simultaneous sessions due to shortage or RAM.

Supereg, 2020-04-07:
> the sessions problem suggests that you have more than 3 clients trying to conenct to the device. we currently have a limit of 3 simultaneous sessions due to shortage or RAM.

When looking at the `HAPIPAccessoryServer` inside `HandlePedningTCPStream` the `HAPPlatformTCPStreamManagerAcceptTCPStream` is called first. This function (in the default POSIX PAL) returns a `kHAPError_OutOfResources` if no more TCP session can be opened. And thus fail directly. The Shellies however fail some lines below when the AccessoryServer tries to allocate a new Session. Looking at your init code you correctly set both maxTCPsessions and maxSession (to 3). Haven't looked at your TCPStreamManager PAL however. But seems like there is some mismatch? @rojer

Supereg, 2020-04-07:
> code 201 means it cannot find the wifi network, as in - doesn't see any APs to connect to. this is weird.
> unfortunately, there isn't much i can do about it - the wifi machinery is in the libraries [...]

That's weird 😅 I can tell you that the network was definitely up and running 😂
But haven't experienced a dropout since I restarted after that dropout. Anyway I can help you debug this further? Or are we just trying to upgrade the wifi stack and hope it's gone?

ursusHU, 2020-04-25:
I have the same issue here, after a few days of installation of the Homekit firmware the unit just timed out, won't pair again. If I reinstall the shelly firmware works flawless. For testing I have reinstalled now the firmware 1.8 only on one Shelly-1 the same result now, and I'm not able to connect anymore.

aidanboyce, 2020-05-13:
Hi, same issue here. I have 2 Shelly-1 devices. 1 connects without issue (and is further away) the 1 I have the problem with (is really close to the wifi access point) connects to an alternative network but not the network I want it to connect to.

Here's the output from the console

[image]

jurban85, 2020-05-21:
I have the same issue. Have now installed two shelly 1 and both of them randomly disappear from my network. The strange thing is, bothe start to work when I restart my router. It seems that my router is doing some weird channel changes randomly and bother Shellys have trouble to reconnect to my network. Does anybody have the same experience?

andyblac, 2020-05-21:
Do you have auto channel setup on your router ? Where the router auto will choose the channel what it thinks it is the best unused Chanel, I’d try disabling as I doubt the Shelly will support roaming that is required on the client for the feature to work correctly.

jurban85, 2020-05-21:
> Do you have auto channel setup on your router ? Where the router auto will choose the channel what it thinks it is the best unused Chanel, I’d try disabling as I doubt the Shelly will support roaming that is required on the client for the feature to work correctly.
> 
> It sounds like the router is changing Chanel and Shelly is stuck on old channel.

That’s what I thought as well. I gave my network a fix channel (recommended was 11) and after 30 minutes, one shelly disconnected the other one was still in the network. 🙁 the strange thing is that one might work flawlessly while the other one is offline. And it’s not always the same shelly. Both have the same issue.

jurban85, 2020-05-21:
One additional thought, could this issue be caused by my mesh repeaters? I was checking the connection and it seems that they are not trying to access one of those repeaters because they are closer to the router itself. It doesn’t seem that they have trouble with changing access points. Does anybody have similar experiences

ursusHU, 2020-05-22:
I think it is somehow related to the network we use(I have a simple router from my ISP, next week I will build my mesh, so I will see if anything changes), but definitely this firmware handles it differently than the original Shellies. With them I have no issues at all independently how I set my network. Currently I have 20 shellies around and 1 has bad coverage, time to time it gets unreachable in HomeKit, but always manages to get back without further actions.

andyblac, 2020-05-22:
@jurban85 can you lock device to a particular AP, in Unifi you control minimum RSI of a client that is request for a device to use the AP help stops a client jumping around AP's

jurban85, 2020-05-22:
> @jurban85 can you lock device to a particular AP, in Unifi you control minimum RSI of a client that is request for a device to use the AP help stops a client jumping around AP's

That's the thing: I can't. My router and its mesh repeater doesn't have this option. I will try to disable the mesh network for a few days, maybe this will help.

jurban85, 2020-05-22:
Newsflash. Right now, one of two shellys was offline again. Instead of restarting my speed port pro router, I have switched the WiFi channel from 9 to 1. After I have done this, shelly was online again! This kinda ruins my mesh ap theory but my issue seems to be related to different WiFi channels. 

I have turned off automatic channel selection but my problem still happens quite often. 

Does anybody have an idea?

rojer, 2020-05-22:
@jurban85 not yet. i am listening to all the conversation here, keep the information coming.

jorgerray, 2020-06-04:
I have shellys 1 and 2.5 (no blind) with your firmware for homekit that disconnect from my wifi network after restarting the internet network. The only way for them to work again is to disconnect them from the electrical network and reconnect them.
Would it be possible to add a scheduled daily auto reset button at a specific time? Thnaks

ursusHU, 2020-06-04:
I have installed the mesh network, one unit has the HK firmware since yesterday. Fingers crossed as currently the new Shelly Homebridge plugin gives only headaches ;)

felixmueller, 2020-06-11:
@rojer Since I also have issues with repeatedly disconnecting Shelly 1 devices, what about an optional auto reboot feature as suggested by @jorgerray ?
The implementation could offer a time setting for automatic reboots, e.g. every 24 hours.
This could solve big issues with disconnecting and not reconnecting Shellys which I experience quite often. I have to regularly disconnect my whole place from power to solve this.
Thank you so much!

rojer, 2020-06-11:
we can add reboot if wifi connection is not re-established for some time

felixmueller, 2020-06-11:
@rojer It depends if wifi disconnect is the only reason for becoming unresponsive. And if the device is always able to detect it and handle accordingly. A timed reboot count solve all other problems, if the device has other issues.

ursusHU, 2020-06-11:
Let me share my experience after installing a tplink mesh system. All the units are working now fine since a week. I have the connection problem only with one Shelly-1 PM Which is in the basement (with original firmware no problems were occurring). The funny thing that after firmware change the Shelly is disappearing from the tplink app. It is visible with the original firmware.

felixmueller, 2020-06-14:
@rojer After experiencing some more permanent wifi disconnect issues over the weekend, I came to the conclusion that an optional auto reboot feature in case of the wifi not reconnecting (as suggested by you) would be very very helpful in my case!
Is there any chance of implementing it in the near future? Thank you so much for your work and the firmware!

andyblac, 2020-06-14:
I had a strange thing happen yesterday, I have 9 Shelly devices a mix of 1's / 2.5's on this firmware and Dimmers / RGBW2 on official via HomeBridge. anyway, yesterday I had to cut power as I am getting work done in the house, when I powered back up, half of the shelly's would not show in homekit nor could I see the WEBUI, but they all had IP address shown on router, I cut the power again hoping it would force them to reboot up properly, but now a different lot would not come online. The devices that had official firmware on then had no issue s both times, they booted up five and both WEBUI and in HomeKit via homebridge.

I tried doing a force 'reconnect' in Unifi software manger on each client, and eventually they all came back online, it almost like HomeKit whilst it is doing it connection thing, is locking up the device, as I have also noticed if you hammer HomeKit, by opening / closing / switching rooms etc, that ask for the device to update it status, it causes the device to top off too.

too me it almost if the device can't handle multiple requests to and from homekit in quick succession, and when you have a load of devices all asking to sync with HomeKit, if it does not respond in time the device gets hung up.

so @rojer is there something you can put in the code that if the device has had no response from HomeKit after a request was sent, try to reconnect the Wifi or reboot the device.

jurban85, 2020-06-16:
Guys I wanted to leave some feedback. I have had the same issue, my shellys kept disconnecting after a few hours, the only thing helped was rebooting my router (speedport pro). Meanwhile I have up and went back to original FW. I have also installed the homebridge plugin but it didn’t go well. I decided to rename my 5ghz network. Since the HB plugin did also not work, I gave this custom fw a chance and for whatever reasons my shellys are online for 2 days now. I don’t know if rojer did some magic to the latest release or changing ssids names was the key but it works for me now! Thanks for your input guys I am a happy man now!

andyblac, 2020-06-19:
one of my shelly2.5 has dropped off this morning no ip or listed in router, was thinking... we have a reset action of 10 x on/off, could we have a reboot action with 5 x long on/off (1 sec hold each press)

andyblac, 2020-06-19:
seem Official Firmware is having same issues ????

```
Device WI-Fi disconnect - BETA program version 2. 
We make some improvement. 
Please do not use this thread for another issues or anything. I need short and clean answers: Work or NOT
Because tehre is some reports for occasional disconnects from Wi-Fi network, we make firmwares for Shelly1 and Shelly2.5 with workaround fix based on latest firmware version. This is happens for less than 1% of the costumers and it's related to specific router settings or network infrastructure or another Wi-Fi devices connected to the same network. 
Simptoms: Devices lost Wi-Fi connection and need power cycle to connect again. If your devices do not has this simptoms you not need to install the fix. 

If you have such a issue, you can test this fix.
How to install it: 
You need to know device IP. Then execute this command from any browser: 
http://deviceIP/ota?url=URL
where URL is: 

Shelly2.5: 
http://repo.shelly.cloud/firmware/dimo/SHSW-25.zip

You can revert devices any time to standart version. If BETA fix is sucessfull we will implement it in next frimware version for all deveces. 

Please report in this thread. 
```

felixmueller, 2020-06-20:
Just a quick update of my experience with disconnecting Shellys: I have a Mesh WiFi in my home using a repeater in the upper floor. The disconnecting Shellys are more often the ones closer to the Mesh repeater than to the Mesh router. Maybe it is a Mesh issue? I have zero problems with other devices.
I still hope for an auto reboot feature after a specific time or event to solve there issues. Thank you so much @rojer !

andyblac, 2020-06-22:
here the link to the FaceBook support group discussing this issue on Shelly 1/2.5
https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/2875875842511761/

felixmueller, 2020-06-24:
Another update from my side: I have now reverted my most problematic Shellys to the stock firmware. Since then, I have had no more wifi disconnects. I now use the shelly-homebridge plugin. Automations that trigger on switching operations now work reliably, with the HomeKit firmware they unfortunately only did it in half the time.

75TiagoMatias, 2020-06-24:
Hi,

Same issues on a Plug-S. From time to time during the day it disconnects and is only available about 5/10min after disconnection.

Already changed Automatic channels to 11 Fixed Channel on 2.4Ghz network... No luck...

When I go to the IP back-end of the device, when it's not available, I can't connect. But when it reconnects, the uptime is running 2/3 days ago... I mean... When disconnects it does not turns off or reboots, only doesn't have access to the network...

ursusHU, 2020-06-24:
> Another update from my side: I have now reverted my most problematic Shellys to the stock firmware. Since then, I have had no more wifi disconnects. I now use the shelly-homebridge plugin. [...]

I had the same issue, since I updated the wifi network to a mesh network (tplink deco m5), everything works smoothly with the HomeKit firmware as well.

rojer, 2020-06-25:
again, apologies for lack of concrete action on this but i am reading all the comments.
i think there are two issues at play here:
1) disconnections due to some sort of bug/issue
2) disconnections due to expensive crypoto + aggressive APs
we definitely need to fix (1), which i think is the most prevalent issue people experience, fixing (2) is more complicated due to non-preemptive sdk we use.
my next step is to try updating wifi libraries to address (1). i'll get on it soon.

juankibcn, 2020-06-26:
Same issue here. I’ll wait for your update.

tdXFS, 2020-06-26:
I have the same issue, the only way to restore the connection is to reboot the shelly or restart the access point

meraki-master, 2020-06-26:
Same issue here. I see a lot of failed authentications in the log when this happens. Rebooting the access point helps from time to time. Power cycling the shelly helps every time.

rojer, 2020-06-28:
ok, i updated SDK and toolchain and built Shelly 1 and Shelly 2.5 firmware. please test attached builds and let me know.
[shelly-homekit-Shelly25.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/4842015/shelly-homekit-Shelly25.zip)
[shelly-homekit-Shelly1.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/4842016/shelly-homekit-Shelly1.zip)

andyblac, 2020-06-28:
loaded it on 2 of my shelly 1's see how it goes, if no issues I'll do the rest of my Shelly's :)

juankibcn, 2020-06-28:
I've loaded it on 1 shelly 1 and 1 shelly 2.5
Now i have:
- 1 x Shelly 1 with 2.0.1
- 1 x Shelly 1 with 2.0.2
- 1 x Shelly 2.5 with 2.0.1
- 1 x Shelly 2.5 with 2.0.2
- 1 x Shelly 2.5 with original firmware

Starting the countdown....

andyblac, 2020-06-29:
here is official response from Shelly for Shelly 1 / 2.5 they had disconnecting issues in 1.7.x firmware, they fixed it in 1.7.4 here is what they did, @rojer maybe help us ? 

```
June:25 UPDATE: Dear All - this fix looks successfull workaround for costumers which have experiance with disconnecting with their devices. Now we will implemented  this fix in all The Shelly devices and next week will release firmware 1.7.4 for all of them. About the fix: We added additional check (ARP Check). If devices lost connection for 1 min we reboot Wi-Fi stack, even that SDKs do not report us nothing. This is workaround for the problem. We working to implement newest Espressive SDK in firmware 1.8 which should fix the real problem, not just to workaround it. 
```

rojer, 2020-06-29:
yes, more or less

juankibcn, 2020-06-29:
At the moment, I’ve lost the WiFi connection of one of my Shelly with 2.0.1 firmware. 
2.0.2 online during 16 hours... 
TBC

[209 more replies]

closed 2020-08-05 by rojer