mongoose-os-apps/shelly-homekit #34: HomeKit AND Alexa

Alwiz79, 2020-03-31:
Is it somehow possible to have the Shelly 1 listed in HomeKit AND Amazon Alexa.
I'd like to crontrol with the Apple Home App, but also have the possibility to tell Alexa what to turn it on/off

GrafKanex, 2020-04-01:
I also search for the same solution for the shelly 2.5 but till now without success

rojer, 2020-04-01:
this is possible in principle, mongoose os has the [aws iot cloud library](https://github.com/mongoose-os-libs/aws) already but currently having both AWS and HomeKit enabled means firmware won't fit in the limited amount of flash we have available. we need to increase available space.

consoleman, 2020-04-01:
You have to use the Shelly skill and Homebridge for HomeKit access

andyblac, 2020-04-01:
try this https://github.com/RavenSystem/esp-homekit-devices

rojer, 2020-04-01:
if we bring in the library, you will be able to use skill with the device directly.

bulgaro80, 2020-04-05:
> https://github.com/RavenSystem/esp-homekit-devices

No, doesn't work with Alexa, just with HomeKit

bulgaro80, 2020-04-05:
> Is it somehow possible to have the Shelly 1 listed in HomeKit AND Amazon Alexa.
> I'd like to crontrol with the Apple Home App, but also have the possibility to tell Alexa what to turn it on/off

me too... I tried with Homebridge but it's too complex to manage and keep it running. I would have something already embedded natively in the firmware

Supereg, 2020-04-13:
Just checking in to leave my opinion here.
Could we please leave the HomeKit firmware as simple as possible and reduce the bloat added to it. I personally would hate the idea to open any other communication channel besides the HomeKit protocol. At least introduce a build flag which to enable or disable inclusion of code designed for such a feature.

macomic, 2020-06-19:
I think a lot of us use Alexa rather Siri. An integration would be very nice. 👍🏼

ursusHU, 2020-09-21:
Also other products (Netatmo, eufy, tado etc.) are able to communicate paralell with Homekit/Google/Alexa. This would be a nice feature. The Homebridge solution is working, but unfortunately it causes issues in case of some mesh routers (coap problem), this firmware however works nice, this could be a good improvement.

andyblac, 2020-09-21:
TBH, having Alexa and google added to this firmware would turn me off it, I would probably never update again. for me that is a big NO, NO.

rojer, 2020-09-21:
@andyblac hm, interesting. can you elaborate why?
i quite like the idea of supporting multiple ecosystems.

andyblac, 2020-09-21:
plus I like that fact the HomeKit dose not require ANY WAN access.

bulgaro80, 2020-09-21:
The issue about multiplatform compatibility with Shelly is the EPROM size, too small to install a firmware that is supported by Alexa and Home Kit

andyblac, 2020-09-21:
> The issue about multiplatform compatibility with Shelly is the EPROM size, too small to install a firmware that is supported by Alexa and Home Kit

that too, would just limit even more what we could add to HomeKit side of it, RAM is limited already.

rojer, 2020-09-21:
> The issue about multiplatform compatibility with Shelly is the EPROM size, too small to install a firmware that is supported by Alexa and Home Kit

not necessarily the case, i believe we can support both with some optimizations.

macomic, 2020-09-21:
> The issue about multiplatform compatibility with Shelly is the EPROM size, too small to install a firmware that is supported by Alexa and Home Kit

Maybe not after it was new Coded from scratch. What happened last.

rojer, 2020-09-21:
space concerns aside, what are other disadvantages?
needless to say, it would be off by default, so no wan access by default.

andyblac, 2020-09-21:
> space concerns aside, what are other disadvantages?
> needless to say, it would be off by default, so no wan access by default.

yes, but anything you add to the firmware would have to be compatible with both enabled, this would reduce available RAM to HomeKit side. and limit what we could do in the future surely ?

macomic, 2020-09-21:
If you have the decision ether on or off. Then you can't have any disadvantages.

andyblac, 2020-09-21:
> If you have the decision ether on or off. Then you can't have any disadvantages.

of course it is read above.

Supereg, 2020-09-21:
@rojer how does the Alexa sdk look like on the shelly side? Is this provided by Amazon or a custom implementation?

rojer, 2020-09-21:
hah, interesting glitch there: looks like another comment got mixed up with mine, i take it that this part

> I don't trust Google / Amazon in a any way shape or form, I try to stay clear of any IoT thats supports those.

is yours, @andyblac ?

@Supereg i've had a brief look and it looks looks like just a couple MQTT topic subscriptions. mongoose os already has extensive support of mqtt, i don't think we'll need to bring in any additional libraries.

macomic, 2020-09-21:
IMO Alexa Integration would help the firmware to reach much more households and additionally talented helpers to write code or test new firmwares in new environments

rojer, 2020-09-21:
i think having choice is good.
privacy concerns are addressed by having these features off by default - if you don't trust google or amazon, don't use them.
space impact needs to be evaluated. this firmware is and will remain primarily about homekit support, but if we can afford it, i don't see why we shouldn't add it as an option.

andyblac, 2020-09-21:
> i think having choice is good.
> privacy concerns are addressed by having these features off by default - if you don't trust google or amazon, don't use them.
> space impact needs to be evaluated. this [...]

just out of curiosity how much free RAM do we have atm for additional features. like schedules, dummy switches etc.

macomic, 2020-09-21:
BTW I read somewhere that a Part of Apple cloud services use AWS

rojer, 2020-09-21:
right now we have about 7K code space (flash) left on our biggest target (Shelly25, which includes power metering).
there are ways to reclaim code space:
1) disable HAP logging, saves 44K, easy
2) reduce fs size from 256K to 128, reclaims 128K for code, also easy but adds a constraint to stock firmware compatibility: must be upgrading from 1.8 or newer, older versions cannot reconfigure fs size in OTA updates. as times passes and 1.8 gets out into the field, this becomes less of a restriction. plus, it's always possible to update stock fw before migrating.

i did a quick experimant and adding full TLS support (required for AWS connections) requires 107K flash space, which leaves 65K to spare with (1) and (2) both applied or about 21K with just (2). this is quite comfortable.

in terms of RAM, with HAP ADK optimizations applied we are quite comfortable now with about 25K free. this goes down during OTA updates and TLS handshake will require additional resources, so we may need to implement a solution where HAP connection is shut down for the duration of the update to free resources. this is not very difficult to do.

all in all, i don't see major resource constraints to implementing AWS support.

ttimpe, 2020-10-15:
This woule be very useful and remove the dependancy of the homebridge-shelly plugin for me.

davidfierro, 2020-11-02:
> This would be very useful and remove the dependancy of the homebridge-shelly plugin for me.

+1

Ioan-Popovici, 2021-03-01:
Interesting ideea, I won't use it but I have friends that have mixed environments.

LamaXVIII, 2021-03-15:
So, do you think there's any chance of having both HomeKit and Alexa?

rojer, 2021-03-15:
there is a chance but it's very distant, there are way higher priority items on the list.

LamaXVIII, 2021-03-15:
> there is a chance but it's very distant, there are way higher priority items on the list.

Okey, I realise. Do you think there is a comfortable alternative?

Kuki77, 2022-11-30:
Is this topic "dead"? I'd also appreciate to use it with Homekit and Alexa.

ttimpe, 2022-11-30:
This should be merged with Matter support as that is what we really need.

timoschilling, 2024-09-03:
Adding Support for Alexa didn't make sense nowadays, for that Matter is the solution. To support Matter in this project would be a mess, Matter support for Shellys should be an own project. That's why I close this issue.

closed 2024-09-03 by timoschilling