mongoose-os-apps/shelly-homekit #36: AP Mode not working?

Kroellie, 2020-04-05:
Hi, my shelly 1 flashed with your firmware lost it's wifi connection. Now when I reset the Shelly 1 using 5 times on/off, I hear the clicking sound from the shelly, but no AP is presented. What could be wrong, or what can I do?

andyblac, 2020-04-05:
iirc it is 10 on/off to reset in this firmware

rojer, 2020-04-05:
as long as you can hear relays clacking rapidly, it means it switched to AP mode.
as to why you can't see the AP - i have no idea...
serial console output would help.

burton58, 2020-08-07:
I have the same issue with two shelly 2.5's. I reseted both (power off, 10x on/off) and got the lights flickering. But I do not see any WIFI that matches Shelly. Or is there a specific Wifi name/pw?

Tried it multiple times but without success

andyblac, 2020-09-20:
is this still an issue ?

andyblac, 2020-09-22:
I'll close this ticket, please reopen if still an issue.

closed 2020-09-22 by andyblac

fichtelmann, 2021-11-24:
I reactivate a shelly 2.5 today after a long time. Does not appear inside my configured Wifi. So I reset the device (5x switch). Hear the relais sound. But I can find the shelly AP to configure it new. Try it also with 2 more resets -> no success. Try the rest button on the back for 10 seconds. Relais were clicking. Still no success. Can't find the AP network. What shall I do ?

raj47i, 2025-11-10:
Same Problem for me, for two of my 2.5s