mongoose-os-apps/shelly-homekit #37: Original firmware for Shelly 1PM

bulgaro80, 2020-04-05:
Hi all,

I have updated all my Shelly 1PM and they work fine with HomeKit, but for one of them I need to revert back to the original firmware because I need to keep Alexa compatibility.
Is the firmware for Shelly 1 available in the Code page compatible even with Shelly 1PM?
If yes, how can I upload it? I'm used to us the following command line, but files names are different:
esptool.py -p /dev/cu.usbserial-00000000 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m \ 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 main.bin
Any support is welcome, many thanks!!
Stefano

andyblac, 2020-04-05:
I don't think Shelly1 firmware linked on age for 1PM, maybe a user has a copy they can share, one you get it try this method, https://github.com/mongoose-os-apps/shelly-homekit/blob/master/docs/flashing.md

bulgaro80, 2020-04-05:
Many thanks Andy for the procedure! I hope someone can share the original Shelly 1PM firmware.

Stefano

rojer, 2020-04-05:
http://api.shelly.cloud/firmware/SHSW-PM_build.zip is the original firmware url

rojer, 2020-04-05:
added links to README

closed 2020-04-05 by rojer

bulgaro80, 2020-04-05:
Hi all, last question: I downloaded the original firmware but considering the files within SHSW-PM_build.zip, what is the best tool/command line to uso for uploading the firmware?
Thanks!
Stefano
PS: I tried to use mos (as suggested by andyblac but the command line suggested doesn't work).

rojer, 2020-04-05:
use web interface, there's a form to upload firmware at the bottom of the page

bulgaro80, 2020-04-05:
unfortunately running some tests I erased the firmware and trying to install something. So I do not have access anymore to the web interface. That's the reason I was looking for a command to be launched via command line

rojer, 2020-04-05:
oh, hang on... you'll need hardware info file, otherwise stock firmware won't boot.
@kzyapkov do you have device recovery procedure documented somewhere?

kzyapkov, 2020-04-07:
Not for all devices, working on this. In the mean time @bulgaro80 can try https://github.com/ioprev/shelly-firmware or ask in the facebook support group or via email.

bulgaro80, 2020-04-11:
Many thanks @kzyapkov, I used this link [](https://github.com/ioprev/shelly-firmware) and it worked!