mongoose-os-apps/shelly-homekit #38: Shelly2.5: Power measurement support

RRostron, 2020-04-05:
good evening

i flashed the latest builds to my shellys ,  but on the shelly2.5s that i have, its not showing the power usage on SW1, it stays on 0w,  SW2 seems to work ok as that gives a reading.

rojer, 2020-04-06:
yep, you're right. stupid last minute bug on my part, fix incoming.

RRostron, 2020-04-06:
thanks rojer, 
another thing i noticed is on one of the shellys SW2 it shows a  negative - reading

rojer, 2020-04-06:
i noticed this on one of my devices too, except in that case SW1 is negative while SW2 is positive. absolute values seem correct but the sign is not. not sure why, will take a look.

rojer, 2020-04-06:
ok, i pushed 1.7.1 for shelly 2.5, please update

closed 2020-04-06 by rojer

RRostron, 2020-04-06:
hi rojer

that works thanks!