mongoose-os-apps/shelly-homekit #43: Water valve support

Snekatis, 2020-04-08:
Is it possible add support for Water valve? So HomeKit will see it as a real Water valve?

[image]

ursusHU, 2020-06-03:
And also Garage door pleasae :)

Montardit, 2020-07-15:
Yes, please! Sprinkles support!

danny1605, 2020-11-10:
Please, integrate the irrigation type !

lucaspinelli85, 2020-12-15:
it would be great for my garden, please integrate it!

andyblac, 2021-01-25:
pull request here https://github.com/mongoose-os-apps/shelly-homekit/pull/433

danny1605, 2021-02-03:
There is already new information regarding Water valve support ?

timoschilling, 2021-02-03:
Basic valve support is now merged, timer support maybe follow.

rojer, 2021-02-14:
please help test at https://github.com/mongoose-os-apps/shelly-homekit/issues/498

Montardit, 2021-03-16:
Shellys configured as water valve can't be added to automatizations. If I configure it as as switch it is selectable but if it is configured as a valve not. Not as a irrigation nor as basic valve. I've configured it with a shelly 2.5. They don't show in the accessories/ambient selector list of the automatization.

timoschilling, 2021-03-17:
@Montardit I can confirm that, but it looks like a bug or missing feature in the home app. I created an irrigation valve with the HomeKit Accessory Simulator (a developer app from Apple) and it behaves the same. Google for that and you will find some more reports about that. But you can use apps like Home+ or Eve to create a irrigation related automatisation, maybe using a scene to start the irrigation from a automatisation.

Montardit, 2021-03-17:
It's a little weird, I hope they'll fix this. It's strange you cannot 
water your plants regularly if you have an irrigation valve... I've 
tried with Eve and it works fine. Home+ is not free.

Thanks!!!

El 17/3/21 a les 3:50, Timo Schilling ha escrit:

danny1605, 2022-07-13:
hello, is there anything new regarding the timing of irrigation?

blitzi2000, 2022-10-28:
maybe adding a new HAP Type in the roller shutter mode so can be seeing as a water tap/sprinkler in homekit? that would allow opening and closing and automatize with a time setting...