mongoose-os-apps/shelly-homekit #46: Shelly1 no response in Homekit

petergPRG, 2020-04-12:
I have included a total of three Shelly1 in HomeKit. I have assigned a fixed IP to all devices and all devices are reachable under their IP. Two run without problems with SW version 1.3.3 ( 20200217-232831/1.3.3-g518dd07-master )
One Shelly always loses contact in HomeKit. This Shelly has the SW version 1.7.0 (20200331-194631/1.7.0-ga963921-master)
When I reboot this Shelly via the web browser he is accessible in HomeKit. After the first (sometimes second) switching via HomeKit the connection is lost.
Is there a way to flash this shelly to the 1.3.3 version?

andyblac, 2020-04-12:
download the 1.3.3 FW from releases, open the web page on the shelly scroll down to Firmware, chose the FW zip file and update.

petergPRG, 2020-04-12:
Thanks for answer. I tried this, but when I install the shelly HomeKit "flash" with the command: Shelly1: http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip
I have again the 1.7.0

andyblac, 2020-04-12:
ye it grabs the latest he has hosted, that diff to git builds

andyblac, 2020-04-12:
get fw from here https://github.com/mongoose-os-apps/shelly-homekit/releases to upload manually.

petergPRG, 2020-04-12:
Thx Andy, I downloaded the file, but I can't upload it to the shelly over the web interface. any Idea?

andyblac, 2020-04-12:
should be ale to 
[image]

see at bottom of page, click on upload, choose zip file click ok

petergPRG, 2020-04-13:
yep, successful , thanks a lot

andyblac, 2020-04-13:
np, glad you managed.

RedPlumber, 2020-05-15:
Same issue here, after updating few original shelly to fw 1.9 I often lose connection to the device in homekit. That's very frustrating. I updated one of the devices to 1.9.1 but it does not work.

andyblac, 2020-09-20:
is this still an issue ?

rojer, 2020-09-20:
this predates all the fixes that went into 2.0, @petergPRG please upgrade and reopen if this is still an issue.

closed 2020-09-20 by rojer