mongoose-os-apps/shelly-homekit #47: 2x Shelly 1: HomeKit Automations not working

felixmueller, 2020-04-13:
Hi there,

I installed multiple Shelly 1’s in my home, primarily to use them for automations.

For instance: when Shelly A is switched off, switch off Shelly B as well. 

Unfortunately, this does not work at all. When I create a new „When an accessory is controlled“ automation for Shelly A (disable) then disable Shelly B, nothing happens. 

Maybe it is kind of a timing issue, it is possible to trigger a smart bulb when switching a Shelly, but not another Shelly. 

Did someone experience the same issue and is there any solution for this?

Thanks a lot!

andyblac, 2020-04-13:
works here, what are you using as a HomeKit Hub ?

felixmueller, 2020-04-13:
I have an Apple TV and a HomePod in my network. The ATV is always the active Hub.

andyblac, 2020-04-13:
I had issue with ATV as hub, trying disabling as a Hub in gui.

felixmueller, 2020-04-13:
I just disabled the Home Hub feature on my ATV and the HomePod took over. And after disabling and re-enabling the Automations it now works!

Thank you very much @andyblac !

I do not understand why this is still an issue and Apple didn’t fix it yet.

closed 2020-04-13 by felixmueller

andyblac, 2020-04-13:
yeah, but thats Apple for you.

qx54, 2020-10-06:
I have an Apple TV as hub and Lifx bulbs. I'm about to buy Shelly 1's and 2.5's.

@andyblac & @felixmueller, are you saying that shelly-homekit won't work with my Apple TV as hub?

felixmueller, 2020-10-06:
Hi @qx54

I switched from this firmware do the homebridge-shelly plugin. 

Since then, I had no more issues with automations. 

The problems I had were not caused by my Apple TV. The Shelly firmware was not operating reliably.

qx54, 2020-10-06:
Hi @felixmueller 

Thanks for your feedback. I'm considering Homebridge now.

felixmueller, 2020-10-06:
@qx54 

I don't want to blame the firmware for my issues. I think its awesome software. 

It just didn't work out in my case. Maybe it was the WiFi or something else.

The Homebridge plugin however worked very reliably from day one.

rojer, 2020-10-06:
bugs happen, we need to figure this out. #178 is probably the same issue.

qx54, 2020-10-06:
I'm aware of this and from browsing a couple of issues related to my interests (e.g. single, double, long press) I can tell you're very active and doing a tremendous job!

But because I already have Sonos devices which by default don't integrate well with homekit, this might just be the push I needed to buy a Raspberry and install homebridge. So it's more about Sonos and not so much about this issue ;)

andyblac, 2020-10-06:
For me I find homebridge way less reliable than this firmware. I have no issues with firmware

My automations issues are down to the HomePod as it effects all my HomeKit devices

I personally would recommend this firmware over homebridge.

Rappelsack, 2020-10-07:
I guess this thread is now the active one? 

I wrote my issues already here: https://github.com/mongoose-os-apps/shelly-homekit/issues/178

So my next step was:
shut off the Homepod (so that only my iPad is active and the hub). Switching all automations to deactivate and activate again. Restart iPad.
And: the shutter (only this one who is displayed as a bridge) still is down even it should go up at 06:30 am this morning.
And all Shelly 1 (floor heating) stayed off this morning.

So I did the same troubleshooting but now with the Homepod as active hub:
same problems. 

Therefore I suggest that it must be the firmware of the shellys? It worked finde all the way to 2.1.8.