mongoose-os-apps/shelly-homekit #49: SHELLY 2.5 DUAL HOMEKIT SETTINGS

bolagnaise, 2020-04-14:
I am using a shelly 2.5 on a dual light switch in my lounge room. The problem i have is that you can only assign both switches to a single room, and one switch is for my outdoor light. It would be awesome if we could assign each switch their own homekit code and add them as individual switches for individual rooms in the home app.

andyblac, 2020-04-14:
reported here https://github.com/mongoose-os-apps/shelly-homekit/issues/18

bolagnaise, 2020-04-14:
Ok, so i guess my only option is to add another shelly 1 to the circuit to control the outdoor light. Thanks for info.

rojer, 2020-04-14:
yeah, that's #18. not so easy to solve unfortunately, due to lack of memory.

closed 2020-04-14 by rojer