mongoose-os-apps/shelly-homekit #50: Change Web UI to use WebSocket for RPC

rojer, 2020-04-14:
Mongoose OS supports RPC over WebSocket.
We should open a channel and keep it open to make RPC more efficient. This will also allow regular updates of power readings and other status without sending HTTP request each time.

rojer, 2020-09-29:
challenges:
 * ensure connection is re-established as necessary (e.g. when device reboots)
 * when applying new state updates, do not stomp over pending changes to fields made by the user.

rojer, 2020-09-29:
https://github.com/mongoose-os-libs/wifi-setup-web-ui can be used for inspiration

shellyrgbw, 2020-09-29:
> Mongoose OS supports RPC over WebSocket.
> 
> We should open a channel and keep it open to make RPC more efficient. This will also allow regular updates of power readings and other status without sending HTTP request each time.

shellyrgbw, 2020-09-29:
> https://github.com/mongoose-os-libs/wifi-setup-web-ui can be used for inspiration

juancitomadrid, 2020-09-30:
I can work in UI.
Let me know needs.

rojer, 2021-01-03:
good job @ZeevoX thank you very much!
UX is much better now

closed 2021-01-03 by rojer

andyblac, 2021-01-03:
@rojer any chance of beta release ?

rojer, 2021-01-03:
i don't intend to cut 2.8 just for this, so we won't call it a beta, but it makes sense to cut a build that can be tested.
i think we should just have latest build always available

rojer, 2021-01-03:
i filed an issue for this - https://github.com/mongoose-os-apps/shelly-homekit/issues/379

meanwhile, i've manually cut a one-off - http://rojer.me/files/shelly/2.7.0-latest/