mongoose-os-apps/shelly-homekit #51: Bypass auto off.

andyblac, 2020-04-14:
would it be possible to bypass the auto off, on a double press of the physical switch ?. So it would just ignore the auto off for that session only, if one has been configured.

rojer, 2020-04-15:
can you describe in detail how you see it working and why would it be useful?

andyblac, 2020-04-15:
ie in Garage i have it set to auto turn off lights after 20mins, as mostly that long enough, but some times i work in there, so need longer, it would be nice to press the light switch twice so it will bypass auto off and remain on until i manually turn it off.

andyblac, 2020-05-17:
@rojer any more info needed ?

rojer, 2020-05-17:
i understand your use case but i'd like to see more interest before adding it to the firmware.

andyblac, 2020-05-17:
ok, np.

Jaguaza, 2020-05-17:
> es decir, en el garaje lo tengo configurado para apagar automáticamente las luces después de 20 minutos, ya que la mayoría de las veces es suficiente, pero algunas veces trabajo allí, por lo que necesito más tiempo, sería bueno presionar el interruptor de la luz dos veces para evitar el apagado automático y permanecer encendido hasta que lo apague manualmente.

Aqara button type ?, that you can create several rules according to the keystrokes, one, two or three.

This would be interesting.

andyblac, 2020-05-17:
no, this request is here https://github.com/mongoose-os-apps/shelly-homekit/issues/9, i am talking about the physical switch and auto power off in shelly UI.

Jaguaza, 2020-05-17:
> no, esta solicitud está aquí [# 9](https://github.com/mongoose-os-apps/shelly-homekit/issues/9) , estoy hablando sobre el interruptor físico y el apagado automático en la interfaz de usuario shelly.

ok

andyblac, 2020-09-19:
@rojer now we have detached mode which supports single, double, long press, would be possible to add support for this, when in normal HAP mode.

rojer, 2020-09-19:
actually, yes, now that we have double press detection, this would be trivial.
but how to present it in the UI? or just make it the default behavior? i don't imagine many people would accidentally double-press a switch. but we need to document this behavior.

andyblac, 2020-09-19:
> actually, yes, now that we have double press detection, this would be trivial.
> but how to present it in the UI? or just make it the default behavior? i don't imagine many people would accidentally double-press a switch. but we need to document this behavior.

maybe via a check box, only shown / enabled on default if auto off is enabled.

rojer, 2020-09-19:
not a bad idea. we should also hide auto-off time when auto-off is disabled.

rojer, 2020-10-22:
now i'm thinking long press to disable auto-off would be even better.
easy to do, too.

rojer, 2020-10-22:
@andyblac please try this - https://rojer.me/files/shelly/2.4.0-beta1/
seems to be working in my quick testing.

andyblac, 2020-10-23:
> @andyblac please try this - https://rojer.me/files/shelly/2.4.0-beta1/
> seems to be working in my quick testing.

works great. thx

rojer, 2020-10-23:
ok, will be in the next release.

closed 2020-10-23 by rojer