mongoose-os-apps/shelly-homekit #53: “Auto off” functionality in mili-seconds....

Flying150, 2020-04-19:
Hello Folks,

The 1.8 release is awesome! Thanks a lot ;o)
I face a specific application that would require pulses in mili-seconds.....
The regular Shelly firmware can be set up this way. Any chance that the next release could consider a format: like: 00:00:00:01 or 00:00:00:05 ?

Thanks for your support in advance,
Flying150!

rojer, 2020-04-19:
yes, it's possible. we can change `sw.auto_off_delay` to floating point and string format to `HH:MM:SS.sss` with `sss` being the millisecond part.
it's an easy one, any takers?

rojer, 2020-04-19:
@handy-manny-swe perhaps?

handy-manny-swe, 2020-04-19:
Sure @rojer, but I have some concerns that I would like to address first. This might lead to consuming the one and only hardware timer that is available on the ESP8266 which the Shelly1 is based on. Are we ok with that, or do we plan for using the hardware timer for something else?

If we need to support intervals less than 10ms the hardware timer is recommended according to the [mongoose documentation](https://mongoose-os.com/docs/mongoose-os/api/core/mgos_timers.h.md). @Flying150 would a 10ms interval be enough accurate for your needs, or would we need to support values down to one millisecond?

rojer, 2020-04-19:
we definitely won't be supporting intervals less than 10ms, there's simply no point as relay won't be able to switch that fast.

handy-manny-swe, 2020-04-19:
Okay, so that would keep us at the software timer. Good. I'll try to find some time for this next week.

rojer, 2020-04-19:
yeah, software timer will be adequate.

Flying150, 2020-04-20:
Agreed gentlemen 👍
Looking forward to the implementation of the change ....and thank you soooo much in advance...

Best wishes

yeah, software timer will be adequate.

closed 2020-04-22 by rojer

herrmanthegerman, 2023-09-15:
Hello,

now that "**auto off**" with sub-seconds resolution has been implemented, I wonder whether an "**auto on**" feature with sub-seconds resolution would be feasible. I'm asking as I'd like to use a shelly plug to simulate short power outages. The original firmware for shelly plug includes an "**auto on**" function but only with seconds resolution.
Thanks.

TimelyBanana, 2025-02-19:
Hi, since 3.40.20, this functionality is now broken, and has the HH:MM:SS format. It is no longer possible for me to set 500ms on my Shelly 1 Mini Gen 3.