mongoose-os-apps/shelly-homekit #55: Outlet device type?

freddair, 2020-04-20:
Hi and a thousand thanks for your outstanding work which makes getting Shelly into HomeKit so easy for anyone! I am just wondering if it’s possible to define a Shelly as an alternative HomeKit device type such as an outlet? That would be very helpful. Thanks, Fred

rojer, 2020-04-20:
absolutely. this should be easy to do, we have "on" characteristic that is the same as switch and can wire "in use" to always be true.

gregDEweb, 2020-04-20:
This would be very helpful but with the recent change in fw1.8.0 the Shelly can be used for a lot of different equipments. 
I‘m using it now as a garage door opener, water valve, sprinkler, lock, switch and probably some others will come up with some more ideas. 
Question is if it will be possible to include as many as possible.

rojer, 2020-04-22:
I think we should support as many as it makes sense and make it a choice in the web UI.
Outlet device makes sense, so we should support it.
This will need a bit of refactoring of the code, but is certainly doable.

schemhad, 2020-04-23:
Awesome project, thank you very much. 

I would like to define one of my shelly's (shelly1) as a door lock device type. My door opener works fine as a "Switch" in Homekit, but the device type would change the icon in homekit and the siri voice commands. 

Therefor i could say "Hey Siri, open the Door" instead of "Hey Siri, switch the Door on." ;-)

schemhad, 2020-04-23:
Could it work if I change the Type from Switches to Locks by myself?

```
static HAPAccessory s_accessory = {
...
    .category = kHAPAccessoryCategory_**Switches**,
...
```

```
static HAPAccessory s_accessory = {
...
    .category = kHAPAccessoryCategory_**Locks**,
...
```

andyblac, 2020-04-23:
think you would need to change characteristics as well locks don't have a ON etc...

```
...
static const HAPCharacteristic *shelly_sw_on_char(uint16_t iid) {
  HAPBoolCharacteristic *c = calloc(1, sizeof(*c));
...
```

rojer, 2020-04-23:
that's right, you need to make sure that the right set of characteristics is exported by the service.
the right set is described in the HAP reference manual (you can download it [here](https://developer.apple.com/support/homekit-accessory-protocol/) for free).

schemhad, 2020-04-23:
OK, so I think for my simple electric door latch I would need the accessory-type "_**8.26 Lock Mechanism**_", because the Type "_**8.11 Door**_" requires characteristics like e.g. the "current position" which I don't have, right? It's not a fully motozied door with contacts everywhere, just the opener on my intercom...

I'm afraid, that the change with all characteristics will exceed my "**skills**".

So I'm hoping for a future release with more devices like e.g. the "**_lock mechanism_**".

freddair, 2020-05-09:
Und feelings on when this may be implemented? Unfortunately I have no coding skills other wise I would love to start with the lock mechanism. Thinking about flashing HAA in the meantime but it’s so much more complicated than shelly-HomeKit...

rojer, 2020-05-09:
selectable device types is literally the next thing to do on my list.

rojer, 2020-05-16:
@schemhad i am implementing lock mechanism and want to make sure: "on" state of the shelly switch should correspond to "unsecured" state, and "off" to the "secured", correct?
logic being: default / de-energized relay state should correspond to "secured", which is the more secure state.

rojer, 2020-05-17:
pushed a 2.0.0 pre-release - https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.0.0
please test and let me know.

Jaguaza, 2020-05-17:
Good Morning.

Could a fan device be added, so that it is reflected in homekit even if it is with the on off function, for the fans that we control with shelly? Thank you

Borja90, 2020-05-17:
> pushed a 2.0.0 pre-release - https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.0.0
> please test and let me know.

Thanks a lot! This is great and an awesome work you are doing! Looking forward to see how it keeps improving and adding this and devices types

freddair, 2020-05-17:
I updated to 2.0.0 on a Shelly 1 which is installed behind a light switch.

I changed the device type to outlet, saved and rebooted. Device change didn’t reflect in home app immediately, so I tried to change to lock. Then home app showed an outlet, but not responding. Later it showed a lock not responding. After another reboot, it was responding and I could use it as a lock. 

That confirmed to me that it worked after some delays and reboots, so I changed it back to a switch  then it was again not responding in home app, even after another reboot.

Then I downgraded to 1.9.1, but it is still not responding in the home app.

Any idea? Would be kind of sad if we had to reset the HomeKit ID and have all automations and scenes reconfigured every time.

rojer, 2020-05-17:
@freddair you actually don't need to reboot after save, the device reboots automatically.
in my testing i found that device type changes quite reliably.
the fact that device is not responding after downgrade is expected - apple home app actually doesn;t allow firmware downgrades and disables devices if it finds an older firmware version.

so, can you try going back to 2.0.0 and try changing device types without manually rebooting? it's possible that additional reboot confuses the app.

rojer, 2020-05-17:
@Jaguaza 
> Could a fan device be added, so that it is reflected in homekit even if it is with the on off function, for the fans that we control with shelly? Thank you

yes, i think it's reasonable. but please file a separate issue for it.

freddair, 2020-05-17:
@rojer Thanks for explanation, it now works.

rojer, 2020-05-17:
@freddair ok, thanks.
to reduce confustion i removed auto-reboot and instead changed the prompt to ask the user to reboot manually. rebooting the device may change state of the outputs, we shouldn't be doing it willy-nilly. it will also be better UX in case they want to make some more changes before rebooting.

anyway, i pushed 2.0.1, this is now deployed.

closed 2020-05-17 by rojer