mongoose-os-apps/shelly-homekit #56: Mode explanations

JacVid, 2020-04-20:
Hello,

I'm trying this very interesting firmware, but i couldn't find any answers to the device web configuration features after flashing (fw 1.8).

About Type:
- Momentary: what it does ?
- Toggle: what is it for ?
- Edge: what is the use case ?
- Detached: New feature, wha is it for ?

About Initial State:
- Last: OK
- ON: OK
- OFF: OK
- Input: can you explain ?

About Auto OFF:
- Is it a time you define to turn the device OFF, no matter its previous state (it goes to OFF every night at 11PM for exemple) ?
- Or is it the time the device turn off after being turned ON (when switched to ON, it turns off after 5min) ?

Thanks for your explanations.

JacVid, 2020-04-20:
Thank you very much @andyblac !

Nevertheless, I don't get the difference between Toggle and Edge ? I'm sorry :(

andyblac, 2020-04-20:
np, let say you have a light that is controlled by 2 switches, one at bottom of stairs we call it SW1 and one at top SW2.

LIGHT OFF, you switch on with SW1, walk upstairs turn of with SW2

now let say you add a shelly to the light, it would need to put into EDGE mode for this to work.

JacVid, 2020-04-20:
Thanks for your explanation !

But I still try to understand what could be the difference in that case. Because the Toggle mode will do the same, right ? 

In your case, if SW1 or SW2 changes its status, then SW Input is open or close to drive the light, so it has the same behavior as Toggle mode...

I'm really sorry but I really don't get the difference between Toggle and Edge Mode. For me, it seems to be identical regarding what you showed me.

Anyway thanks for your time, I really appreciate :)

freddair, 2020-04-20:
Hi, here is what I found out:

TOGGLE always relates the relay state to the power state ON or OFF coming from the switch. This works the same way for 1, 2 or 3 switches in a hallway. When you switch the shelly via the home app or automations, it may not be in sync with the switch state any more and then you have to press the switch two times to switch the relay again.

EDGE means that the relay changes its state every time when the switch also changes. It doesn’t matter if the switch is on or off, it will alsbald toggle the relay state. That’s a good mode if you want to use a wall switch and automations / home app without the effect mentioned above.

Kind regards from Berlin
Frederik

JacVid, 2020-04-20:
Thanks @freddair !

Your explanation is very useful. 

In that case, toggle mode is useless and I would prefer use Edge Mode, even with a single rocker, because then you are able to turn the light ON/OFF via the app or an automation without issue if want to use the rocker back to manually change the status, right or I'm still completely confused ?

Kind regard from Paris

freddair, 2020-04-20:
Correct. Edge mode is your friend.

There are other use cases where toggle mode makes sense, I use it for my coffee maker where I built in a Shelly 1 and use the automatic physical switch-off ... :)

closed 2020-04-20 by JacVid

andyblac, 2020-04-20:
just keep in mind even with EDGE mode a single rocker will NOT act normally.

ie normal ON = down

if you turn ON via switch (down), then turn OFF via HomeKit, to turn ON again via switch would be UP. 

this is NOT wife friendly, this i why i had to change my switches to PUSH button type and use MOMENTARY setting.

JacVid, 2020-04-20:
@andyblac, thanks for your reminder. 

I agree, but what if in Toggle mode, regarding your case, the rocker doesn't change the status of the light, it is not wife friendly either, right ?

Instinctively, when the light is OFF, you change the rocker state to turn it ON, no matter if it is up or down. So to me Edge seems to be the best config.

andyblac, 2020-04-20:
yeah in TOGGLE, 1st you would have to switch OFF (UP), then ON (down), your correct, 

again this is why i why out all my Switches to Push Button switches, and use MOMENTARY in settings. As each press is the same (ON or OFF).

ie
press to turn ON, then in HomeKit turn OFF, to turn ON again on switch you just press the button.