mongoose-os-apps/shelly-homekit #58: Shelly 1 won't update

MGLange, 2020-04-21:
Hi,
My Shelly 1 is on version 1.3.3. Of your firmware. I can't update it or return it to stock firmware.
Build ID: 20200217-232831/1.3.3-g518dd07-master

Can't update via the web interface, can't connect to homekit, can't revert to stop stock via the web interface.

I initially installed the firmware via the provided URL:
Shelly 1: http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip 
Can someone please help me?

What am I doing wrong?

To be honest I'm new to all this (Github) so forgive me if I'm in the wrong place to ask this.

rojer, 2020-04-21:
no, you've come to the right place.
what error do you get when trying to update?

MGLange, 2020-04-21:
When I use the update function in the web interface: -1 Archived file size is unknown

When I use the provided URL (http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip): Not Found

rojer, 2020-04-21:
the url method does not work on the homekit firmware, only on stock.
updating from homekit to new version or returning to stock should be done with form upload.

"Archived file size is unknown" suggests that ZIP file was recreated or modified.
firmware accepts non-streaming ZIP files only. did you re-pack it or something?

petergPRG, 2020-04-21:
Update function with Safari on a Mac is not possible. I've had the same Problem. Use firefox and it will work

MGLange, 2020-04-21:
I downloaded the zip file on my mac. Had to compress it because mac automatically unzips.

I'll try Chrome or Firefox. Keep you posted

rojer, 2020-04-21:
> Had to compress it because mac automatically unzips.

there's your problem. make sure you upload _exactly_ the file you downloaded, as not all zips are created equal.

closed 2020-04-21 by rojer