mongoose-os-apps/shelly-homekit #59: Naming Shelly Devices...

andyblac, 2020-04-21:
i have a few shelly devices now, and it starting to get confusing when i open the GUI on a device. Could we allow naming the device in the GUI, or at least display the HomeKit device name in GUI ?

rojer, 2020-04-21:
the Home app does not propagate name changes from to the device. Name is a read-only characteristic. what we can do is add ability to change the name we present to the Home app, so if you set it from web UI and do not override in the Home app, it will change in both. would that work?

andyblac, 2020-04-21:
i manually change HomeKit accessory names anyway, as long as we can give a name the device in WEB UI, so i can tell what device I am in, that's enough for my needs.

rojer, 2020-04-21:
sure, that can be done.

rojer, 2020-04-21:
we already have a config option for it, even (`swX.name`). it's just a matter of surfacing it in the ui.

closed 2020-04-22 by rojer

rojer, 2020-04-22:
committed a simple solution. device needs to be rebooted for the name change to be picked up by the ios device (can be done from the web interface).