mongoose-os-apps/shelly-homekit #62: version number not correct in homekit

andyblac, 2020-04-23:
as you can see in screenshots below, web ui shows correct version 1.9.1, but homekit showing 1.8

[image]
[image]

but if i build it myself and upload that fw, it shows in homekit correct.

rojer, 2020-04-23:
does not repro for me.
home app may not have updated the version when you were looking.

andyblac, 2020-04-23:
[image]
[image]

rojer, 2020-04-23:
version string reported on web ui and sent to ios app are literally the same.

andyblac, 2020-04-23:
> does not repro for me.
> home app may not have updated the version when you were looking.

i have been on 1.9.0 since yesterday ???, it was still showing 1.8, until i built it myself ?

rojer, 2020-04-23:
well, i flashed the firmware to my unit before uploading to the site and it shows 1.9.1 in both web ui and home app.

andyblac, 2020-04-23:
from github resources ?

rojer, 2020-04-23:
the same zip file, yes. technically, it was just before it became github resource :)

andyblac, 2020-04-23:
hmmm, i'll try reflashing fw file github again.

andyblac, 2020-04-23:
well, sorry but it back as 1.8

[image]

rojer, 2020-04-23:
no, it is definitely 1.9.1. i unpacked it and ran strings on the binary:
```
[rojer@nbd ~/shelly-homekit master]$ unzip ~/Downloads/shelly-homekit-Shelly1-1.9.1.zip
[rojer@nbd ~/shelly-homekit master]$ strings switch1-1.9.1/switch1.bin | grep 1.9.1
1.9.1
20200423-164607/1.9.1-geb66659-master
```

andyblac, 2020-04-23:
yeah, as webui is correct, but as you can see HomeKit is showing 1.8

rojer, 2020-04-23:
i don't know, it's a mystery.
in fact, i tried downgrading my device and it stops working in the home app. i actually remember reading that ios will not allow downgrades, so it stops communicating if version is below 1.9.1 in my case.

andyblac, 2020-04-23:
it worked here, strange

andyblac, 2020-04-23:
can you try flashing the above ?

rojer, 2020-04-23:
updated shelly1 with your file - worked.
[image]
[image]

andyblac, 2020-04-23:
strange, maybe HomeKit is caching FW ?

andyblac, 2020-04-23:
i'll keep an eye on it, with your next release.

schemhad, 2020-04-28:
Could you please try an hardreset of the shelly (power off, power on, switch 5 times) and flash the latest release again? I can not find the fault you indicate with my iPad 4 and Shelly1 ...

andyblac, 2020-04-28:
tried that already, i'll try fw 1.9.2 from github once it is released

closed 2020-06-05 by andyblac