mongoose-os-apps/shelly-homekit #65: hostname change

andyblac, 2020-04-28:
is it possible to edit the hostname, like you did you switch names ?

[image]

rojer, 2020-04-28:
yes, it is possible. it is specified in `device.id` config variable

MilanBacaCz, 2020-07-31:
and will it be accessible from config gui?

andyblac, 2020-09-19:
@rojer could this be added, be handy when you have multiple shelly's for the script to quickly see which is which.

andyblac, 2020-09-19:
@rojer does this look correct ?, if so I'll create a pull request 

https://github.com/andyblac/shelly-homekit/commit/eb66793be77a97ac83190a3c8278659fd6318b66

how can I test such things locally ?

andyblac, 2020-09-19:
@schemhad if you have time could take a look at this ?

schemhad, 2020-09-19:
That's why i've set the pagetitle dynamically in the recent pull request. 😉

But i would place it in the wifi container.

andyblac, 2020-09-19:
does my change look correct (just move to wifi), I'm new as JS, just trying to understand from existing code.

rojer, 2020-09-19:
> @rojer does this look correct ?, if so I'll create a pull request
> 
> [andyblac@eb66793](https://github.com/andyblac/shelly-homekit/commit/eb66793be77a97ac83190a3c8278659fd6318b66)

not quite. this needs to be sent to the device as Device.SetConfig RPC, and the device needs to apply it to the config.
a small code change is needed.

> 
> how can I test such things locally ?

there's a way to iterate on web changes without reflashing the device.
1) make the changes to `fs_src/index.html`
2) compress it:

```
[rojer@nbd ~/allterco/shelly-homekit master]$ make fs/index.html.gz
gzip -9 -c fs_src/index.html > fs/index.html.gz
```

3) upload to the device using the `mos` tool:

```
$ mos --port ws://shelly1-68b743.local/rpc put fs/index.html.gz
```

4) refresh the webpage in the browser, test the changes

rojer, 2020-09-25:
https://github.com/mongoose-os-apps/shelly-homekit/commit/95c21e455b836f81bdf320ce28ddbec400484d9e adds the API feature, @schemhad please add GUI when you have the time.

rojer, 2020-09-27:
i was adding system settings section for roller-shutter and added hostname item to it.

andyblac, 2020-09-27:
@rojer just tested 2.2.0-beta1 for this, after changing the host name, you can indeed access the webUI via the new hostname, But the hostname does change when scanning via DNS-SD. can you check your are getting same result ?

as a result the flash_script no longer sees the device with changed hostname.

rojer, 2020-09-27:
there's a slight issue there, yes. while _hap._tcp name does get fully updated, _http._tcp does not until the device is rebooted.
it remains in a half-updated state:
```
=;wlp0s20f3;IPv4;shelly25-1;_hap._tcp;local;shelly25-1.local;192.168.11.75;9000;"ci=14" "sf=0" "s#=1" "pv=1.1" "md=Shelly2.5" "id=85:54:60:AE:B7:6D" "ff=0" "c#=36"
=;wlp0s20f3;IPv4;shelly25;Web Site;local;shelly25-1.local;192.168.11.75;80;"arch=esp8266" "fw_id=20200927-161341/2.1.8-21-g1adc21d-rsh" "id=shelly25"
```
here i renamed shelly 25 to shelly25-1, HAP has been fully renamed and hostname has been updated, but the web service is still advertised under the old name.
while it's possible to fix in the firmware, it's not very easy. instead, if you just use $7 instead of $4 in the script, it should be ok.

andyblac, 2020-09-27:
this what is get
[image]

andyblac, 2020-09-27:
oh yeah using _hap

[image]

but this mean we would have to check every device for compatibility, this will slow search down a lot.

rojer, 2020-09-27:
hm, true. i'll think about it, but at the moment all that's needed is a reboot.
can you confirm taht reboot fixes it?

andyblac, 2020-09-27:
yes it does, so maybe just do an auto reboot after name change ?

[image]

andyblac, 2020-09-27:
but anyway, thinking about it I'll still have to check all _http as we ca not now search for 'shelly' in $7 any more.

andyblac, 2020-09-27:
checking all _hap add a lot of time to scan, best stick with _http

andyblac, 2020-09-27:
PR https://github.com/mongoose-os-apps/shelly-homekit/pull/152 should fix scanning, but a the reboot is required after hostname change.

andyblac, 2020-09-29:
@rojer did you add a automatic reboot after setting Hostname ?

rojer, 2020-09-29:
not yet, but i'll get to it before 2.2

andyblac, 2020-09-29:
ok I'll leave this open for time being then.

rojer, 2020-09-30:
ok, i fixes the issue on the 2.2 branch, will be in the next beta

closed 2020-09-30 by rojer

andyblac, 2020-10-01:
confirmed working thanks.

viritt, 2021-01-03:
A quick question!
I am using 2.7.0 and I can successfully change the "Name" under General Settings.
According to the Wiki, this changes the hostname.

However I see that on DHCP request, it sends Option 12 (Hostname) as the previous hostname "shellyswitch25-MACADR"
Also, under Settings > Device > Model: Shelly25, ID: shellyswitch25-MACADR has the previous name.

Were all these supposed to be set by the same "Name" change?
Am I missing something ?

andyblac, 2021-01-03:
> A quick question!
> I am using 2.7.0 and I can successfully change the "Name" under General Settings.
> According to the Wiki, this changes the hostname.
>
> However I see that on DHCP request, it sends [...]
device ID renames the same but hostname should be changed

```
Host: http://Lounge-Light-SW.local
Device Name: Lounge-Light-SW
Device ID: shelly1-F28095
IP: 192.168.3.38
Model: Shelly1
Current: HomeKit 2.7.0
Latest: HomeKit 2.7.0
```

viritt, 2021-01-03:
Not sure where you got that, from /rpc/Shelly.getInfo I get this.

`{"device_id": "shellyswitch25-E0B932", "name": "shelly-Sala", "app": "switch25", "model": "Shelly25", "stock_model": "SHSW-25", "host": "shelly-Sala.local", "version": "2.7.0", "fw_build": "20201222-230927/2.7.0-g1041f8b-master", "uptime": 205, "failsafe_mode": false, "wifi_en": true, "wifi_ssid": "aaa", "wifi_pass": "***", "wifi_rssi": -62, "wifi_ip": "192.168.2.61","hap_cn": 2, "hap_running": true, "hap_paired": true, "hap_ip_conns_pending": 0, "hap_ip_conns_active": 1, "hap_ip_conns_max": 9, "sys_mode": 1, "rsh_avail": true, "gdo_avail": true, "debug_en": false, "sys_temp": 54, "overheat_on": false, "components": [{"id": 1, "type": 4, "name": "Persiana Sala", "in_mode": 1, "swap_inputs": false, "swap_outputs": false, "cal_done": true, "move_time_ms": 25789, "move_power": 157, "state": 0, "state_str": "idle", "cur_pos": 0, "tgt_pos": 0}]}`

I do not refer to the http info that the shelly broadcasts through Bonjour, but to the actual hostname it sends the DHCP server upon boot.
Whenever I sniff dhcp request packets, its Hostname (option 12) comes with device_id instead of name.
Shouldn't it also broadcast the new "hostname" instead of device_id ?

andyblac, 2021-01-03:
```
    "info": {
        "device_id": "shelly1-F28318",
        "name": "Kitchen-Light-SW",
        "app": "switch1",
        "model": "Shelly1",
        "stock_model": "SHSW-1",
        "host": "Kitchen-Light-SW.local",
        "version": "2.7.0",
        "fw_build": "20201222-230821/2.7.0-g1041f8b-master",
        "uptime": 452860,
        "failsafe_mode": false,
        "wifi_en": true,
        "wifi_ssid": "ABFi-IoT-2.4",
        "wifi_pass": "***",
        "wifi_rssi": -53,
        "wifi_ip": "192.168.3.34",
        "hap_cn": 26,
        "hap_running": true,
        "hap_paired": true,
        "hap_ip_conns_pending": 0,
        "hap_ip_conns_active": 7,
        "hap_ip_conns_max": 9,
        "sys_mode": 0,
        "rsh_avail": false,
        "gdo_avail": true,
        "debug_en": false,
        "components": [
            {
                "id": 1,
                "type": 0,
                "name": "Kitchen Light",
                "svc_type": 0,
                "in_mode": 0,
                "in_inverted": false,
                "initial": 2,
                "state": false,
                "auto_off": false,
                "auto_off_delay": 600.0
            }
        ]
    }
```

andyblac, 2021-01-03:
i know it works, as my script does a dns browse to list all shelly devices on the network.

timoschilling, 2021-01-04:
I can confirm what @viritt says. My router (a Fritz!Box) lists the shelly's with there devise ID instead of the devise name. 
That doesn't conflict with that what you say @andyblac. I can confirm that your script works well too.

andyblac, 2021-01-04:
seems ok via dns-sd

```
andrews-mbp:tools andrewblackburn$ dns-sd -B
Browsing for _http._tcp
DATE: ---Mon 04 Jan 2021---
13:16:12.115  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
13:16:12.116  Add        3   7 local.               _http._tcp.          shellyrgbw2-B09A29
13:16:12.116  Add        3   7 local.               _http._tcp.          shellydimmer2-40F52000B1C9
13:16:12.116  Add        3   7 local.               _http._tcp.          shellyrgbw2-B0A3EF
13:16:12.116  Add        3   7 local.               _http._tcp.          shellyrgbw2-B075D3
13:16:12.116  Add        3   7 local.               _http._tcp.          shellyrgbw2-B03CAE
13:16:12.116  Add        3   7 local.               _http._tcp.          shellydimmer-F3CD89
13:16:12.116  Add        3   7 local.               _http._tcp.          Andys-Bedroom-SW
13:16:12.116  Add        3   7 local.               _http._tcp.          shellydimmer2-40F520015A2D
13:16:12.116  Add        3   7 local.               _http._tcp.          shellyrgbw2-B0C8BB
13:16:12.116  Add        3   7 local.               _http._tcp.          Kitchen-Light-SW
13:16:12.116  Add        3   7 local.               _http._tcp.          Garage-Light-SW
13:16:12.116  Add        3   7 local.               _http._tcp.          Landing-Alarm-SW
13:16:12.116  Add        3   7 local.               _http._tcp.          MediaRoom-Lamps-Button
13:16:12.116  Add        3   7 local.               _http._tcp.          Lounge-Light-SW
13:16:12.116  Add        3   7 local.               _http._tcp.          Bathroom-SW
13:16:12.116  Add        3   7 local.               _http._tcp.          shelly1l-84CCA8ADF044
13:16:12.116  Add        2   7 local.               _http._tcp.          Mums-Bedroom-SW
13:16:39.251  Add        2   7 local.               _http._tcp.          shellydw2-D8BFC01A85C5
```

viritt, 2021-01-04:
I confirm too it works well via dns-sd.

However, It does not work well via standard DHCP request. Option 12 - Hostname is requested with device id and not device name. Probably 99% of routers will find the device via this method and not with dns-sd.

andyblac, 2021-01-04:
one for @rojer then.

rojer, 2021-01-04:
agree that DHCP request option value should change.
ID is not supposed to change with change of the name and will always stay unique.