mongoose-os-apps/shelly-homekit #66: Interaction with other HomeKit devices

gonzague, 2020-05-01:
Hey guys
I don't know if it just me .. : I have a Shelly 2.5 running your firmware (thanks for that by the way).

It is powering a mirror's light in my bathroom. The bathroom contains a movement sensor (Philips Hue) and a Hue Tap.

In HomeKit, you can say "if button one of the Hue Tap is press" do "this". Or "if a movement is detected" do "this". I tried linking : if a movement is detected, turn on the mirror (with the Shelly) to no avail.

Is it expected / Can it be fixed somehow? :)

Thanks a lot

andyblac, 2020-05-01:
you'll need to do an automation in HomeKit to turn on Shelly2.5 switch if movement is detected.

"A Sensor Detects something"
[image]

choose the sensor.
[image]
[image]

choose what to turn on
[image]

end up with something like
[image]

gonzague, 2020-05-01:
Hey Andy, thanks that's what I had done but it wasn't working and .... mystery of Apple, I went into my bathroom .. and it started working all of a sudden. thanks!!

closed 2020-05-01 by gonzague

gonzague, 2020-05-01:
It is unstable though.. It worked and now doesn't work anymore. I'll investigate

andyblac, 2020-05-01:
what hub are you using ?, if you have an HomePod, i recommend to use that rather than AppleTV's, ATV's are not reliable.

gonzague, 2020-05-01:
I'm using an Apple TV :-( crap haha

andyblac, 2020-05-01:
Yeah ATV’s since iOS 12 are the worst 🙁

cuttymello, 2020-05-03:
I have the exact same problem using an iPad as hub.. It just doesn't work

andyblac, 2020-05-03:
yeah, HomePod seems to be the best option atm.