mongoose-os-apps/shelly-homekit #69: OTA flash fails.

jurban85, 2020-05-09:
Hey people,

I have a new Shelly and I wanted to flash the stock fw ota. Now I have done this before and it worked like a charm, this time it doesn’t. 
I get the following message:
{“status”:”updating” , “has_update” : false, “new_version” : “ “, etc. 

Afaik there should be an entry with the version number. Does anybody know how to fix this without flashing it via serial. I don’t even have the hardware to do so. 

Thanks!

b-e-n-e, 2020-05-11:
same here with a new shelly1:
{"status":"updating","has_update":false,"new_version":"","old_version":"20191217-140757/1047-long-id-for-shelly-devices@...."}

rojer, 2020-05-11:
> I wanted to flash the stock fw ota

let me clarify: you are trying to go back from homekit to stock fw?

b-e-n-e, 2020-05-11:
can only speak for myself, I am trying to go from stock fw to homekit via OTA and this url:
http://ip.of.my.shelly/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip

andyblac, 2020-05-12:
snap, just got 4 new Shelly 2.5's they won't flash via OTA `http://192.168.3.53/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip`

````
{"status":"updating","has_update":false,"new_version":"","old_version":"20191217-074901/1047-long-id-for-shelly-devices@ab2bdce7"}
```

andyblac, 2020-05-12:
ok fixed issue, @b-e-n-e @jurban85 check to make sure you shelly's have an working internet connection, my Shelly was not getting a working DNS.

b-e-n-e, 2020-05-12:
@andyblac 
Thanks a lot for the hint! I had a wrong gateway IP entered in the shelly. Strange thing is, that I had an info about a new fw available for the shelly. That info disappeared after my first try to flash the homekit fw OTA. Because of this info about the new fw I was thinking that internet connection is working properly.
Now everything is working fine! OTA update worked and homekit integration is done.

Thank you!

andyblac, 2020-05-12:
np glad you got it working.

seritos, 2020-05-14:
@b-e-n-e Where you able to go from the stock FW to 3rd party FW? I am trying it, but I can't... 

Would you please share how to assembly those files? is just as simple as to paste the FW I would like? 

Like: 

http://shellyip/ota?url=http://thehackbox.org/tasmota/tasmota.bin

andyblac, 2020-05-14:
@seritos you can do OTA firmware that is monogo os not tasmota, also the fw has to bin zip containing all fw files, not just a bin.

seritos, 2020-05-14:
Thanx Andy for your reply. 

Is not possible anyhow to do so? This would help a lot to the community!

THnx

andyblac, 2020-05-14:
from what i understand of it, no, you can OTA flash a different OS of the device aka Tasmota.

seritos, 2020-05-14:
Would it be possible to request this improvement? 

Thanks!

andyblac, 2020-05-14:
you'd have to ask @rojer , but i don't think it is possible.

seritos, 2020-05-14:
@rojer Would it be possible? 

Thanx!

rojer, 2020-05-14:
it may be possible to create an intermediate mongoose-os based firmware that would be used to flash other firmware, e.g. tasmota.
however, i am not interested in doing that, personally.

seritos, 2020-05-14:
Thanks @rojer for your question. Would you support me/us if I would try to do this intermediate FW?

Looks like there are 5 bin files zipped and control each file with hash sha1 y sha256....

If we could somehow get the documentation, or somewhere start, would be awesome to get right into.

Thanx!

rojer, 2020-05-14:
> Thanks @rojer for your reply. Would you support me/us if I would try to do this intermediate FW?

sure.

> Looks like there are 5 bin files zipped and control each file with hash sha1 y sha256....

you don't need to care so much about mongoose firmware format, more about how to flash the 3rd party one, once yours is running on the device.
i see 3 major steps:
1) get a basic mongoose firmware going that is OTA-able from stock. you can easily do it by starting with this one here and stripping it down.
2) learning the layout of the 3rd party firmware - what bits need to where
3) writing some mongoose code that will download and flash 3rd party firmware to the right places

you need to be careful not to write to the same location where code that is doing the writing is currently running from, so some additional relocation may be required, but it's doable if the 3rd party firmware is less than 1MB in size.

> 
> If we could somehow get the documentation, or somewhere start, would be awesome to get right into.

i'll give you some pointers later (on mobile now) 

> 
> Thanx!

jurban85, 2020-05-14:
Just wanted to give you guys an update. I was able to flash the Shelly via OTA on the next day. Seems like something the fw was not available for some strange reasons. Also the GitHub Page was not accessible on this day. Afterwards it worked just fine. Thanks for your effort and thank you for making this easy way of flashing this devices!

miamilabs, 2020-05-15:
Wondering if it will work for rgbw2 controller?
I saw people selling already working HomeKit rgbw2 and some one told that they are using this OS.

rojer, 2020-05-16:
ok, i'm going to close the issue here.
we can continue discussion about alternative firmware on https://github.com/arendst/Tasmota/issues/8436.

closed 2020-05-16 by rojer