mongoose-os-apps/shelly-homekit #70: Inter-device control actions

freddair, 2020-05-13:
Hi,

I have two separate HK homes in one multi-party-house in a single network.

In one home I use a Shelly 1 as a door opener connected to the intercom.

I would like to have a dummy device in the other home that mirrors the lock mechanism so that I can use it to open the front door from both HK homes.

Is it possible to set this up using http commands or any other way?

I tried it with HAA firmware where I can send http commands but currently it’s not possible to achieve this... maybe with Shelly-HomeKit or as a new feature?

Thanks a lot!

andyblac, 2020-05-13:
hi,

take a look at https://apps.apple.com/us/app/pushcut-shortcuts-automation/id1450936447 it maybe able to help you achieve what your after.

freddair, 2020-05-13:
Thanks, I‘ll have a deep look! :)

Although Shortcuts didn’t allow my neighbor to use my lock mechanism and/or it was very unreliable depending on what I tried. We need something independent from unreliable home hubs (Apple TV / HomePod...)

andyblac, 2020-05-13:
from my understanding pushcuts, links homekit to IFTTT

freddair, 2020-05-15:
Needs to run on a dedicated iOS device, seems not to be a good idea to me...

freddair, 2020-06-16:
@rojer: any thoughts on this? Here is what I mean: Direct Device to Device communicaton (DDD) in the original shelly firmware.

https://www.shelly-support.eu/lexikon/index.php?entry/71-kommunikation-shelly-zu-shelly-direct-device-to-device-communicaton-ddd/

andyblac, 2020-11-30:
@freddair this may help you on this one, https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#API

freddair, 2020-11-30:
Thanks @andyblac, meanwhile I use Shelly-Homekit for the lock and I could use a virtual lock via homebridge or an existing HAA device to send http commands - nice idea isn't it 😇

timoschilling, 2022-12-14:
To use the Shelly Stock DDD communication, your Shelly's must be on the same network. I assume that your neighbour isn't on the same network, so DDD never will work for you.

If your wifi can reach his apartment, you can give him a Shelly connected to your wifi and HomeKit home, configured as stateless input an configure a automatisation that triggers the door.

I will close this as the feature request can never solve this issue, when not on the same network and if on the same network, there are other solutions.

BTW: DDD is still requested in https://github.com/mongoose-os-apps/shelly-homekit/issues/1058, but more direct, so I will keep #1058 for that topic.

closed 2022-12-14 by timoschilling