mongoose-os-apps/shelly-homekit #71: shelly 1PM can't update OTA

Tomer51m, 2020-05-16:
Hi,
I am trying to update my Shelly 1PM but for some reason it will not update. 
I tried reseting few times and changing wifi networks.

It looks like it's trying and the LED is blinking, but at the end of the process I go to the shelly IP address and see the old firm working perfectly.

This is the response I get when trying to update:
`{"status":"updating","has_update":false,"new_version":"20200312-124223/v1.6.1@112262fd","old_version":"20200312-124223/v1.6.1@112262fd"}`
new version and old version are the same, but no homeKit support 🤷‍♂️

any help with be appreciated.

andyblac, 2020-05-16:
what OTA url are you using ?

Tomer51m, 2020-05-16:
http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1PM.zip
Changing the A.B.C.D to the IP of the shelly on my network.

andyblac, 2020-05-17:
hmm, the only issue i have had when try to use OTA, is if the shelly does not have access to a working internet / DNS. make sure your shelly has outside access, some people firewall their IoT network.

Tomer51m, 2020-05-17:
How can I check if it has internet access?
I do have firewall. I will try to disable it.

andyblac, 2020-09-15:
is this still an issue ?

andyblac, 2020-09-19:
@rojer OP has not responded maybe close it ?

rojer, 2020-09-19:
yes, @Tomer51m please reopen if it's still an issue

closed 2020-09-19 by rojer