mongoose-os-apps/shelly-homekit #72: Shelly RGBW2 Support

miamilabs, 2020-05-16:
Dear Community,

i saw there are few people which managed to run mongoose OS on RGBW2 controller and charging Double price for the controllers, i would even pay this once as i already bought my controllers for my home and don't send them back as i love this product and want to support the developer.

Are there any plans to bring support for Shelly RGBW2 or maybe some open source guide how to use mongoose-os with RGBW2 Controller?

Thanks you all.
Stay safe.

andyblac, 2020-06-22:
I have 2 of these, atom I am using them with Homebridge plugin, let just say it pretty unless, colours are all wrong, and you have zero control over white light, I hope when support is added to this firmware it works better :)

josetavares83, 2020-10-30:
Hey.. Please add support for RGBW devices 🙏🙏🙏

andyblac, 2020-10-30:
Patients young padawan 😄 , @rojer is really busy on this project, and he only has so much free time. it will come, we just have to be patient.

miamilabs, 2020-11-09:
> Hey.. Please add support for RGBW devices 🙏🙏🙏

As long the software is in build, i am currently using  RavenSystem / haa.
If you need this urgently as me, you can reflash your RGBW2. Make sure you pull a backup of your current OS.

andyblac, 2020-11-09:
too complicated to configure, plus as far as i know you can only flash via serial

andyblac, 2020-11-25:
just a heads up, for those using RGBW2 and Homebridge Shelly's new firmware 1.9.0 update fixes the awful crashes the devcie was getting.

RedAnon, 2020-12-07:
Would be real nice to get at least something simple like multiple white LED strips working first. I bet that's what most people will use.

[image]

RGB/RGBW shenanigans can wait a bit. 😄

andyblac, 2020-12-07:
haha, nice try 😆

TimE90, 2021-02-03:
Hey, thanks for these firmwares! Are there any news on the RGBW2 support?

timoschilling, 2021-02-03:
I'm currently working on the RGB version, an RGB+W version and 4xW version will follow.

BlancoDeMallloca, 2021-02-23:
@timoschilling: Thanks for your geformt with this issue. Andy updates on this? I am not able to code but would be willing to support the testing of RGBW2 functions.

andyblac, 2021-02-23:
> @timoschilling: Thanks for your geformt with this issue. Andy updates on this? I am not able to code but would be willing to support the testing of RGBW2 functions.

it's coming along, we still need to do a few more things, sorry can't give you a timeframe.

timoschilling, 2021-02-23:
I'm 99% finished with rgb support, rgbw and 4xW will follow in serrate PRs / releases.

bkleef, 2021-02-23:
@timoschilling about the 4xW, is it possible to exclude some channels from HomeKit?
In my situation, only 3 of the 4 channels are connected, causing 1 of the 4 lamps to show unwanted in HomeKit.

timoschilling, 2021-02-23:
Yes can chose between 1, 2, 3 or 4 single channels

timoschilling, 2021-02-24:
A preview version for the RGB support is attached to the PR https://github.com/mongoose-os-apps/shelly-homekit/pull/529.

miamilabs, 2021-03-08:
Thank you so much! Finally i can remove those VMs. Will test it next days when i arrive back to Europe.

meistermolli, 2021-03-23:
Hi, is your multiple white Support the same as the CCT ?

Dimmalble and from warm white to daylight white?
like this:
warmweiß (2.700 K) - tageslicht (6.500 K)

timoschilling, 2021-03-23:
We can support multiple things, for the first 4xW will only support 4 channel white, but we can later add WW/CW or WW/NW/CW too.

u20p17, 2021-03-30:
Hi,
at the moment I am using a RGBW2 with original firmware 1.10.1 but have an issue that the mixed colours do not look right (specially yellow/orange). I am using a RGBWWCW 5in1 strip, but only connected RGBWW to the Shelly. Do you think that this software could solve the problem? Will there be a feature in the future to make a colour calibration? When do you expect to release a firmware which supports RGBW? 

Thank you guys for your great work!

andyblac, 2021-03-30:
give the beta of RGB a try, regarding RGBWWCW as the he RGBW2 only suports 4 channels, you will never be able to fully use that strip with it you will have to choose between WW or CW, or when WHITE control is added to this firmware you could in theory use 2 RGBW2's 1 for RGB and 1 for WW / CW.

andyblac, 2021-03-30:
atm the beta firmware only supports RGB channels, we will add RGBW, RGB + W, W x 4, in future.

u20p17, 2021-04-01:
Hi,
just checked how I can update the firmware of the rgbw2, but couldn't find the firmware on http://rojer.me/files/shelly/ ... how can I flash this firmware?

timoschilling, 2021-04-01:
It's not official released yet, but here is the latest build, you can flash it with the [flash script](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Flashing#Script) via `--local-file`.
[rgbw2.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6245495/rgbw2.zip)

u20p17, 2021-04-01:
Hi,
I now flashed the firmware but the result is more or less the same than with the original firmware--> if I select yellow, it looks more like green...
[image]
[image]

Furthermore, I have a normal light switch for turning the light on/off. If I select Toggle as input mode I can turn it off and on, but in the on state the light has a strobe effect (so it turns on/off quite fast). Whats the problem there?

timoschilling, 2021-04-01:
You post the log output during a state change

andyblac, 2021-04-01:
@u20p17 regarding colour, sound like a led strip limitation, not all strips are equal.  yellow should give 100% RED and GREEN, can you confirm this ?

u20p17, 2021-04-01:
Hi, attached the log file, where I activated the switch for about 3 secounds and then deactivated again. It started flashing like before:
[log.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/6245984/log.txt)

@andyblac : If I turn on red it looks perfektly like red. If I turn on green, it looks also perfekt. But a combination of them (which should result in yellow) looks like lightgreen... I do not understand this...

andyblac, 2021-04-01:
> @andyblac : If I turn on red it looks perfektly like red. If I turn on green, it looks also perfekt. But a combination of them (which should result in yellow) looks like lightgreen... I do not understand this...

it is a single led chip or old type 3 seperates ?

timoschilling, 2021-04-01:
@u20p17 the log doesn't match your UI screenshot, you UI says Hue: 60, your log says always Hue: 37.

u20p17, 2021-04-01:
@andyblac : it is a 5in1 chip --> [link](https://de.aliexpress.com/item/4001115198583.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=20120359594&aff_fcid=d8b6d83eddb24b77a6f6ae7647e09a26-1617305603137-01147-_mNZDRXt&tt=MG&aff_fsk=_mNZDRXt&aff_platform=default&sk=_mNZDRXt&aff_trace_key=d8b6d83eddb24b77a6f6ae7647e09a26-1617305603137-01147-_mNZDRXt&shareId=20120359594&businessType=ProductDetail&platform=AE&terminal_id=31c6fd77eedd476199a125683a6179e3)

@timoschilling : Sorry, I changed the colour in the mean time. Now a log with the yellow colour: 
[log.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/6246102/log.txt)

andyblac, 2021-04-01:
and siri test ?

u20p17, 2021-04-01:
when I tell siri to turn on the yellow light it turns the light on but it looks lightgreen again.

andyblac, 2021-04-01:
> > @andyblac : If I turn on red it looks perfektly like red. If I turn on green, it looks also perfekt. But a combination of them (which should result in yellow) looks like lightgreen... I do not understand this...
> 
> it is a single led chip or old type 3 seperates ? also, i know it is daft question but.... do you have wires in correct chanels ?
> 
> ie ask siri for red does it show red?, then ask for green.

try that

timoschilling, 2021-04-01:
The log looks good now.
```
4672865163 shelly_hap_rgb.cpp:194  state: on, brightness: 100, hue: 60, saturation: 100
4672870807 shelly_output.cpp:71    Output 1: 1.000000 (switch)
4672875928 shelly_output.cpp:71    Output 2: 1.000000 (switch)
4672880273 shelly_output.cpp:74    Output 3: OFF (switch)
```
1=R, 2=G, 3=B, so it looks fine to me

I would say your strip is bad or need special color coding.

Maybe you can try a hue value between 60-120 to get a better green.

u20p17, 2021-04-01:
if i ask sire for red I get red, for blue I get blue and for green I get green. So it looks fine like @timoschilling  also mentioned.

If I use a hue value of about 40 it looks more like yellow, even if it looks more like warmwhite then.

andyblac, 2021-04-01:
> if i ask sire for red I get red, for blue I get blue and for green I get green. So it looks fine like @timoschilling also mentioned.
> 
> If I use a hue value of about 40 it looks more like yellow, even if it looks more like warmwhite then.

sounds like a hardware strip issue.

u20p17, 2021-04-01:
i am from austria. do you have a recommenation for austria/germany too?

andyblac, 2021-04-01:
> i am from austria. do you have a recommenation for austria/germany too?

ask @timoschilling he's from Germany. he'll have better idea.

timoschilling, 2021-04-01:
I have an rgb from AliExpress. @jobl1978?

jobl1978, 2021-04-02:
> i am from austria. do you have a recommenation for austria/germany too?

Hi compatriot ;-),
@u20p17 i currently use this strip for my tests RGBW firmware extension tests. It has also a little bit of greenish but it's ok and bright. 

https://www.amazon.de/Streifen-Farbwechsel-300LED%60s-Ratgeber-Warmweiß/dp/B07R8WFBQZ

Hue: 60% Saturation&Brightness: 100%

[image]

jobl1978, 2021-04-02:
@u20p17 Some theory to understand better what's going on...

In the picture you can see many colors but the RGB colorspace which the RGB Strip can show is only inside the triangle. As you can see hitting yellow there is much harder then hitting red, blue or green. The three corners of the triangle represents full red, green and blue. If yellow is not exactly between the red and green corner of the triangle you will get a greenish or  reddish light. Those corners are never at the same position between different strips. Some strips has those corners calibrated to achieve better results and the cheaper once are shifted off more and does not hit the correct point for yellow. This can only be prevented if strips are calibrated or rgb controller itself provides a software based calibration where the correct position of the red, green, blue and white will be teached in. After that every RGB value will be transformed to the correct position to calc the correct output signal (PWM) for every channel.

[image]

u20p17, 2021-04-02:
thanks for the description. would it be possible to implement a colour calibration also in the RGBW2 firmware in the future?

for now I also bought a new led strip from [led-shop-austria](https://www.led-shop-austria.at/index.php?main_page=product_info&cPath=16_113_208&products_id=1224 ). Not the cheapest one, but lets give it a try.

mrmaximas, 2021-04-04:
What about RGBW2 in WhiteOnly Mode? Thanks!

jobl1978, 2021-04-04:
> What about RGBW2 in WhiteOnly Mode? Thanks!

RGB+W and WWWW will be the next thing coming up, but first we will release RGB/RGBW support

mrmaximas, 2021-04-04:
> > What about RGBW2 in WhiteOnly Mode? Thanks!
> 
> RGB+W and WWWW will be the next thing coming up, but first we will release RGB/RGBW support

W only (CH4) will be good too. Thanks!

jobl1978, 2021-04-04:
WWWW means up to 4 channels, you can deactivate unused channels

mrmaximas, 2021-04-04:
deactivate unused channels - good idea!

mrmaximas, 2021-04-04:
i'm using six rgbw2 with old light fixtures via CH4 only )

me61ic, 2021-04-20:
> It's not official released yet, but here is the latest build, you can flash it with the [flash script](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Flashing#Script) via `--local-file`.
> [rgbw2.zip]([image]rgbw2.zip)

Any idea why I get this error message when trying to flash my RGBW2 with the beta homekit firmware given here

timoschilling, 2021-04-20:
@me61ic I your shelly in white or in color mode? I must be in color mode. Try to update your flash script, by pull it via git. Or try http://192.168.68.118/ota?url=http://rojer.me/files/shelly/2.9.0-alpha2/shelly-homekit-ShellyRGBW2.zip

andyblac, 2021-04-20:
> Any idea why I get this error message when trying to flash my RGBW2 with the beta homekit firmware given here

bug found in script, plesase try this version, report back if works ok
[flash-shelly.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6343328/flash-shelly.zip)

andyblac, 2021-04-20:
> @me61ic I your shelly in white or in color mode? I must be in color mode.

for script not true, colour mode can be anything.

andyblac, 2021-04-20:
ok found additional bug, this one should fix flashing from stock to hk in any color mode with local-file
[flash-shelly.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6343407/flash-shelly.zip)

me61ic, 2021-04-20:
> ok found additional bug, this one should fix flashing from stock to hk in any color mode with local-file
> [flash-shelly.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6343549/flash-shelly.zip)

now I get this:
[image]

andyblac, 2021-04-20:
> > ok found additional bug, this one should fix flashing from stock to hk in any color mode with local-file
> > [flash-shelly.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6343549/flash-shelly.zip)
> 
> now I get this:
> [image]

where did you get zip file from ? 

please this zip file: 
[shelly-homekit-ShellyRGBW2-2.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6343725/shelly-homekit-ShellyRGBW2-2.zip)

andyblac, 2021-04-20:
try this script version should check if zipfile is a vailed flash file
[flash-shelly-2.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6343768/flash-shelly-2.zip)

me61ic, 2021-04-20:
> flash-shelly.py 3.zip

I get the same
[image]

andyblac, 2021-04-20:
[flash-shelly.py 3.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6343979/flash-shelly.py.3.zip)
try that one.

me61ic, 2021-04-21:
> @me61ic I your shelly in white or in color mode? I must be in color mode. Try to update your flash script, by pull it via git. Or try http://192.168.68.118/ota?url=http://rojer.me/files/shelly/2.9.0-alpha2/shelly-homekit-ShellyRGBW2.zip

I was not able to flash any of the given firmwares through Phyton locally, in fact I was always getting flash shelly version 2.7.0 even though I was deleting the old zips and copying the new ones in the Tools folder. 

Then I tried this other option that timoschilling sent and I was able to flash that firmware to my RGBW2 but now in order to be able to see the shelly homekit settings I need to set my wifi to the shelly RGBW2, connect to its IP directly in its own WiFi and then I can see the Homekit settings, but it does not let me to connect it to my house wifi, once I enter the password, I hit save, takes me to another screen but does not seem to save it to the shelly RGBW2, I can not see it in my house wifi and only way to connect to it again is to enter its own shelly wifi and IP again. Any ideas what I am doing wrong?

[47 more replies]

closed 2022-02-03 by timoschilling