mongoose-os-apps/shelly-homekit #74: Dummy switch creation

Jaguaza, 2020-05-17:
Proposal:
Create an option in the web interface to create dummy switch.

Objective:
Apart from having the shelly device, having a fake device to be able to add it to automations.

Example:
Automations by a geolocation trigger for connection and disconnection of the alarm in homekit ask for authorization to carry it out in a push notification.

If you create a rule that says when I leave home, turn off and turn on the dumy switch.
And a second rule that says when the switch turns on it turns on the alarm.

Since the switch is not a homekit security device, it does not ask for geolocation authorization.
And since the switch is supposed to be turned on you do not ask for authorization to connect the alarm.

These dummy switche devices are widely used in Homekit.

rojer, 2020-05-17:
it would actually be easy to do but i'll need more people to confirm its usefulness.

andyblac, 2020-05-17:
why not just use Shortcuts, to enable / disable the alarm ?

This is what i do.

[image]
[image]

Jaguaza, 2020-05-17:
> en realidad sería fácil de hacer, pero necesitaré más personas para confirmar su utilidad.

Ok I don't think that's a problem.

It is a widely used device.

Jaguaza, 2020-05-17:
Good Morning.

It was just a simple usage example.

Jaguaza, 2020-05-17:
> ¿Por qué no usar atajos para activar / desactivar la alarma?
> 
> Esto es lo que hago, y no requiere confirmación del usuario maestro.
> 
> [image]
> [image]

Good Morning.

It was just a simple usage example.

andyblac, 2020-05-17:
ah ok...

xuleto, 2020-05-17:
Estaría genial que implementaseis esta opción a parte de utilizarlo para el sistema de seguridad se podría utilizar él multitud de situaciones. Estaremos atentos!

divemasterjm, 2020-05-17:
Would be a nice enhancement +1

edubox, 2020-05-17:
+1 I think that this will be useful. Not only in the case that @Jaguaza mentions, but other ways. Many systems have it.

jjulveval, 2020-05-17:
+1.  It is a great idea and very useful

muval, 2020-05-17:
Good morning, the updates you are putting in the firm for the shellys are very good, it would be interesting the option of having a dummy switch also when possible.

jesegura, 2020-05-17:
+1. I join the partner's proposal, I find it interesting to have that dummy switch

danimart1991, 2020-05-17:
This would be perfect por HomeKit users that look for easy options as possible.

Borja90, 2020-05-17:
+1 
It would be great to have this implemented giving lots of posibilities. Thanks!

ElBielas, 2020-05-17:
Hi guys I think it's a great idea the dummy option !!!

alcar21, 2020-05-17:
+1 
i too am very interested

rojer, 2020-05-17:
wow, wow, ok, i get it :)

rojer, 2020-05-17:
so, the plan is to add more ["sw" sections](https://github.com/mongoose-os-apps/shelly-homekit/blob/master/mos.yml#L20). currently we only have one for each physical switch but we can have more. set gpio for them to -1 and modify code to ignore all the gpio activity but behave the same in all other respects (state changes, notifications).
we'll need to figure out the UI aspect as well, so we don't populate like 10 of them but allow adding and removing "virtual" switches as needed.

Jaguaza, 2020-05-17:
> entonces, el plan es agregar más [secciones "sw"](https://github.com/mongoose-os-apps/shelly-homekit/blob/master/mos.yml#L20) . Actualmente solo tenemos uno para cada interruptor físico, pero podemos tener más. establezca gpio para ellos en -1 y modifique el código para ignorar toda la actividad de gpio pero se comporte igual en todos los demás aspectos (cambios de estado, notificaciones).
> también tendremos que descubrir el aspecto de la interfaz de usuario, de modo que no rellenemos como 10 de ellos, sino que permitamos agregar y quitar conmutadores "virtuales" según sea necesario.

thanks

alfons1000, 2020-05-17:
I’m very interested!!

Pixel-Chris, 2020-05-19:
I think the optimal solution would be to integrate this to issue #9 !!!

Jaguaza, 2020-05-19:
> I think the optimal solution would be to integrate this to issue #9 !!!

+1

andyblac, 2020-09-21:
I guess this should be another easy one, now that code rewrite is done.

rojer, 2020-09-21:
indeed. but only after the roller shutter is done!

andyblac, 2020-12-06:
as stateless switch is now coded, this should be fairly easy to add ? could it be added for 2.7.0 ?

rojer, 2020-12-13:
not for 2.7.0, but it is much easier after recent refactoring effort. maybe soon.