mongoose-os-apps/shelly-homekit #75: FAN device request.

Jaguaza, 2020-05-17:
Petition:
Fan device.

Utilities:
For shelly controlled fans we can declare them as such even if their function is only on / off. Although possibly with new shelly devices you can add intensity etc. But for now with an on / off is enough.

rojer, 2020-05-17:
if anyone feels like doing this before i get to it, here's an example change - https://github.com/mongoose-os-apps/shelly-homekit/commit/1ef5e0c5dd5f1032905e8ef14d5c6b104b77f6bd

Jaguaza, 2020-05-17:
@rojer Sorry for disturbing you.
We have the fan with swicht Mode for on/off
[image]

Its not necesary either fan device because i think there is not a shelly device who can emulate a fan's remite control, It were a Nonsense.
We can Close this issue and if you want se can open other to plan the garage's device, because It Will be very used whith a shelly 1 by so many users.
[image]

Jaguaza, 2020-05-19:
@rojer ???

rojer, 2020-05-20:
i still think it makes sense to add a fan device.

Jaguaza, 2020-05-20:
> i still think it makes sense to add a fan device.

If you see it meaningfully, go against more devices better.  I can only say thank you.

I told Dimi if i could get contac whit you. I am spanish facebook's oficial support groups administrator.
As you think i can help you, Take on me.

felixmueller, 2020-05-20:
Just a quick thought: isn't it true that every Outlet in HomeKit can be set as a Fan in the device settings (Show as:)?

Jaguaza, 2020-05-20:
> Just a quick thought: isn't it true that every Outlet in HomeKit can be set as a Fan in the device settings (Show as:)?

Yes and no.  In switch you just change the icon, but the device is still a switch.  The Fan device in addition to an on and off can be associated with intensity or power of the fan spin.

andyblac, 2020-05-24:
just the Shelly's allow the ability to control the voltage ?, thought they where just on/off and controlling the speed of a fan requires a voltage regulator ?

rojer, 2020-05-24:
all characteristics of the fan service except "active" are optional. we will not be able to provide speed control, but that's fine.

andyblac, 2020-09-20:
is the really required when you can simply choose "FAN" in HomeKit switch type settings ?

[image]

rojer, 2020-09-20:
yep, i think that's enough

closed 2020-09-20 by rojer