mongoose-os-apps/shelly-homekit #77: Remove one of the two relays from Homekit

prrtzt, 2020-05-19:
Hi,

One of my Shellies only uses 1 of the relays, but both show up in Homekit.

I am wondering if it would be possible to add an option to hide one of the switches from Homekit?

Thanks!

col-tom, 2020-05-19:
Just move them to (dummy) room, e.g. _unused_

prrtzt, 2020-05-19:
That's what I've been doing, but it's not very clean :)

Jaguaza, 2020-05-19:
> Hi,
> 
> One of my Shellies only uses 1 of the relays, but both show up in Homekit.
> 
> I am wondering if it would be possible to add an option to hide one of the switches from Homekit?
> 
> Thanks!

Why don't you unify the relays in hk as a single device?

prrtzt, 2020-05-19:
If you do that, then you cannot turn the light on and off in a single click. When you click on the device, it shows two switches, including the unused one.

Same issue if you ask Siri. If I say, turn off Wardrobe, it answers "There are more than one, you can ask me to turn off a specific one, or all".

Overall, it' is not very clean.

ursusHU, 2020-06-03:
> Same issue if you ask Siri. If I ask "turn off the Bedroom", Siri will answer "There are more than one, you can ask me to turn off a specific one, or all".Did you try turn off bedroom lights ? It works well for me (shelly, plug (set as light), bedside lamp etc. is turned off)

freddair, 2020-06-07:
You can “group” devices together so that they act as a single device both for Siri and the Home app.

andyblac, 2020-06-07:
shelly 2.5 is a single device with 2 switches, grouping won't help.

freddair, 2020-06-07:
What dou you mean by “it won’t help”? I just tried it out and it works as expected. You only see one switch. I included screenshots to show what I’ve done.[image][image][image][image][image]

andyblac, 2020-06-07:
the OP wants to hide the 2nd switch, as it not used, how does grouping help ?

freddair, 2020-06-07:
The second switch is hidden within the group, which is shown as a single switch. It’s just a possibility, everyone can decide for their own if it suits their needs.

Jaguaza, 2020-06-07:
I do not see it necessary, having the grouping of devices in homekit, it is a utility that right now I do not see necessary.

rojer, 2020-07-03:
this is possible, switches can be disabled and will not appear in homekit, we just need to expose a control for this.

andyblac, 2020-09-19:
this is now completed in https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.1.7, just select "Disabled" in switch type.

[image]

andyblac, 2020-09-19:
@rojer think this one can be closed as well.

rojer, 2020-09-19:
yep, this one is done, released in 2.1.7

closed 2020-09-19 by rojer