mongoose-os-apps/shelly-homekit #79: Switch 1 : no control without direct WiFi access

Whiterabbit64, 2020-05-22:
Hello.

My HomeKit hub is my HomePod so all my homekit accessories are accessible even if I'm not connected to my wifi. Except the Shelly 1. Is that a normal behavior ?

rojer, 2020-05-22:
Shelly only has wifi access

Whiterabbit64, 2020-05-22:
Yes I know, but I use my HomePod as a hub to remotely control all my home kit devices whern I'm away from home.
So as the Shelly is integrated in my HomeKit setup whay can't I control it remotely through the hub ?

rojer, 2020-05-22:
i see. i don't know, tbh.

andyblac, 2020-05-22:
@Whiterabbit64 shelly still requires a link between HomePod and Shelly, and as shelly only has wifi,  it will loose link if WiFi drops.

Whiterabbit64, 2020-05-22:
I don’t get the point. When I leave home, my Homekit equipment and my wifi remain powered.
So I can control remotely all of them. The Shelly is still connected to the Wifi and the HomePod.

The only device that I can’t control is the Shelly because in home application it appears offline.

Whiterabbit64, 2020-05-23:
Problem solved.
The issue was with the HomePod. I’ve moved it to a less complex and more compatible Wifi network.

closed 2020-05-23 by Whiterabbit64

andyblac, 2020-05-23:
glad you got it sorted :)