mongoose-os-apps/shelly-homekit #80: Switch 1 and HomePod instructions

Whiterabbit64, 2020-05-22:
Hello.

I can "drive" my Switch 1 with "Hey Siri" from my iPhone and from the Home application but each time I try with Hey Siri from my HomePod, the Switch 1 is becomes unreachable for few minutes.
Is that normal ?

andyblac, 2020-05-22:
no, should be no difference however you drive the drive, vi home app, hey siri, etc, is you HomePod the Hub ?

Whiterabbit64, 2020-05-22:
> no, should be no difference however you drive the drive, vi home app, hey siri, etc, is you HomePod the Hub ?

Yes the HomePod is the hub.

I suspect an issue with WiFi.

My WiFi network is made of several SSID:
- 2 dedicated for 2.4GHz devices (1 with normal DTIM and 1 for Shelly devices with a modified DTIM)
- 1 for 5GHz devices
- 1 dual band.

That's a total of 4 SSID (Cisco controlled AP). It's a bridged network, so they IP range is the same for all devices and the DHCP server is my Internet router.

The HomePod has a bug with dual band SSID so it's connected on my 5GHz SSID.
My iPhone is connected on normal 2.4GHz SSID
My iPad is connected on 5GHz SSID
The Shellys are connected to the DTIM modified 2.4GHz SSID

The device on which I have less issues is the iPhone.
The device on which I have most of the issues is the HomePod.
Ipad is in between :)

Whiterabbit64, 2020-05-23:
Problem solved.
I’ve moved the HomePod to a less complex Wifi network. Now everything works fine.

closed 2020-05-23 by Whiterabbit64