mongoose-os-apps/shelly-homekit #81: Unable to flash Plug S

Cibernello, 2020-05-22:
I was able to successfully flash Shelly 1 with 1.9.1 firmware. It works just fine. When I tried to do the same with Plug S it did not work. I run the update procedure (as I did with Shelly 1) but after the reboot of the device I still see the stock firmware. Any suggestion? Thanks

IvanGitter, 2020-05-27:
Hi, I have got the same problem with Plug S. I tried to update the firmware to version 2.0.1 with the following URL:

`http://192.168.1.14/ota?url=http://127.0.0.1:8887/build/shelly-homekit-ShellyPlugS.zip`

 The response from the Plug is reported below

`{"status":"updating","has_update":false,"new_version":"20200309-104506/v1.6.0@43056d58","old_version":"20200309-104506/v1.6.0@43056d58"}`

Any clue?
Thanks a lot.

andyblac, 2020-05-27:
@IvanGitter try using the local ip and not the loopback IP of your PC

IvanGitter, 2020-05-27:
Thanks @andyblac but it does not work. I tried with the URL reported below and it worked. So there is something in my pc/lan/... that prevents the plug to update the firmware.

`http://192.168.1.14/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyPlugS.zip`

andyblac, 2020-05-27:
glad you got it flashed, it does require your PC to be running a web server so it can host the file.

Cibernello, 2020-05-29:
Unfortunately I still cannot flash it.

I used the the url 
http://10.0.0.81/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyPlugS.zip

And I got this message

{"status":"updating","has_update":false,"new_version":"20200309-104506/v1.6.0@43056d58","old_version":"20200309-104506/v1.6.0@43056d58"}

When plug s reboots I still see the old firmware. I followed the same process for Shelly 1 and it went perfectly. Plug S douse not update. Why? I am sure the IP address of the device is correct... please help me

andyblac, 2020-09-30:
is this still an issue ?

Cibernello, 2020-09-30:
No, I was finally able to flash the firmware with one of the latest updates 

Cheers

Marco

Inviato dal mio incredibile iPhone 11 Pro

andyblac, 2020-09-30:
thanks for letting me know, I'll close ticket.

closed 2020-09-30 by andyblac

AdriaanBakker74, 2022-09-28:
I did run into the same problem. Unable to update