mongoose-os-apps/shelly-homekit #82: Shelly Access Security

marmitol, 2020-05-24:
Hello.
It's not an Issues, it's a security request for future versions. It would be nice if the management could be accessed via https, username and password.

It's a suggestion.

Thank you very much,

Whiterabbit64, 2020-05-24:
> Hello.
> It's not an Issues, it's a security request for future versions. It would be nice if the management could be accessed via https, username and password.
> 
> It's a suggestion.
> 
> Thank you very much,

I confirm, it’s a must have feature.

andyblac, 2020-05-24:
can i ask why ?, do you guy's exposed port 80 on your IoT VLAN to the web ?

Whiterabbit64, 2020-05-24:
> can i ask why ?, do you guy's exposed port 80 on your IoT VLAN to the web ?

Because a WiFi network can’t be secured at 100%. So if an attacker goes through, then it’s better to have a second security. So securing Shelly Homekit with https and user/pass would be great.
If not, once the Wifi is hacked, then opening the garage doors is not an issue 😁

andyblac, 2020-05-24:
ok, noted.

andyblac, 2020-05-24:
then also, maybe an option to allow the user to chose the port that webpage is hosted would also help ?, port 80 is very common....

rojer, 2020-05-24:
HTTPS with a proper certificate is impossible. self-signed is theoretically possible but wouldn't make much sense. authentication - agree, we should have it.

Jaguaza, 2020-05-24:
I think with a username - password and serious port choice the most recommended. yes.

andyblac, 2020-05-24:
that what i was thinking HTTPS there would not be enough space on device, but HTTP with login should be enough.

Jaguaza, 2020-05-24:
> que lo que estaba pensando HTTPS no habría suficiente espacio en el dispositivo, pero HTTP con inicio de sesión debería ser suficiente.

Other spur type, tasmim fimware use http. I don't think it's a big problem. Even the user is default which is one less security option.

epiphany11, 2020-10-06:
HTTP Access would be fine (username + password) like shelly original firmware.

sebastianheise, 2020-11-25:
Only somewhat related to this ticket, but: 
Currently WPA2 encryption for Wifi is supported. 
Is it possible/useful to also support WPA3 as well?

nachumk, 2020-12-16:
+1
This is pretty important for me as well - access to one's WiFi would enable someone to control secure devices that are implemented with Shelly1. Whereas other HomeKit devices can't be controlled via WiFi alone. Locking down access to these devices with a secure HTTPS login would be of great use.

andyblac, 2020-12-16:
we would need an API token aswell if this is done.

timoschilling, 2020-12-17:
I we use HTTP Basic Auth for the UI we can use it for the API as well, `http://<user>:<password>@<host>/rpc/...` should work.

andyblac, 2020-12-17:
> If we use HTTP Basic Auth for the UI we can use it for the API as well, `http://<user>:<password>@<host>/rpc/...` should work.

no way!!!, i will use clear text password in an API request.

timoschilling, 2020-12-17:
What's the difference between an clean text password and an api key? Both is readable if we don't have https and the user:password will be encoded as an token and send via header.

andyblac, 2020-12-17:
> What's the difference between an clean text password and an api key? Both is readable if we don't have https and the user:password will be encoded as an token and send via header.

i do not trust giving the clear text password to another manufacure device (ie stock shelly fw aka Button1 to control a homekit shelly). that device "could" have access to other cloud servcies and other remote services.

with an API Token have can not get acces to the webUI of the shelly homekit device,

andyblac, 2020-12-17:
the API token can be any random key, that can be changed at any time by the user.

andyblac, 2020-12-18:
i'll leave this open as it is a good request.

trendchiller, 2021-01-28:
+1 from me for password protected HTTP(s) GUI

rojer, 2021-01-28:
so let me outline the solution as i see it. i think we can use HTTP digest authentication, for both serving contents and RPC.
RPS already supports digest auth, it just needs to be enabled. and when used over HTTP/WS, it takes credentials that browser sends, so we should be all good.
we just need to enabled digest auth for serving the index page, configure RPC auth and tighten ACLs to require authentication for calls such as SetState or SetConfig. in fact, authentication should be required for everything, except a very limited set of calls: Sys.GetInfo should be allowed without auth, and whether Shelly.GetInfo should require auth is up for discussion. we may start with leaving it open. it is used by the flasher script to check versions, so i think at least limited information should be accessible without authentication. sensitive stuff can be moved to a different, authenticated call and UI modified to use it.

timoschilling, 2021-01-29:
I think we should not enable it for the http UI, only for websocked and rpc. Then the UI can be served and the UI can ask the User for the credentials

rojer, 2021-01-29:
sure, if we can have a nice login form that's better UX.
but if we can find a way to facilitate built-in digest auth so we don't need to worry about generating tokens, cookies, etc, that'd make things easier.
i don't know if it's possible to trigger digest auth process for a background load, maybe it is.

timoschilling, 2021-01-29:
> i don't know if it's possible to trigger digest auth process for a background load, maybe it is.
The reason why web sockets didn't need CORS is why they didn't inherit the cookie or basic / digest auth from the web page, like it happens by xhr requests. So there is a need that the user must give the password to the js app.

rojer, 2021-01-30:
we send CORS headers with our responses though, it should be fine.

rojer, 2021-04-10:
FYI, i've been laying foundations for this in the background: https://github.com/mongoose-os-libs/rpc-common/commit/30294b862bc7bd33d30556b2777005dfd1130254 https://github.com/mongoose-os-libs/rpc-common/commit/f948357dde68e67bcf8347b9a9a49468c6ee5c4c https://github.com/mongoose-os-libs/rpc-common/commit/0c4abc2a173b29797a1f57a421317131bc692c40 https://github.com/mongoose-os-libs/rpc-common/commit/dfbf4a083198aeccbc1415c64011b19d30eb10c7
https://github.com/mongoose-os-libs/mongoose/commit/bc084dad9a9c35145fa7001c47402d733c0353f3

we can now have HTTP digest auth on UI load that gets transparently applied to RPC channels.
i have password protection working in my test setup, just need to add UI support for setting and changing password, reset, etc

andyblac, 2021-04-10:
> FYI, i've been laying foundations for this in the background: [mongoose-os-libs/rpc-common@30294b8](https://github.com/mongoose-os-libs/rpc-common/commit/30294b862bc7bd33d30556b2777005dfd1130254) [mongoose-os-libs/rpc-common@f948357](https://github.com/mongoose-os-libs/rpc-common/commit/f948357dde68e67bcf8347b9a9a49468c6ee5c4c) [mongoose-os-libs/rpc-common@0c4abc2](https://github.com/mongoose-os-libs/rpc-common/commit/0c4abc2a173b29797a1f57a421317131bc692c40) [mongoose-os-libs/rpc-common@dfbf4a0](https://github.com/mongoose-os-libs/rpc-common/commit/dfbf4a083198aeccbc1415c64011b19d30eb10c7)
> [mongoose-os-libs/mongoose@bc084da](https://github.com/mongoose-os-libs/mongoose/commit/bc084dad9a9c35145fa7001c47402d733c0353f3)
> 
> we can now have HTTP digest auth on UI load that gets transparently applied to RPC channels.
> i have password protection working in my test setup, just need to add UI support for setting and changing password, reset, etc

how will this work for rpc and API requests ?

rojer, 2021-04-11:
Digest authentication is widely supported. `curl --digest -u user:pass`, for example.

timoschilling, 2021-04-11:
That’s nice I will start working on the Frontend implementation

rojer, 2021-04-11:
the bad news is that modern version of the digest auth, using sha256 hash, is not supported by browsers, they only support md5 ([chrome issue](https://bugs.chromium.org/p/chromium/issues/detail?id=1160478), [ff issue](https://bugzilla.mozilla.org/show_bug.cgi?id=472823)).
i was suprised by that and currently thinking what to do. md5 is a very old technology, and though digest auth uses it in a way that is at least somewhat non-trivial to exploit, it's not great. not great, but very convenient to use, very little effort needed to implement, no custom cookies or headers, browser will manage it.

andyblac, 2021-04-11:
how will this work for script, i was hoping you implement a token type type thing, so we don't have to keep typing username / password.

rojer, 2021-04-15:
this should be all you need - https://docs.python-requests.org/en/master/user/authentication/#digest-authentication

rojer, 2021-04-15:
for simplicity, we'll start with just one user with full access to everything.
RPC will be password protected. i am debating whether to allow some basic information without authentication or just protect everything. i'd like to keep everything closed for simplicity, but we may want e.g. name in the open just to be able to request credentials for a particular device with a name rather than just device ip address or mac / id.

rojer, 2021-04-24:
this should be working now, try 2.9 alpha3 - https://github.com/mongoose-os-apps/shelly-homekit/issues/617

timoschilling, 2022-02-04:
Can this issue be closed?

Whiterabbit64, 2022-02-05:
I think so 👍

closed 2022-02-05 by timoschilling

q20, 2023-05-17:
So, hear me out, chaps. How about _disabling_ the web interface after initial configuration of WiFi and HomeKit setup code? Each Shelly device then has its own combination of physical button presses to enable the web interface, allowing for reconfiguration. This way, once a device has been provisioned for HomeKit, it's either succesfully adopted into Apple Home (hoorah) or requires reconfiguration (cryptic button presses).
This negates any need for SSL.
An ESP HomeKit project offers exactly this sort of setup, so all credit goes to them for the idea:

https://github.com/RavenSystem/esp-homekit-devices/wiki

PS: how about also generating an adoption QR code in the current interface, before it's disabled?   😉

jactmp, 2023-05-17:
Hi, 
User's opinion: Always on access to web configuration interface (optionally guarded by password) is crutial advantage over RavenSystem's solution. Shelly modules are usually connected to home WiFi behind router/firewall so security is not an issue here. 
Please make it optional (if at all).

nrjohnstone, 2023-09-07:
Why does everyone think their home WiFi is amazingly secure? It's not about getting hacked externally as almost all compromises are usually from devices behind the network that make an outbound call to then download rootkits and other payloads. How many of you are blocking outbound calls ?? Probably no one because that is painful to administer.

Also, how many allow kids with devices to join your main WiFi network ? Do their friends know the password? How easy do you think it is for a compromised device to suddenly be "behind" your firewall and start sniffing packets on your network.

Because of the fact that there is no encryption on the Shelly HTTP UI, it's trivial to sniff the username/password if you are in the network and then start doing all kinds of malicious stuff, like flashing the shelly with a compromised version of the firmware etc...

The Shelly UI should really be protected by HTTPS, or at least be able to be disabled as q20 and then renabled with a magic physical button press etc... (which doesn't help for those Shelly devices that don't have buttons by the way ...)

5xPa, 2023-09-07:
I for one have blocked all IOT devices from the internet by a firewall rule. The Shelly homekit does try to get the time. I setup in my router a re route to send them to a local NAS running a time server.
I did have some Meross smart plug.  I found Meross smart plug were continually phoning home. No longer use them.

stultitiophobia, 2023-09-07:
> The Shelly homekit does try to get the time. ...
Which fqdn is used ?

5xPa, 2023-09-08:
I asked Timo if there was anyway of manually setting the ntp server or indeed asked to add this feature. He replied the time request was part of the mongoose operating system. The server was time.google.com 8.8.8.8

5xPa, 2023-09-08:
Here is the issue I raised https://github.com/mongoose-os-apps/shelly-homekit/issues/1241