mongoose-os-apps/shelly-homekit #83: Static ip address like the original firmware

Makaba05, 2020-05-31:
DHCP makes sometimes a problem when a device doesn’t response quick enough. So, a static ip-address option like at the original Firmware would be good.

rojer, 2020-05-31:
yeah, can be done. config already supports configuring it (wifi.sta.ip, wifi.sta.netmask, wifi.sta.gw), just need to be exposed via settings.

Jaguaza, 2020-05-31:
> yeah, can be done. config already supports configuring it (wifi.sta.ip, wifi.sta.netmask, wifi.sta.gw), just need to be exposed via settings.

Good!!!!

michaelbublat, 2020-08-12:
Is there any documentation on how to expose via settings?

smarthome-fan, 2020-09-18:
and a 2. way:
flash orignal firmware from shelly. set ip and save. flash shelly-homekit
 firmware. now you have a static ip too in this firmware ;-)

lucaspinelli85, 2020-11-16:
Can I edit the IP address from HomeKit firmware an set it static without come back to original firmware, set it, and return to HomeKit firmware? Because I loose the switch option when come back to original.

andyblac, 2020-11-16:
> Can I edit the IP address from HomeKit firmware an set it static without come back to original firmware, set it, and return to HomeKit firmware? Because I loose the switch option when come back to original.

not in this fw atm, mabe just reserve an IP in your router DHCP servers table.

healeydave, 2021-01-08:
+1 for static IP settings in HK firmware config screen.

simonfield76, 2021-01-08:
> > Can I edit the IP address from HomeKit firmware an set it static without come back to original firmware, set it, and return to HomeKit firmware? Because I loose the switch option when come back to original.
> 
> 
> 
> not in this fw atm, mabe just reserve an IP in your router DHCP servers table.

Are you planning to put an option in the firmware in the next future? I do not like making ip reservation in my firewall

rojer, 2021-01-09:
definitely makes sense to have it but as usual - no ETA

rojer, 2021-02-14:
closing in favor of #501

closed 2021-02-14 by rojer