mongoose-os-apps/shelly-homekit #85: Firmware update fails

ursusHU, 2020-06-04:
Is there anything I can do if I stuck between the original and the shelly firmware? Shelly page doesn't load shelly homekit update page says not found. Is it possible to install something OTA on the unit? Or I have a piece of papaer holder?

rojer, 2020-06-04:
please provide more detail: what page doesn't load, what's the address? what error do you get? what is the "homekit update page" you are referring to?

ursusHU, 2020-06-05:
I made several updates today. Using the address given in the description. 2 Shelly 2.5 was updated successfully, 1 Shelly 1 was updated successfully, and unfortunately 3 failed. For the Shelly1 I used this link with the ip address of the Shelly (static IP was set in the Shellies)
http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip
The process starts, however after several minutes of waiting if I enter the given ip address the page was loading the Shelly interface without any functionality and stating that the device is not accessible. Afterwards I tried again the above mentioned link, to reflash it again but it was giving the answer not found. However now I checked the link and the Shelly is reachable through the Shelly homekit interface. Which could mean that somehow the browser cache was causing problems.
Unfortunately I tried to restart 2 units with the power cycle 5 on/off  method, so they are in AP mode now, but of course I'm not able to connect them through the Shelly app. Is there a way to get back the units to work? 
Summary: possibly the 3 Shelly1 were flashed correctly, but somehow the browser cache tricked me. Now I have 2 Shellies which were resetted (power cycle 5 on/off on the switch), and now I'm looking for a way to get them back into the system. Is there a way?

ursusHU, 2020-06-05:
Ok, I have managed to get the unit to AP mode with the 10 on/offs and it is running your firmware. Sorry for bothering you with this issue.

rojer, 2020-06-05:
ok, no problem

closed 2020-06-05 by rojer