mongoose-os-apps/shelly-homekit #89: Extended documentation

alexanderbjelos, 2020-06-14:
Hello, 

can anyone describes the functionality of the different types and input-modes?

Alex

pivale, 2020-07-14:
Hello all,

First of all, kudos to everyone working on the project!
I still don't have a shelly, but because of this project I will order a few of them soon.

Would also be interested in reading some more documentation, specifically regarding:
- types of wall "dumb" switches supported (or what type of states are supported on the switch input contacts)
- recommended connections for different approaches, like behind a switch, behind a power outlet, behind a double switch, behind a double power outlet, behind a two button roller shutter. Or linking to the supplier page with that information.

andyblac, 2020-09-20:
take a look at the new [wiki](https://github.com/mongoose-os-apps/shelly-homekit/wiki)

closed 2020-09-20 by andyblac